Ysurac/openmptcprouter #1888: Erreur de certificat à la fin de l'installation

Rathorian, 2021-06-09:
Bonjour

J'ai installé une première fois sur un VPS OVH le script OpenMPTCProuter et à la fin j'ai toujours une erreur de certificat : 
```
[Wed Jun  9 14:08:10 CEST 2021] Create new order error. Le_OrderFinalize not found. {
  "type": "urn:ietf:params:acme:error:rateLimited",
  "detail": "Error creating new order :: too many certificates already issued for: ovh.net: see https://letsencrypt.org/docs/rate-limits/",
  "status": 429
}
[Wed Jun  9 14:08:10 CEST 2021] Please add '--debug' or '--log' to check more details.
[Wed Jun  9 14:08:10 CEST 2021] See: https://github.com/acmesh-official/acme.sh/wiki/How-to-debug-acme.sh
```

J'imagine que c'est parce que j'ai pas changé le nom du VPS et qui porte du coup le nom de domaine `ovh.net` mais est-ce bloquant pour la suite ?

Merci beaucoup

Ysurac, 2021-06-09:
Non ce n'est pas bloquant pour la suite et c'est effectivement parce que le VPS porte le nom ovh.net.

Rathorian, 2021-06-09:
> 
> 
> Non ce n'est pas bloquant pour la suite et c'est effectivement parce que le VPS porte le nom ovh.net.

Merci pour la réponse ultra rapide.

Je ferme le ticket.

Bonne journée

closed 2021-06-09 by Rathorian