Ysurac/openmptcprouter #1889: 404 When trying to access Login Page from either RPi4 or VirtualBox

thetechgeekster, 2021-06-09:
## Expected Behavior
When going to 192.168.100.1 it should present the Login Screen.

## Current Behavior
When I try to log in from either RaspberryPi 4 or a virtualized instance in VirtualBox, I get a 404 error in every browser I have available. 
[image]

Ysurac, 2021-06-09:
What release do you use ?
Make sure you don't have something already using this IP.

thetechgeekster, 2021-06-09:
I am using openmptcprouter-v0.57.3-r0+15225-bfc433efd4-x86-64-generic-ext4-combined-efi . For the virualized instance I only have a USB Network adapter passthroughed directly to the VM with no other network adapters (or DHCP)

Ysurac, 2021-06-09:
Why using efi release ? You have a EFI only system ?
Make sure you have an IP on 192.168.100.x network for your client computer (even if you shouldn't have a 404 if not the case).

thetechgeekster, 2021-06-09:
I can try with the non EFI version. I though it might be better. Through the passthroughed adapter I get this IP 
[image]
so I think it is getting an IP from openwrt. Could it be that I need to setup a static IP in the openwrt instance via terminal?

thetechgeekster, 2021-06-09:
This is the result of ip addr in the VM by the way
[image]

Ysurac, 2021-06-09:
Both seems to be correctly configured.
You can try a "curl 127.0.0.1" on the OpenMPTCProuter VM console to check if you get something or not.

closed 2021-09-13 by github-actions[bot]