Ysurac/openmptcprouter #1891: Support for inexpensive GLiNet routers

albertdb, 2021-06-10:
Hi,

I think it would be awesome to support for some inexpensive GLiNet routers such as the GL-MT300N-V2. They are already based on OpenWRT, so most of the work should already be done (I hope it so).

That kind of routers won't allow aggregating big pipes, but for cases where you just need to (or can) bond slow+slow to obtain something decent, would be nice.

Since they have at least 2 Ethernet ports, it would also allow many combinations (ethX as WANs and wlan as LAN, or eth0+wlan as WANs and eth1 as LAN), something that cannot be directly done on a NanoPI RS2 or a Raspberry Pi, due to lack of interfaces.

Thank you very much

Adorfer, 2021-06-10:
just as a warning, from using using openvpn and wireguard with those unist: do not expect more than 12MBit/s as total badwidth from a  MT7628AN SoC!

Ysurac, 2021-06-10:
CPU is for me really too slow, and there is also a small flash and small memory.

albertdb, 2021-06-10:
Yes, I know the limitations, but when you have to bear with sub-5Mb connections, 12Mb is like a dream.

closed 2021-09-13 by github-actions[bot]