Ysurac/openmptcprouter #1893: BUG: using __this_cpu_add() in preemptible

rstanislav, 2021-06-11:
System is Raspberry pi 4B, raspbian 64 bit kernel v5.4.83 with last mptcp version (patch from here) , under load dmesg have every few seconds this message:
(is this case program was ss-redir, but there is few other cases with openvpn and other network applications, bug message is always the same (looks like some error in mptcp_retransmit_skb? )

> [ 2230.022188] BUG: using __this_cpu_add() in preemptible [00000000] code: ss-redir/520
> [ 2230.022218] caller is mptcp_retransmit_skb+0x14c/0x2a0
> [ 2230.022231] CPU: 1 PID: 520 Comm: ss-redir Tainted: G         C O      5.4.83-v8+ #1
> [ 2230.022237] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
> [ 2230.022242] Call trace:
> [ 2230.022251]  dump_backtrace+0x0/0x158
> [ 2230.022257]  show_stack+0x24/0x30
> [ 2230.022264]  dump_stack+0xd8/0x150
> [ 2230.022275]  __this_cpu_preempt_check+0x130/0x150
> [ 2230.022283]  mptcp_retransmit_skb+0x14c/0x2a0
> [ 2230.022290]  mptcp_write_xmit+0x3ec/0x460
> [ 2230.022301]  __tcp_push_pending_frames+0x60/0xe8
> [ 2230.022310]  tcp_push+0x100/0x1a0
> [ 2230.022316]  tcp_sendmsg_locked+0xaa8/0xd78
> [ 2230.022322]  tcp_sendmsg+0x40/0x60
> [ 2230.022330]  inet_sendmsg+0x4c/0x70
> [ 2230.022339]  sock_sendmsg+0x54/0x60
> [ 2230.022346]  __sys_sendto+0x11c/0x148
> [ 2230.022354]  __arm64_sys_sendto+0x30/0x40
> [ 2230.022363]  el0_svc_common.constprop.1+0x7c/0x1a8
> [ 2230.022371]  el0_svc_handler+0x34/0xa0
> [ 2230.022378]  el0_svc+0x8/0x204

I have opened issue also on MPTCP project page - https://github.com/multipath-tcp/mptcp/issues/429

Is there something i can do to help identify issue ?

Ysurac, 2021-06-11:
What is RPI load average when you have the issue ?

rstanislav, 2021-06-11:
ss-redir using about 4-10% cpu during speedtest(2 USB modems, ~ 15-17Mbit total speed), avr is around 3-4%, rpi only doing routing, minimal install without any additional programs. During speedtest or almost any other network load there is usually 4-6 BUG messages in dmesg

Ysurac, 2021-06-11:
USB modems use a powered USB hub ?

rstanislav, 2021-06-11:
They are directly connected to RPi, but RPi usb ports have + pin cut and directly connected to PSU 5.2v/5A with common ground. There is no messages about undervoltage or anything like that is PRi log.

Ysurac, 2021-06-11:
You can try to update to 0.58rc.

rstanislav, 2021-06-11:
I'm not using openmptcprouter, only kernel patches for kernel 5.4 from openmptcprouter repo, i have clean system with minimal packages & scripts for automated routing configuration, self build kernel and shadowssocks\openvpn compilled from sources.

After more heavy testing 1 more BUG in dmesg:

> [ 2802.063494] BUG: using __this_cpu_add() in preemptible [00000000] code: ss-redir/493
> [ 2802.063526] caller is mptcp_process_data_ack+0x67c/0xb18
> [ 2802.063541] CPU: 3 PID: 493 Comm: ss-redir Tainted: G         C O      5.4.83-v8+ #2
> [ 2802.063548] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
> [ 2802.063553] Call trace:
> [ 2802.063564]  dump_backtrace+0x0/0x158
> [ 2802.063572]  show_stack+0x24/0x30
> [ 2802.063581]  dump_stack+0xd8/0x150
> [ 2802.063593]  __this_cpu_preempt_check+0x130/0x150
> [ 2802.063602]  mptcp_process_data_ack+0x67c/0xb18
> [ 2802.063611]  mptcp_handle_options+0x1b0/0x5d0
> [ 2802.063622]  tcp_validate_incoming+0x40c/0x540
> [ 2802.063630]  tcp_rcv_state_process+0xe4/0xcf0
> [ 2802.063637]  tcp_v4_do_rcv+0x1ec/0x2c8
> [ 2802.063645]  mptcp_backlog_rcv+0xa4/0xd8
> [ 2802.063655]  __release_sock+0x78/0xc0
> [ 2802.063662]  release_sock+0x34/0x90
> [ 2802.063670]  mptcp_close+0x214/0x650
> [ 2802.063677]  tcp_close+0x53c/0x558
> [ 2802.063685]  inet_release+0x48/0x70
> [ 2802.063694]  __sock_release+0x4c/0xd0
> [ 2802.063702]  sock_close+0x24/0x38
> [ 2802.063712]  __fput+0x94/0x238
> [ 2802.063719]  ____fput+0x20/0x30
> [ 2802.063729]  task_work_run+0x88/0x138
> [ 2802.063735]  do_notify_resume+0x12c/0x678
> [ 2802.063743]  work_pending+0x8/0x14

closed 2021-06-21 by rstanislav