Ysurac/openmptcprouter #1897: After update from 0.57.3 to 0.58, my interface vanished

kobbybest, 2021-06-14:
## Expected Behavior

My four  interfaces don't vanished

## Current Behavior
After update  my four interfaces vanished
I have installed new image X86-64 bits on my nuc and restore my save

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.58
[image]

  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1026
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-06-14:
Check State->System log

kobbybest, 2021-06-14:
Mon Jun 14 19:18:34 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:18:34 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:18:36 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:18:36 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:18:41 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:18:44 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:18:44 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:18:44 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:18:45 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:18:46 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:18:47 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:18:49 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:18:49 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:18:49 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:18:49 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:18:50 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:18:50 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:18:56 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:18:58 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:18:58 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:18:59 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:00 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:01 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:01 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:03 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:03 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:04 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:04 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:05 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:05 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:10 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:13 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:13 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:13 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:14 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:16 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:16 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:18 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:18 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:18 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:18 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:19 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:19 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:25 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:27 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:28 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:28 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:29 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:30 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:30 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:32 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:33 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:33 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:33 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:34 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:34 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:40 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:42 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:42 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:42 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:43 2021 user.notice OMR-VPS: Can't get vps token, try later ()
Mon Jun 14 19:19:45 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:45 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:47 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:47 2021 user.notice OMR-VPS: Can't get vps token, try later
Mon Jun 14 19:19:48 2021 user.notice post-tracking-post-tracking: Set firewall on server
Mon Jun 14 19:19:48 2021 user.notice OMR-VPS: Can't get vps token, try later
[... cut ...]
Mon Jun 14 19:23:47 2021 daemon.info dnsmasq[3277]: read /etc/hosts - 4 addresses
Mon Jun 14 19:23:47 2021 daemon.info dnsmasq[3277]: read /tmp/hosts/dhcp.cfg01411c - 38 addresses
Mon Jun 14 19:23:47 2021 daemon.info dnsmasq-dhcp[3277]: read /etc/ethers - 0 addresses
Mon Jun 14 19:23:47 2021 user.notice omr-bypass: OMR-ByPass is running
Mon Jun 14 19:23:48 2021 user.notice mptcp: Reloading mptcp config due to ifup of wan6 (wan6)
Mon Jun 14 19:23:49 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0

kobbybest, 2021-06-14:
Dans assistant de configuration, je ne vois pas d'interfaces, j'ai donc essayé d'en rajouté une ou deux, rien ne s'est passé, par contre dans état, je retrouve mes 4 connections plus celle que j'ai crées
[image]
[image]

Du coup je ne peux même pas supprimer les deux que je viens de créer

Ysurac, 2021-06-14:
Il y a eu un soucis lors de la récupération de la sauvegarde, elle n'a visiblement pas été mise à jour.
Il faut aller dans Réseau->Interface et remettre chaque interface dans la zone WAN du firewall.

kobbybest, 2021-06-14:
A priori c'est déjà le cas, dois-je faire quelque chose de particulier en plus
[image]

Ysurac, 2021-06-14:
Je pense qu'il n'est pas dans la bonne zone, la zone a été renommée.
L'importation de la sauvegarde a été faite après la mise à jour ? Pas lors de la mise à jour via l'interface ?

kobbybest, 2021-06-14:
J'ai mis à jour le VPS puis j'ai flashé la nouvelle version avec etcher, puis restaurer ma sauvegarde

Ysurac, 2021-06-14:
Oui c'est le soucis. Je vais sortir une nouvelle version avec un fix lorsque l'on importe une vieille sauvegarde sur une version à jour.

kobbybest, 2021-06-14:
du coup comment je peux m'en sortir, reflasher avec etcher et reconfigurer à la main les 4 connections? et est-ce normal de ne pas voir les interfaces dans assistant de configuration ou c'est lié à la restauration ?

Ysurac, 2021-06-14:
Oui les soucis que tu peux avoir sont liés à la restauration (au final une vieille configuration non comprise).
Tu peux reflasher via etcher (ou via l'interface web sans garder la configuration) et reconfigurer à la main, ou alors tu peux reflasher via etcher (ou via l'interface web toujours sans garder la configuration) la version  que tu avais puis importer ta config et faire la mise à jour via l'interface web en gardant la configuration et elle sera mise à jour.

kobbybest, 2021-06-14:
J'ai fait la deuxième solution, donc restaurer une image vierge de la 057.3, restaurer ma configuration puis mise a jour en 0.58 par l'interface web en gardant la configuration, ca à l'air correcte, merci beaucoup de ton aide

kobbybest, 2021-06-14:
J'ai ce truc la qui boucle, tu saurais de quoi cela peut venir, ça a l'air d'être lié au wifi du NUC

Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.296913] WARNING: CPU: 1 PID: 8760 at cfg80211_probe_status+0x54c/0x770 [cfg80211]
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.300662] Modules linked in: ath9k ath9k_htc ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial pppoe ppp_async option iwlmvm iwldvm ipw cdc_mbim ath9k_hw ath wireguard usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx smsc75xx sierra_net sierra rtlwifi rtl8xxxu rtl8812au rtl8187 rt2x00usb rt2x00lib rndis_host qmi_wwan pppox ppp_generic pl2303 oti6858 mt7601u mos7720 mct_u232 mcs7830 mac80211 lzo libchacha20poly1305 libblake2s keyspan kalmia iwlwifi ipt_REJECT huawei_cdc_ncm garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cypress_m8 curve25519_x86_64 cp210x chacha_x86_64 ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem blake2s_x86_64 belkin_sa ax88179_178a asix ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.300712]  xt_connlimit xt_connbytes xt_condition xt_comment xt_cgroup xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor via_velocity via_rhine usbserial usbnet tulip ts_fsm ts_bm solos_pci slhc sky2 skge sis900 sis190 sch_cake rtl8150 r8712u(C) r8169 r8152 r8125 r6040 poly1305_x86_64 pegasus pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci mmc_spi macvlan lzo_decompress lzo_compress libcurve25519_generic libchacha libblake2s_generic kaweth iptable_raw iptable_nat iptable_mangle
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.321003]  iptable_filter ipt_ah ipt_ECN ipheth ip6table_raw ip_tables hso forcedeth ezusb ethoc et131x ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 e1000e e100 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_acm bnx2 atlantic atl2 atl1e atl1c atl1 asn1_decoder arptable_filter arpt_mangle arp_tables aquantia 8390 8139too 8139cp tcp_nanqinlang tcp_bbr2 sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred configs evdev i2c_gpio i2c_dev spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.342535]  ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig bonding 3c59x ip6_gre ip_gre gre ixgbevf ixgbe igb e1000 ifb nat46 ip6_udp_tunnel udp_tunnel sit sctp libcrc32c mdio ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel netlink_diag tunnel6 tunnel4 ip_tunnel hfcpci hfcmulti veth tun mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo autofs4 br2684 atm nls_utf8 zram zsmalloc natsemi eeprom_93cx6 ena sha256_ssse3 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg md5 kpp hmac ghash_generic gcm echainiv des_generic libdes deflate ctr cbc authenc crypto_acompress nls_iso8859_1 nls_cp437 vfat fat sdhci_pltfm xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl gpio_button_hotplug button_hotplug tg3 b44 ssb ptp realtek broadcom
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.364075]  bcm_phy_lib pps_core mii libphy
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.389766] CPU: 1 PID: 8760 Comm: rpcd Tainted: G        WC        5.4.106 #0
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.393867] Hardware name: BESSTAR TECH LIMITED GN34/AB1A, BIOS 0.02 03/31/2020
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.397896] RIP: 0010:cfg80211_probe_status+0x54c/0x770 [cfg80211]
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.401839] Code: 9a 00 00 00 48 89 df c7 45 d4 01 00 00 00 e8 eb b1 80 e0 85 c0 0f 84 35 ff ff ff 48 89 df e8 9b 37 be e0 b8 a6 ff ff ff eb 81 <0f> 0b 48 89 df e8 8a 37 be e0 b8 ea ff ff ff e9 6d ff ff ff eb db
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.405863] RSP: 0018:ffffc9000412fa50 EFLAGS: 00010202
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.409773] RAX: 0000000000000000 RBX: ffff88827055c600 RCX: 0000000000000000
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.413683] RDX: ffff888274a18008 RSI: 0000000000000000 RDI: ffff888274a18300
[... cut ...]
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.540239] RDX: 0000000000000000 RSI: 00007ffeb9d692c8 RDI: 0000000000000007
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.543989] RBP: 000000000000002e R08: 0000000000000000 R09: 0000000000000000
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.547699] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000000
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.551326] R13: 00007ffeb9d6f854 R14: 0000000000000001 R15: 0000562b5c6add8c
Mon Jun 14 20:15:37 2021 kern.warn kernel: [  444.554840] ---[ end trace c9b499a20b5792a3 ]---

kobbybest, 2021-06-14:
j'ai supprimé ce package kmod-cfg80211 et j'ai plus de message en boucle

closed 2021-06-14 by kobbybest