Ysurac/openmptcprouter #1899: RPI3B+ file system go in readonly for no reason

WazoAkaRapace, 2021-06-15:
## Expected Behavior
The system should run and not reboot by itself

## Current Behavior
After a fresh install, the filesystem is going to read only mode for no reason (New SD Card, flashed with etcher, then old settings imported)
The interfaces keep logging me off.

## Possible Solution
I've tried to repare the filesystem, but after 10 minutes of runtime, it corrupted itself again.

## Steps to Reproduce the Problem

  1. Update from 0.57.X to 0.58.1

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.58.1
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter platform: RPI3

Ysurac, 2021-06-15:
Try another SD card again, check power supply and temperature.
Check also State->kernel log.

AlexFspb, 2021-06-15:
I have the same problem on RPI3 with 0.57.3, but it started to be "read only mode" in few days usually.
It was checked with 2 different SD cards

Adorfer, 2021-06-15:
then check for the 2 remaining points.

FermedePommerieux, 2021-06-17:
Same problem with a RPI4B 3 hours after upgrade to 0.58.2 from the last 0.57, power supply and temperature as usual.
RW after a reboot

starkovv, 2021-07-20:
+1

Got the same issue on NanoPI R4S after 3 days from fresh install:
Filesystem is readonly
DNS issue: can't resolve hostname
Version 0.58.5

FermedePommerieux, 2021-07-20:
This could happen when you install/upgrade with the wrong disk image.

Adorfer, 2021-07-20:
or the 2 remaining points from above.

mvmaksnet, 2021-07-21:
@Ysurac  how to check logs?
I am using rpi 3bplus. I made new image and connected 2 usb modems and after I added third after saving it says readonly. Power is not problem here, I have powered usb hub. What can be problem?

treyguitar, 2021-09-04:
Have the same issue after power cycling, Kernel log says:
[   13.796332] EXT4-fs error (device mmcblk0p2): __ext4_new_inode:1114: comm uci: failed to insert inode 699: doubly allocated?
[   13.811393] EXT4-fs (mmcblk0p2): Remounting filesystem read-only

Ysurac, 2021-09-04:
You can try (when not read only) to set in System->Mount points, "Check filesystems before mount" and "Run filesystem check" on each partition. This will be the default in next release.

closed 2021-12-08 by github-actions[bot]