Ysurac/openmptcprouter #1900: IPv6 Prefix Size?

dragon2611, 2021-06-15:
OpenMPTCPRouter 0.58, when setting the IPv6 range to a /48 it seems OpenMPTCProuter is configuring a /60 on the LAN interface and only handing out a /62 via DHCPv6-PD.

If I Request a /56 with a prefix hint I get a /61 offered to my router (My lan is not directly connected to OpenMPTCProuter).

I don't see much in the way of options to change the Lan side IPv6 configuration.

Ysurac, 2021-06-15:
This can be defined via ```uci set network.lan.ip6assign='60' && uci commit network```.
I will add this setting back in Network->Interfaces in a 0.58.2 release.

Adorfer, 2021-06-15:
What is  (with 58.x) the minimum IPv6_prefix on the VPS for having "non-NAT6" on the OMR-LAN?

dragon2611, 2021-06-15:
I'd think you'd need a routed prefix unless you are going to resort to some horrible ND/Proxy hacks on the VPS side.

Adorfer, 2021-06-15:
My question is not "if", but "minimum size". I guess a /128 routet prefix will not be sufficient, which is totally obvious, but is it now a /64 or can it work with DHCPv6 (instead of SLAC), so even with a single additional /64? or is it still needing a /48?

Ysurac, 2021-06-15:
You need to have one IP for the VPS, one IP for the LAN interface and at least one IP for a client, so I would say a /126 if you use static IPs or DHCPv6 correctly configured (with IPs for VPS and LAN not also distributed)

Ysurac, 2021-06-15:
But don't forget that as output IP this will be VPS IPv6 if a proxy is used.

Adorfer, 2021-06-17:
"if a proxy is used": which operation mode of OMR has to be set "not use use proxy mode"? this referrers to v2ray vs shadowsocks vs. "none"? so it has to be set to "none" to use public IPv6 in the OMR-LAN?

closed 2021-09-21 by github-actions[bot]