Ysurac/openmptcprouter #1902: Passage de 0.58 a 0.58.1 par interface WEB reste bloqué à 100% en téléversement

kobbybest, 2021-06-15:
Affiche une fenêtre après la barre pour écrire le firmware

## Current Behavior
Reste bloqué à 100 %
J'ai également testé de 0.57.3 à 0.58.1 et même problème, cela fonctionne avec la 0.58

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Ysurac, 2021-06-15:
Via système->Mise à jour système ou via Système->Sauvegarde / Mise à jour du micrologiciel ?

kobbybest, 2021-06-15:
oui

Kalimeiro, 2021-06-15:
J'ai eu le même problème avec un RPI4, j'ai rafraichis la page puis j'ai recommencé et c'est passé.

kobbybest, 2021-06-16:
Oui le système ne supporte que EFI, je viens d'essayer avec la 0.58.2 et cela fait pareil, j'ai essayé de rafraichir plusieurs fois comme l'indique Kalimeiro ainsi que sur plusieurs pc et navigateurs et cela reste bloqué à 100%. Mon contournement est de sauvegarder la conf , reflasher et restaurer la conf.
A priori, l'image sans EFI fonctionne mais je ne peux plus booter.

YoloBaz, 2021-06-16:
Même problème ici. Bloqué à 100%...
openmptcprouter-v0.58.2-r0+16336-b36068d35d-x86-generic-generic-ext4-combined.img

Ysurac, 2021-06-16:
Oui j'ai eu le soucis sur une installation, mais quand j'ai voulu avoir l'erreur réelle en mettant le navigateur en mode développeur, c'est passé directement.
Sinon il est toujours possible de faire la mise à jour via SSH avec sysupgrade.
Avez-vous le soucis en utilisant "système->Mise à jour système" sinon (qui s'occupe lui même de télécharger la dernière image) ?

YoloBaz, 2021-06-16:
Je n'avais pas cette option: "système->Mise à jour système". J'ai tenté sysupgrade via SSH et cela a fonctionné.

Ysurac, 2021-06-16:
L'option est présente depuis la 0.58

YoloBaz, 2021-06-16:
J’étais avec la 0.57... j'essaierai cette option à la prochaine update, ce sera plus simple que par SSH (pour moi !). Merci !

Kalimeiro, 2021-06-16:
Je viens de tester la mise à jour 0.58.1 vers 0.58.2 via "système->Mise à jour système" : RAS.

kobbybest, 2021-06-17:
Je viens de tenter la maj de 0.58.1 vers 0.58.2 via système->Mise à jour système" , il a téléchargé l'image mais la mauvaise (sans l'efi) du coup il n'a jamais réussi a rebooter, c'est dommage car en effet cette option facilite grandement les maj, il faudrait rajouter une coche pour lui indiquer de prendre l'image EFI par exemple ou qu'il propose le choix avant de télécharger

Ysurac, 2021-06-17:
ah oui c'est un cas que j'ai pas pris en compte. Le mieux serait que je puisse détecter que l'image utilisée est déjà EFI...

Ysurac, 2021-06-17:
Ca sera corrigé dans la v0.58.3 qui est en compilation.

kobbybest, 2021-06-24:
via système->Mise à jour système" cela fonctionne maintenant correctement en choisissant la bonne image EFI

closed 2021-06-24 by kobbybest