Ysurac/openmptcprouter #1903: Bonding issue with 2 smart phones

goofguy1, 2021-06-15:
## Expected Behavior
TetheringWAN and TetheringWAN2 connections should be bonded together in aggregated mode, and not in failover mode

## Current Behavior
TetheringWAN and TetheringWAN2 are running in failover mode, so when WAN goes down WAN2 takes over with a short delay, instead of bonding the connections together. 

## Specifications

  - OpenMPTCProuter version:  OpenMPTCProuter v0.58.1 2021-06-15 04:55
  - OpenMPTCProuter VPS version:  OpenMPTCProuter VPS 0.1024
  - OpenMPTCProuter platform: RPI4/x86_64
[image]

Ysurac, 2021-06-15:
How do you check that speed is not aggregated ? Did you try using via ssh "omr-test-speed" and check in MPTCP->Bandwidth tab ?

goofguy1, 2021-06-15:
I was doing an RTMP test with my gopro and simulated a real world disconnect by unplugging a phone from USB tether and also tried ejecting the sim card tray to lose the connection but not the USB tether.

Both times resulted in livestream dropping connection instead of the 2nd phone taking over the load. Resulting in failover instead of aggregation as the signal came back up on the 2nd phone after 3-5 seconds

Ysurac, 2021-06-15:
So no issues on aggregation, only on failover ?
Check log in State->System log.

goofguy1, 2021-06-15:
Network bonding/aggregation isn't working, it results in failover

Ysurac, 2021-06-15:
So did you try test I tell you ?
Did you check also MPTCP support of each WAN ?

goofguy1, 2021-06-15:
[image]
[image]

Tested both on Enabled | Master + Backup | Master + Enabled | All instances resulted in failover. One sim card in the phone is carrying the other. Not aggregated.

goofguy1, 2021-06-15:
[image]

goofguy1, 2021-06-15:
2nd test with both tethering set to Enabled, First one was set to master + backup
[image]

Ysurac, 2021-06-15:
This doesn't seems to be aggregated. Did you check MPTCP support in Network->MPTCP of each WAN ?

umekzg, 2021-06-15:
Mobile phones are on the same subnet.
DHCP pool is hardcoded in android tethering service.

goofguy1, 2021-06-15:
> 
> 
> This doesn't seems to be aggregated. Did you check MPTCP support in Network->MPTCP of each WAN ?

I've checked, the settings are followed directly in the documentation

> 
> 
> Mobile phones are on the same subnet.
> DHCP pool is hardcoded in android tethering service.

so I have to swap to modems instead then? I can't use phones?

gnosoz, 2021-06-17:
Correct, you can use only one android phone per time and the other sim cards must be utilized with simple usb routers that can be found anywhere for 40eur

goofguy1, 2021-06-17:
> 
> 
> Correct, you can use only one android phone per time and the other sim cards must be utilized with simple usb routers that can be found anywhere for 40eur

Thank you, could you please link a few modems? I live in singapore and its hard to get unlocked modems so I can use worldwide sim cards when I travel to livestream. There's soo many to pick that I don't know which ones are compatible

closed 2021-06-20 by goofguy1