Ysurac/openmptcprouter #1904: OpenWRT Base and Packages Versioning

SonicFM, 2021-06-16:
Hi,
may i ask what version of OpenWRT "OpenMPTCPRouter" is based on?
Latest stable releases like 19.07.7?

Im asking because i cannot decode if, for example, https://github.com/Ysurac/openmptcprouter/commit/98661eda4315c7a5f5ac65a52035fe460dfee187 is OpenWRTs stable branch or master/develop aka snapshot or if its always "Snapshot" but pinpointed to an specifiy commit, to avoid incompatibilities.

If the last one is the case, how do you decide when to pinpoint to an specific commit?

Since you use your own repo for the OpenWRT and your Packages, the same questions applys to the packages provides by OpenWRT. Are they snapshot or stable or maybe mixed?

Adorfer, 2021-06-16:
it is Openwrt-master afaik.

Ysurac, 2021-06-17:
Fixed commit on master branch of OpenWrt for all.

SonicFM, 2021-06-17:
Thx for the replies! :)
And how do you decide when to pinpoint to an specific commit? Only by Kernel-Version?

These questions came up, since i have problems with low tx-power with mt7615e Wifi-Chip/Card. Googling around tells me it should have been fixed by the devs of the driver/firmware of kmod-mt7615-firmware / kmod-mt7615e at around 20th September 2020, if i'm not mistaken.

closed 2021-09-02 by SonicFM