Ysurac/openmptcprouter #1905: VPS Provider: Linode - Tested yet?

gsxryan, 2021-06-16:
I didn't see linode tested in the compatibility list.  They have very good offering in US.  The $10 VPS having 8Gb down 2Gb up.  

I'm having trouble with it working, which is why i'm asking if it has been tested yet.

I'll be creating a vultr VPS to compare, but i need more than 1Gb transfer to see if i can support 2x 1Gb links.

gsxryan, 2021-06-17:
I wiped and reinstalled with the latest version and it seems to be working great now with a single ISP link.  I'll update once I get two links connected.

gsxryan, 2021-07-05:
Tested on a 2.2Gb Merged Fiber network: 1 @ 1200, 1@950.
Using mostly defaults besides switching cubic to BBR

[image]

Ysurac, 2021-07-05:
Seems to work well.
I added Linode in the documentation.
Thanks

gsxryan, 2021-07-05:
Thanks!  I've been happy with it.  I added a few notes above after you commented:

The $5 VPS can burst for a few seconds higher than 1Gb, but the Linode QoS quickly regulates it back down.
And, no encryption was used for the 2.1Gb screenshot.  I want to do further testing on that as well.

closed 2021-10-08 by github-actions[bot]