Ysurac/openmptcprouter #1908: Realtime Traffic scale keeps getting bigger even if no downloads or uploads have occurred.

darthclide, 2021-06-17:
## Expected Behavior
The Realtime Traffic page should scale with recent traffic.

## Current Behavior
It scales bigger and bigger until all traffic is shown as just a few pixels at the bottom as shown here in this screenshot:
[image]

## Possible Solution
Revert the realtime traffic page to how it was in the previous version?

## Steps to Reproduce the Problem

1. Refresh page
2. Wait a few minutes.

## Context (Environment)
Instead of trying to hunt down numbers to match wan names, it is far easier to look at colors to quickly check which device is using the most data at a given moment.

## Specifications

  - OpenMPTCProuter version: 0.58.2
  - OpenMPTCProuter VPS version: Version 0.1026 5.4.100-mptcp
  - OpenMPTCProuter platform: RPI4

darthclide, 2021-06-18:
Upon further examination, simply refreshing the page seems to create random results:
[image]

darthclide, 2021-06-23:
I keep looking at my log to see if there are any errors, but this is still happening. Any suggestions would be nice.
[image]

Ysurac, 2021-06-23:
I don't think it's a big issue, only a display choice.

darthclide, 2021-06-23:
Hmmm, I never made any changes? Where is this "display choice" setting?