Ysurac/openmptcprouter #1910: Aggregation KO after upgrade to 0.58.3

hlehoux2021, 2021-06-19:
## Expected Behavior

Expected behavior is to aggregate 2 4G routers with Openmptcprouter.
Has been working for several month with version 0.57.x

Thank you so much for help

## Current Behavior

Only one wan is recognized and working. the other is KO, "pas de reponse au ping"
System log reports:
Sat Jun 19 14:02:27 2021 user.notice post-tracking-post-tracking: wan3 (wan3) switched off because check error and ping from 192.168.2.2 error (4.2.2.1,8.8.8.8,80.67.169.12)
Sat Jun 19 14:02:27 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.2.1 dev wan3
[image]

## Specifications

  - OpenMPTCProuter version: 0.58.3
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter platform: RPI4

closed 2021-06-19 by hlehoux2021

Adorfer, 2021-06-20:
@hlehoux2021 You gave up or you found a solution?

hlehoux2021, 2021-06-20:
##EDIT: solved by:

factory reset of both Archer MR600 and Archer MR200 modems, then firmware update
reinstall latest debian on vps
redo config from scratch of server abd vps with latest versions