Ysurac/openmptcprouter #1912: No Output, No server IP, No WAN IP, Can't get public IP from ShadowSocks

Yersi88, 2021-06-21:
## Expected Behavior

openmptcprouter displays green checkmark next to VPS

## Current Behavior

Everything actually appears to be working as expected but, Network Overview claims "No Output" next to VPS:
[image]

I don't even know what this means. Please help

## Specifications

  - OpenMPTCProuter version: 0.57.3 
  - OpenMPTCProuter VPS version: VPS 0.1024
  - OpenMPTCProuter platform: x86_64

**Config:** 

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.force='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='raspberrypi'
dhcp.@host[0].ip='192.168.100.156'
dhcp.@host[0].mac='DC:A6:32:A8:93:1F'
dhcp.@host[1]=host
dhcp.@host[1].name='-'
dhcp.@host[1].ip='192.168.100.190'
dhcp.@host[1].mac='24:62:AB:D6:E5:5C'
dhcp.@host[2]=host
dhcp.@host[2].name='-'
dhcp.@host[2].ip='192.168.100.164'
dhcp.@host[2].mac='44:A5:6E:5C:78:41'
dhcp.@host[3]=host
[... 1578 lines cut ...]
v2ray.omrtest.listen='127.0.0.1'
v2ray.omrtest.s_socks_auth='noauth'
v2ray.omrtest.s_socks_udp='1'
v2ray.omrtest.s_socks_ip='127.0.0.1'
v2ray.omrtest.s_socks_userlevel='0'
v2ray.main_reverse=reverse
v2ray.main_reverse.enabled='1'
v2ray.main_reverse.bridges='omrbridge|omr.lan'
v2ray.omrexit=outbound
v2ray.omrexit.protocol='freedom'
v2ray.omrexit.tag='out'
v2ray.omrrouting=routing_rule
v2ray.omrrouting.type='field'
v2ray.omrrouting.inbound_tag='omrbridge'
v2ray.omrrouting.outbound_tag='omrout_tunnel'
v2ray.omrrouting.domain='full:omr.lan'
v2ray.omrroutingo=routing_rule
v2ray.omrroutingo.type='field'
v2ray.omrroutingo.inbound_tag='omrbridge'
v2ray.omrroutingo.outbound_tag='out'
v2ray.main_routing=routing
v2ray.main_routing.enabled='1'
v2ray.main_routing.rules='omrrouting' 'omrroutingo'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'

Yersi88, 2021-06-21:
Correction, everything is not working. DNS not working, browser not resolving web pages unless DNS is manually configured in windows.

Ysurac, 2021-06-21:
Update to latest release.

Yersi88, 2021-06-21:
> Update to latest release.

Same after updating both router (restoring config) and vps + vps reboot

Ysurac, 2021-06-21:
You should do a fresh VPS install because all your keys/VPS IP are public now, you didn't remove them before the past.
What is your country ? You are not in a country where shadowsocks can be blocked ?
Ports are open on VPS side ? Did you try with only one WAN ?
You can try another proxy by checking advanced setting box in wizard page.

Yersi88, 2021-06-21:
> You should do a fresh VPS install because all your keys/VPS IP are public now, you didn't remove them before the past.
Haha That's a woopsie, I did fresh install now!
> What is your country ? You are not in a country where shadowsocks can be blocked ?
UK
> Ports are open on VPS side ? Did you try with only one WAN ?
> You can try another proxy by checking advanced setting box in wizard page.

It appears to be working correctly now. I reinstalled the router again and this time I didn't import my old config  but set everything up manually. Might be some weirdness in my old config but I don't really know why cause it was working good until today! 

Anyway, thank you very much for support. I really appreciate when you're doing here <3

closed 2021-06-21 by Yersi88