Ysurac/openmptcprouter #1913: custom x86 hardware with 3 internal 4g Sierra pci modems

Nickmcl024, 2021-06-22:
[    0.000000] Linux version 5.4.106 (runner@fv-az208-531) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+16336-b36068d35d)) #0 SMP Thu Mar 25 05:03:24 2021
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=7e2ddbfd-02 rootwait console=tty0 noinitrd
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009c3ff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009c400-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000085c2afff] usable
[    0.000000] BIOS-e820: [mem 0x0000000085c2b000-0x0000000085c2bfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000085c2c000-0x0000000085c2cfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000085c2d000-0x000000008c4fdfff] usable
[    0.000000] BIOS-e820: [mem 0x000000008c4fe000-0x000000008c859fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008c85a000-0x000000008c898fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000008c899000-0x000000008d034fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000008d035000-0x000000008d3fefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000008d3ff000-0x000000008d3fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000008d400000-0x000000008fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000046dffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 3.0.0 present.
[    0.000000] DMI: AAEON Dejero Rugged Gateway Boxer-6639M/PBA-SKS01, BIOS V04AXM13 03/05/2019
[    0.000000] tsc: Detected 2400.000 MHz processor
[    0.001999] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.002000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.002006] last_pfn = 0x46e000 max_arch_pfn = 0x400000000
[    0.002009] MTRR default type: write-back
[    0.002010] MTRR fixed ranges enabled:
[    0.002011]   00000-9FFFF write-back
[    0.002011]   A0000-BFFFF uncachable
[    0.002012]   C0000-FFFFF write-protect
[    0.002012] MTRR variable ranges enabled:
[    0.002014]   0 base 00C0000000 mask 7FC0000000 uncachable
[    0.002014]   1 base 00A0000000 mask 7FE0000000 uncachable
[    0.002015]   2 base 0090000000 mask 7FF0000000 uncachable
[    0.002016]   3 base 008E000000 mask 7FFE000000 uncachable
[    0.002016]   4 base 008D800000 mask 7FFF800000 uncachable
[    0.002017]   5 disabled
[    0.002017]   6 disabled
[    0.002017]   7 disabled
[    0.002018]   8 disabled
[    0.002018]   9 disabled
[    0.002488] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.003114] last_pfn = 0x8d400 max_arch_pfn = 0x400000000
[    0.010830] found SMP MP-table at [mem 0x000fcd70-0x000fcd7f]
[    0.010840] Using GB pages for direct mapping
[    0.011398] ACPI: Early table checksum verification disabled
[    0.011401] ACPI: RSDP 0x00000000000F05B0 000024 (v02 ALASKA)
[    0.011404] ACPI: XSDT 0x000000008C8660B0 0000DC (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.011408] ACPI: FACP 0x000000008C88C5C0 000114 (v06 ALASKA A M I    01072009 AMI  00010013)
[    0.011413] ACPI: DSDT 0x000000008C866218 0263A5 (v02 ALASKA A M I    01072009 INTL 20160422)
[    0.011415] ACPI: FACS 0x000000008D034C40 000040
[    0.011417] ACPI: APIC 0x000000008C88C6D8 0000BC (v03 ALASKA A M I    01072009 AMI  00010013)
[    0.011420] ACPI: FPDT 0x000000008C88C798 000044 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.011422] ACPI: MCFG 0x000000008C88C7E0 00003C (v01 ALASKA A M I    01072009 MSFT 00000097)
[    0.011424] ACPI: SSDT 0x000000008C88C820 0003BC (v01 SataRe SataTabl 00001000 INTL 20160422)
[    0.011427] ACPI: FIDT 0x000000008C88CBE0 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[... 819 lines cut ...]
[    6.549055] usbserial: USB Serial support registered for moto_modem
[    6.549903] usbserial: USB Serial support registered for motorola_tetra
[    6.550741] usbserial: USB Serial support registered for novatel_gps
[    6.551560] usbserial: USB Serial support registered for hp4x
[    6.552353] usbserial: USB Serial support registered for suunto
[    6.553126] usbserial: USB Serial support registered for siemens_mpi
[    6.555385] wireguard: WireGuard 1.0.0 loaded. See www.wireguard.com for information.
[    6.556285] wireguard: Copyright (C) 2015-2019 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[    6.562801] usbcore: registered new interface driver cdc_mbim
[    6.563849] usbcore: registered new interface driver ipw
[    6.564645] usbserial: USB Serial support registered for IPWireless converter
[    6.567888] usbcore: registered new interface driver option
[    6.568700] usbserial: USB Serial support registered for GSM modem (1-port)
[    6.570498] usbcore: registered new interface driver qcserial
[    6.571439] usbserial: USB Serial support registered for Qualcomm USB modem
[    6.575108] usbcore: registered new interface driver rt2800usb
[    6.578679] usbcore: registered new interface driver rtl8192cu
[    6.581162] usbcore: registered new interface driver ath9k_htc
[    6.583115] kmodloader: done loading kernel modules from /etc/modules.d/*
[    9.630759] 8021q: adding VLAN 0 to HW filter on device eth0
[    9.655270] 8021q: adding VLAN 0 to HW filter on device eth3
[   12.594581] igb 0000:04:00.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   28.922675] igb 0000:08:00.0 eth3: igb: eth3 NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
[   60.443699] i915 0000:00:02.0: Failed to load DMC firmware i915/skl_dmc_ver1_27.bin. Disabling runtime power management.
[   60.444559] i915 0000:00:02.0: DMC firmware homepage

Ysurac, 2021-06-22:
Please use the template.
I need versions, and to know what is the issue...

Nickmcl024, 2021-06-22:
I
Version：5.4.100-mptcp 0.1026

Load：0.04, 0.03, 0.00

Uptime：1d 17h 49m 19s

Proxy traffic：1.03 GiB

VPN traffic：7.36 MiB

Total traffic：1.04 GiB
IPv4 Internet

Connected：YES

IPv6：-

System
System
Uptime：16h 12m 57s

Local Time：2021-06-22 06:33:35

Load Average：0.08, 0.11, 0.09

Kernel Version：5.4.106

Model：AAEON Dejero Rugged Gateway Boxer-6639M

Architecture：Intel(R) Core(TM) i7-6700TE CPU @ 2.40GHz

Firmware Version：openmptcprouter v0.58.3 r0+16336-b36068d3

dont know how to use the 3 internal lte modems that are in the uint

closed 2021-09-26 by github-actions[bot]