Ysurac/openmptcprouter #1916: No server IP, No public IP, Can't get public IP from ShadowSocks

Yersi88, 2021-06-22:
## Expected Behavior
Green checkmark next to VPS

## Current Behavior

Similar issue experienced yesterday on old version of OpenMPTCPRouter. 
[image]

## Specifications

  - OpenMPTCProuter version: 0.58.4
  - OpenMPTCProuter VPS version: 0.1028 5.4.100-mptcp
  - OpenMPTCProuter platform: x86_64

Yersi88, 2021-06-22:
**### Update:** 

When I installed new version of Router I made backup of default config. Restoring default config and setting up manually made the problem go away. 

Very strange, any idea what might be causing this?

Ysurac, 2021-06-22:
Check State->System log

Yersi88, 2021-06-22:
> Check State->System log

I will create log if it happens again! I don't think it will take long :(

goggles24, 2021-06-28:
+1 I'm also seeing this. I will grab a log next time it occurs again.
I am also with Hostworld for my VPS, could be a factor.

Yersi88, 2021-09-26:
I can confirm that this issue didn't manifest itself anymore.

closed 2021-09-26 by Yersi88