Ysurac/openmptcprouter #1917: Can't get Starlink and Tmobile 5g to aggregate.

jakobstavenau, 2021-06-24:
## Expected Behavior
I am excepting that when I bond starlink and tmobile 5g internet together I will see increased bandwidth when I download and upload files.

## Current Behavior
When I do omr-test-speed the wan that is used is whichever is designated as the master. If I do starlink as master then I get 10-15MB/s down. If I do tmobile as master then I get 35-40MB/s down. They do not combine and under MPTCP->Bandwitdh I see that the non master sits almost idle.

## Specifications

  - OpenMPTCProuter version: v0.58.2
  - OpenMPTCProuter VPS version: 0.1026 5.4.100mptcp
  - OpenMPTCProuter platform: x86 i3 NUC

-I am using shadowsocks for proxy and Glorytun UDP
-Multipath Scheduler is default and using cubic for congestion control
-MPTCP support checks good on both wans
-SQM has no effect on or off for aggregation
-VPS is an OVH 2core 500Mbps 4GB ram nearest to my location
-I have tried V2ray and different VPN's they either make things less stable ie (very low bandwidth) or they just break things all together but I am open to trying anything. I am very appreciative of all the hard work you have done any help is welcome. Please let me know what else I can provide to make troubleshooting easier.

Also much less priority side note I cannot get the static route to starlink to show stats on the app. Any suggestions?

[image]
[image]

Ysurac, 2021-06-24:
When you check for MPTCP support, you get the "TCPOptionMPTCPCapable [...] Sender's Key" at the end for both ?
Can you check what you have in State->System log ?

jakobstavenau, 2021-06-24:
> 
> 
> When you check for MPTCP support, you get the "TCPOptionMPTCPCapable [...] Sender's Key" at the end for both ?
> Can you check what you have in State->System log ?

Both end in Sender's Key yes.
I don't see State -> System log.... where do i look?

jakobstavenau, 2021-06-24:
Do you mean status systemlog?

Ysurac, 2021-06-24:
yes...

jakobstavenau, 2021-06-24:
Thu Jun 24 10:49:07 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.118 5a:45:d4:18:0f:57
Thu Jun 24 10:49:07 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.118 5a:45:d4:18:0f:57 Shirlees-Iphone
Thu Jun 24 10:51:41 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 10:51:41 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 10:51:43 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 10:51:43 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 10:53:06 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 10:53:06 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 10:53:07 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 10:53:07 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 10:55:19 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.169 12:54:b2:1a:8a:c5
Thu Jun 24 10:55:19 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.169 12:54:b2:1a:8a:c5 Shirlee-s-iPad
Thu Jun 24 10:56:55 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 10:56:55 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 10:58:35 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 10:58:35 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:00:26 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 11:00:26 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:00:27 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 11:00:27 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:01:27 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.173 42:f4:29:03:01:50
Thu Jun 24 11:01:27 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.173 42:f4:29:03:01:50 Jakes-Iphone
Thu Jun 24 11:01:48 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 11:01:48 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:04:49 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 11:04:49 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:05:26 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 11:05:26 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:18:02 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 11:18:02 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:22:09 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 11:22:09 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:22:23 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 11:22:23 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:22:25 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc
Thu Jun 24 11:22:25 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.153 38:ba:f8:e7:c0:fc DESKTOP-F32V8GC
Thu Jun 24 11:29:31 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.217 8a:f6:5b:1b:69:52
Thu Jun 24 11:29:31 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.217 8a:f6:5b:1b:69:52 ShirleensIphone
Thu Jun 24 11:35:04 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.217 8a:f6:5b:1b:69:52
Thu Jun 24 11:35:04 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.217 8a:f6:5b:1b:69:52 ShirleensIphone
Thu Jun 24 11:36:19 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.217 8a:f6:5b:1b:69:52
Thu Jun 24 11:36:19 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.217 8a:f6:5b:1b:69:52 ShirleensIphone
Thu Jun 24 11:36:32 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.192 f8:75:a4:c5:5a:e8
Thu Jun 24 11:36:32 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.192 f8:75:a4:c5:5a:e8 lenovo
Thu Jun 24 11:37:21 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.217 8a:f6:5b:1b:69:52
Thu Jun 24 11:37:21 2021 daemon.info dnsmasq-dhcp[9434]: DHCPACK(eth0) 192.168.100.217 8a:f6:5b:1b:69:52 ShirleensIphone
Thu Jun 24 11:39:17 2021 daemon.info dnsmasq-dhcp[9434]: DHCPREQUEST(eth0) 192.168.100.217 8a:f6:5b:1b:69:52
[... cut ...]
Thu Jun 24 16:25:25 2021 daemon.info dnsmasq[20296]: using only locally-known addresses for domain lan
Thu Jun 24 16:25:25 2021 daemon.info dnsmasq[20296]: using 1 more local addresses
Thu Jun 24 16:25:25 2021 daemon.info dnsmasq[20296]: read /etc/hosts - 4 addresses
Thu Jun 24 16:25:25 2021 daemon.info dnsmasq[20296]: read /tmp/hosts/dhcp.cfg01411c - 18 addresses
Thu Jun 24 16:25:25 2021 daemon.info dnsmasq-dhcp[20296]: read /etc/ethers - 0 addresses
Thu Jun 24 16:25:25 2021 user.notice omr-bypass: OMR-ByPass is running

jakobstavenau, 2021-06-24:
Small update after rebooting everything. Seems like they are bonding but the connection is very unstable. Starts to ramp up speed but then lands on zero.

[image]
[image]

Ysurac, 2021-06-24:
You may need to enable SQM and set download/upload speed.

jakobstavenau, 2021-06-24:
> 
> 
> You may need to enable SQM and set download/upload speed.

Just tried that, made average download 703K with everything going through starlink (its master)

Ysurac, 2021-06-24:
SQM shouldn't slow your speed more than what is set as download/upload. Check that it's set correctly (in Kbit/s, not Mbits/s)

jakobstavenau, 2021-06-24:
> 
> 
> SQM shouldn't slow your speed more than what is set as download/upload. Check that it's set correctly (in Kbit/s, not Mbits/s)

[image]
[image]
[image]

jakobstavenau, 2021-06-24:
[image]

Adorfer, 2021-06-25:
> They look aggregated but why are they so slow?

In this case please update the issue (expectation/title)
since the issue is currently "does not aggregate". and now aggregation works. I would consider that as solved. 
In other words: 
- What are the "raw line WAN speeds" you do aggregate
- Please have a look at the CPU-Load on VPS & OMR during your tests, i would assume that the VPS used maxes out at about 170-210MBit/s netto traffic with SQM enabled. 

In plus i have a slight feeling, that there some kind of mixup between MBit/s and MB/s in the previous postings.

jakobstavenau, 2021-06-25:
There is no mixup I have posted pictures of all the speeds. 

Starlink gets 150Mbps or so
T-Mobile can get up to 350Mbps

Together they get about 5MB/s aggregated. 

I'll mark it solved. I'm also giving up on troubleshooting.

closed 2021-06-25 by jakobstavenau