Ysurac/openmptcprouter #1918: BLOCK IP INTERNET

sonnyxp2008, 2021-06-24:
Hi everyone, I need to stop the outbound and inbound traffic on a Korean server. Basically some devices have malware and send spam to this server. I paste the information of the server in question [Spam from: 77.xxxxx]
sent 2021-6-21 11:42:50 +0200 (2021-06-21 11:42:50 +0200 on server) (1.1 seconds) Display graph for this sender
from v01qaj7-xxxstserver.com [77.39.211.52] - password@plusnet.co.kr
to [211.231.108.175] - s2150900@daum.net
via [127.0.0.1] - (version 5.18.8)
subject 너의 스마트 폰
size 15596 bytes.

the information is the ip address 211.231.108.175 and I would like to block just that.

Thank you-->

Ysurac, 2021-06-24:
On the VPS, you can add in /etc/shorewall/rules something like
 ```
DROP            net:211.231.108.175        all
DROP            all                                     net:211.231.108.175
```
And ```/etc/init.d/shorewall reload```

sonnyxp2008, 2021-06-24:
sorry i just can't find it

Ysurac, 2021-06-24:
what you can't find ?

sonnyxp2008, 2021-06-24:
it does not find the directory neither on the vps nor on OMR

Ysurac, 2021-06-24:
You have the VPS part installed ? Shorewall is used as firewall on VPS part and ```/etc/shorewall/rules``` file always exist.

sonnyxp2008, 2021-06-24:
that's what it tells me
Last login: Thu Jun 24 10:19:33 2021 from 79.xxxx
root@v01qaj7-:~# /etc/shorewall/rules
-bash: /etc/shorewall/rules: Permesso negato
root@v01qaj7-:~# /etc/shorewall/rules
-bash: /etc/shorewall/rules: Permesso negato
root@v01qaj7-:~#  /etc/shorewall/rules
-bash: /etc/shorewall/rules: Permesso negato
root@v01qaj7-:~# Last login: Thu Jun 24 10:19:33 2021 from 79.31.177.70
root@v01qaj7-:~# /etc/shorewall/rules
-bash: /etc/shorewall/rules: Permesso negato
root@v01qaj7-:~#  /etc/shorewall/rules
-bash: /etc/shorewall/rules: Permesso negato
root@v01qaj7-:~#

Ysurac, 2021-06-24:
You need to edit it with ```nano``` or what you want, it's not an executable.

sonnyxp2008, 2021-06-24:
sorry for my ignorance but where should I put it between these lines?

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#       Accept connection from port > 65000 for shadowsocks and glorytun on the>
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#       Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#       DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT) $FW             vpn
#
#       Redirect all port from 1 to 64999 to the VPN client from the network
#
DNAT            net             vpn:$OMR_ADDR   tcp     1-64999
DNAT            net             vpn:$OMR_ADDR   udp     1-64999
ACCEPT          net             $FW             tcp     65101   # OMR openmptcp>
ACCEPT          net             $FW             udp     65101   # OMR openmptcp>
ACCEPT          net             $FW             udp     65311   # OMR openmptcp>

Ysurac, 2021-06-24:
At the end of the file.

sonnyxp2008, 2021-06-24:
OK thanks so we can see if it works?

Ysurac, 2021-06-24:
You have done a ```/etc/init.d/shorewall reload``` ?
As this IP doesn't answer to ping, there is no simple way to check. If you have telnet, you can try a ```telnet 211.231.108.175 25``` if it's not answer then it's blocked.

sonnyxp2008, 2021-06-28:
Hi Ysurac, the rule is also valid for url?

Ysurac, 2021-06-28:
No, it's only to block IP.
To block a domain, you can use on the router, in Network->DHCP&DNS, dns forwarding, add a domain like the already set "/use-application-dns.net/"

closed 2021-10-01 by github-actions[bot]