Ysurac/openmptcprouter #1922: ntfs-3g et samba disparus après upgrade 0.58.4

gblanchet03, 2021-06-28:
j'ai besoin des packages ntfs-3g et samba pour utiliser mon disque de 1To connecté en usb
où les trouver ?

Ysurac, 2021-06-28:
They should still be in System->Software after updating package list.

gblanchet03, 2021-06-28:
This
Downloading https://packages.openmptcprouter.com/v0.58.4/x86_64/luci/Packages.gz
*** Failed to download the package list from https://packages.openmptcprouter.com/v0.58.4/x86_64/luci/Packages.gz
and that
* opkg_download: Failed to download https://packages.openmptcprouter.com/v0.58.4/x86_64/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.
 
La commande opkg update a échoué avec le code 10.

Ysurac, 2021-06-28:
Il faut vérifier que OpenMPTCProuter a bien la connexion vers Internet qui fonctionne.
Aussi possible de lancer la commande via SSH, en utilisant ```opkg update```

gblanchet03, 2021-06-28:
J'ai ça aussi quand je lance :
https://192.168.100.1/cgi-bin/luci/
typeError :
boardinfo.release is undefined

gblanchet03, 2021-06-29:
@Ysurac  j'ai trouvé 
c'est dnsmasq qui ne se lançait pas pour cause de double baux statiques.
Du coup j'avais soupçonné eth1 sur mon overthebox+ j'ai mis un bridge br-lan sur eth0.2 avec ip statique 192.168.100.1
et tout fonctionne. Je vais mettre à jour mon blog et ouvrirait une nouvelle issue sur ce sujet.
En tout cas on peut fermer celle-ci. Merci

closed 2021-10-02 by github-actions[bot]