Ysurac/openmptcprouter #1923: VPS update in Debian

jcouch1, 2021-06-28:
So I've had my openmptcprouter setup running for close to a year now. I'm running OpenMPTCProuter version 0.57.3 and version 0.58.4.

I'm trying to update my VPS using the instructions in the documentation (https://github.com/Ysurac/openmptcprouter/wiki/Install-or-update-the-VPS).

when i run these commands
wget https://www.openmptcprouter.com/server/debian10-x86_64.sh
sh debian10-x86_64.sh

It keeps me on the same version, never updating.

What am I doing wrong? How do I get it updated?

kams19, 2021-06-28:
Run the script twice without reboot in between. After running it twice
reboot it. That's what worked for me

Ysurac, 2021-06-28:
@jcouch1 check that there is not already a debian10-x86_64.sh script before running the wget command (else you run old release script).

jcouch1, 2021-06-28:
Ok, I know nothing about Debian command line stuff. Is this something you can explain easily? This is probably my problem.

Ysurac, 2021-06-28:
Use this command ```wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh```
or do a ```rm debian10-x86_64.sh``` before your previous wget command

closed 2021-10-01 by github-actions[bot]