Ysurac/openmptcprouter #1924: Intermittent connection, stutters / disconnections in video calls or games, Slow wireguard connection on connected devices

Yersi88, 2021-06-30:
## Expected Behavior

Stable connection without interruptions.

## Current Behavior

1) Stutters in Video Calls (video calls tend to freeze for couple of seconds) 
2) Stutters / disconnections in multiplayer games 
3) SLOW Wireguard connection speeds on connected devices. (I have wireguard clients on connected devices to reach my remote network but the speed tends to be very bad sometimes below 1 MBps and I do experience intermittent connection as well) 

I don't know if these issues are connected but OpenMPTCPRouter tends to be providing unstable connection. I would like to know if there's something I can do about this cause it gets very frustrating. 

## Specifications

  - OpenMPTCProuter version: 0.58.4
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-06-30:
You can try what I put in https://github.com/Ysurac/openmptcprouter/wiki/Optimization

Yersi88, 2021-06-30:
Thank you, I tried V2RAY and BBR2. I will update on results!

Yersi88, 2021-06-30:
Unfortunately, this did not improve the situation, might even make it worse :( 

Any other suggestions? 

And some logs: 
```
Wed Jun 30 23:17:47 2021 user.notice v2ray: Rules UP
Wed Jun 30 23:17:47 2021 user.notice v2ray: Reload omr-bypass rules
Wed Jun 30 23:17:47 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jun 30 23:17:50 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jun 30 23:17:50 2021 daemon.info dnsmasq[19376]: read /etc/hosts - 4 addresses
Wed Jun 30 23:17:50 2021 daemon.info dnsmasq[19376]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Jun 30 23:17:50 2021 daemon.info dnsmasq-dhcp[19376]: read /etc/ethers - 0 addresses
Wed Jun 30 23:17:50 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jun 30 23:17:51 2021 user.notice v2ray: Reload omr-bypass rules
Wed Jun 30 23:17:51 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jun 30 23:17:57 2021 user.notice OMR-VPS: Can't get vps token, try later
Wed Jun 30 23:17:57 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jun 30 23:17:57 2021 daemon.info dnsmasq[19376]: read /etc/hosts - 4 addresses
Wed Jun 30 23:17:57 2021 daemon.info dnsmasq[19376]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Jun 30 23:17:57 2021 daemon.info dnsmasq-dhcp[19376]: read /etc/ethers - 0 addresses
Wed Jun 30 23:17:57 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jun 30 23:17:57 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jun 30 23:18:01 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jun 30 23:18:01 2021 daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 198.41.212.162, 1.0.0.1, 212.27.48.10, 198.27.92.1, 151.101.129.164, 77.88.55.77)
Wed Jun 30 23:18:01 2021 user.notice v2ray: Rules DOWN
Wed Jun 30 23:18:01 2021 daemon.info dnsmasq[19376]: read /etc/hosts - 4 addresses
Wed Jun 30 23:18:01 2021 daemon.info dnsmasq[19376]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Jun 30 23:18:01 2021 daemon.info dnsmasq-dhcp[19376]: read /etc/ethers - 0 addresses
Wed Jun 30 23:18:01 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jun 30 23:18:01 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jun 30 23:18:03 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jun 30 23:18:04 2021 daemon.info dnsmasq[19376]: read /etc/hosts - 4 addresses
Wed Jun 30 23:18:04 2021 daemon.info dnsmasq[19376]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Jun 30 23:18:04 2021 daemon.info dnsmasq-dhcp[19376]: read /etc/ethers - 0 addresses
Wed Jun 30 23:18:04 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jun 30 23:18:15 2021 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 1.1.1.1)
Wed Jun 30 23:18:16 2021 daemon.info omr-tracker-v2ray: Reload V2Ray rules
Wed Jun 30 23:18:17 2021 user.notice v2ray: Rules UP
Wed Jun 30 23:18:17 2021 user.notice v2ray: Reload omr-bypass rules
Wed Jun 30 23:18:17 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jun 30 23:18:23 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jun 30 23:18:23 2021 daemon.info dnsmasq[19376]: read /etc/hosts - 4 addresses
Wed Jun 30 23:18:23 2021 daemon.info dnsmasq[19376]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Jun 30 23:18:23 2021 daemon.info dnsmasq-dhcp[19376]: read /etc/ethers - 0 addresses
Wed Jun 30 23:18:23 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jun 30 23:18:23 2021 user.notice v2ray: Reload omr-bypass rules
Wed Jun 30 23:18:23 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jun 30 23:18:28 2021 user.notice post-tracking-post-tracking: wan3 (usb1) switched off because No access to server API
Wed Jun 30 23:18:28 2021 user.notice post-tracking-post-tracking: Delete default route via 192.168.42.129 dev usb1
Wed Jun 30 23:18:28 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jun 30 23:18:29 2021 daemon.info dnsmasq[19376]: read /etc/hosts - 4 addresses
Wed Jun 30 23:18:29 2021 daemon.info dnsmasq[19376]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Jun 30 23:18:29 2021 daemon.info dnsmasq-dhcp[19376]: read /etc/ethers - 0 addresses
Wed Jun 30 23:18:29 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jun 30 23:18:29 2021 user.notice omr-bypass: Starting OMR-ByPass...
Wed Jun 30 23:18:33 2021 user.notice omr-bypass: Reload dnsmasq...
Wed Jun 30 23:18:33 2021 daemon.info dnsmasq[19376]: read /etc/hosts - 4 addresses
Wed Jun 30 23:18:33 2021 daemon.info dnsmasq[19376]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Wed Jun 30 23:18:33 2021 daemon.info dnsmasq-dhcp[19376]: read /etc/ethers - 0 addresses
Wed Jun 30 23:18:33 2021 user.notice omr-bypass: OMR-ByPass is running
Wed Jun 30 23:18:41 2021 user.notice OMR-VPS: Can't get vps token, try later (can't ping server vps on 162.220.245.215, server API answer on 162.220.245.215)
Wed Jun 30 23:18:44 2021 user.notice post-tracking-post-tracking: wan3 (usb1) switched up
Wed Jun 30 23:19:27 2021 user.notice OMR-VPS: Can't get vps token, try later
Wed Jun 30 23:19:50 2021 daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 151.101.129.164, 77.88.55.77, 1.1.1.1)
Wed Jun 30 23:19:52 2021 user.notice v2ray: Rules DOWN
Wed Jun 30 23:20:17 2021 daemon.info omr-tracker-v2ray: V2Ray is up (can contact via http 198.41.212.162)
Wed Jun 30 23:20:17 2021 daemon.info omr-tracker-v2ray: Reload V2Ray rules
Wed Jun 30 23:20:17 2021 user.notice v2ray: Rules UP
Wed Jun 30 23:20:17 2021 user.notice v2ray: Reload omr-bypass rules
[... 4 lines cut ...]
[... cut ...]
Wed Jun 30 23:21:12 2021 daemon.info dnsmasq-dhcp[19376]: DHCPACK(eth0) 192.168.100.133 a4:e5:7c:8a:a6:82 ESP_8AA682
Wed Jun 30 23:21:26 2021 user.notice OMR-VPS: Setting server config error, try again
Wed Jun 30 23:21:27 2021 daemon.info omr-tracker-v2ray: V2Ray is down (can't contact via http 1.0.0.1, 212.27.48.10, 198.27.92.1)
Wed Jun 30 23:21:30 2021 user.notice v2ray: Rules DOWN
Wed Jun 30 23:21:45 2021 user.notice OMR-VPS: Can't get vps token, try later (can ping server vps on 162.220.245.215, no server API answer on 162.220.245.215)
```

Ysurac, 2021-07-01:
V2Ray doesn't seems to be a good solution for you. You can try others settings.
What is the latency of each connection to the VPS ?

Yersi88, 2021-07-01:
> V2Ray doesn't seems to be a good solution for you. You can try others settings.
> What is the latency of each connection to the VPS ?

Latencies are 12ms, 40ms and 32ms

Ysurac, 2021-07-01:
Should be ok. Load average on both router and VPS are ok ? You tested with less connection (maybe a connection have an issue) ?

Yersi88, 2021-07-01:
Loads seem good router is beefy machine and vps is good as well. I thought there might be problem with connection  so I switched 'master' to other connection. What's the difference between multipath enabled and master? does it make difference on stability? 

I also made new vps on digital ocean maybe hostworld is not up to par

Ysurac, 2021-07-01:
Only one connection need to be set as master, it's the connection used for initial connection (should be a stable connection).

Yersi88, 2021-07-01:
It's just unstable. Router will claim that connections go down like this: 
[image]
or it just says it no server response from connection. 

than after few seconds it comes back up. I'm quite sure the connections are stable and don't go down. 

Or it will claim that it can't connect to VPS admin script for few seconds and than comes back up again. Does this on both Hostworld and Digital Ocean.

Ysurac, 2021-07-01:
The warning are only warnings, not really a problem, some packets can be lost on connection.
Check system logs do know why a connection is detected as down. I show some "No access to server API" in previous log, this mean that server API doesn't answer from connection (it's a check to vps ip address port 65500), you can disable this check in System->OpenMPTCProuter, tab advanced setting, checkbox "Disable server http test".
Can you also give me a full screenshot of status page.

Yersi88, 2021-07-01:
[image]
This is how it tends to look, note that wan2 is now physically disconnected

Ysurac, 2021-07-02:
You should really check system log to know why there is an issue with wan1 for example.
It's a new install or an upgrade ?

Yersi88, 2021-07-02:
It's fresh install. 

I'm getting weird events like this on eth1:
[image]

Ysurac, 2021-07-02:
In this case it's really eth1 that is down, detected via kernel. It's like if physical link is removed.
(at least if there is no info before in log about a down of interface)

Yersi88, 2021-07-02:
I'm using cheap usb to ethernet network card for eth1 maybe it's just crappy card.

Ysurac, 2021-07-02:
r8152, it's the 2.5G usb ethernet no ? I have one and seems to be stable. You can try another USB port maybe.

Yersi88, 2021-07-02:
Yeah it is. Well I disconnected the usb card and plugged the router into switch set to eth0 with macvlan. Will see if there's improvement!

Adorfer, 2021-07-03:
usb-stuff does not work until reliable you put it on a reliable USB hub with a beefy 12V PSU.  it looks like many mainboards can't pull those 2.5GBit/s adapters through with heir power delivery.

Ysurac, 2021-07-03:
I don't have issues with my 2.5Gb/s.  The only problem is to find a VPS that can really deliver more than 1Gb/s ;)
On local test, with a 2.5Gb/s switch this seems to work well.

closed 2021-10-06 by github-actions[bot]