Ysurac/openmptcprouter #1925: info hardware

ascomputer, 2021-06-30:
hello everyone I wanted to know if opentcprouter can be installed on this mini pc Mini PC, ACEPC AK3 Intel Celeron N3350 4GB RAM 64GB thanks

Ysurac, 2021-06-30:
Yes this should work, it's a x86_64 platform and the processor have AES-NI support (and it's a good thing)

ascomputer, 2021-07-03:
i installed x-86_64 version the problem is that it doesn't recognize the lan port. There is a solution?
Thank you

Ysurac, 2021-07-03:
Check network card used and driver that is used, and install it if available.

ascomputer, 2021-07-04:
the lan network card drivers for which operating system should i search?

Ysurac, 2021-07-04:
Linux

Adorfer, 2021-07-05:
i would guess it's [this one](https://www.amazon.co.uk/ACEPC-AK3-Windows-Computer-Ethernet/dp/B08996J3NJ)
can't figure out what NIC is onboard. 
i would suggest to boot some linux live system and use lspci to identify the NIC hardware and find the right kernel module to install on the OpenWRT of the OMR.

ascomputer, 2021-07-05:
the lan card is intel i225v i downloaded the linux driver directly from the intel site.
the file is a .tar.gz how do i install it? you can help me?

Ysurac, 2021-07-05:
Install kmod-igc, available in System->software after refreshing the list.

ascomputer, 2021-07-05:
Running the package manager
Mistakes
Collected errors:
  * pkg_init_from_file: Malformed package file /tmp/upload.ipk.
The opkg install command failed with code 255.

closed 2021-10-08 by github-actions[bot]