Ysurac/openmptcprouter #1928: Load Balancing OMR-ByPass

wlatic, 2021-07-01:
## Expected Behavior

Currently I have Mwan3 running which allows me to group devices together and use sticky connections for them to use specific WANs and failover to over WANs (I have 4), from my understanding OMR-Bypass will allow me to bypass the Multipath VPN, and I can nominate a connection which should be used, however I'm not seeing how I can "Load Balance" these connections in the same way Mwan3 would. I'm also not able to install mwan3 in conjunction with OMR due to the kernel.

Ideally only certain devices will be using the OMR route, the rest will use normal load-balanced WAN groups which have sticky connections.

## Current Behavior
Cannot install Mwan3, cannot see a way to use OMR-bypass to create device groups which will allow use of multiple WANs with sticky connections + failover.

## Specifications

  - OpenMPTCProuter version: v0.58.4r0
  - OpenMPTCProuter platform: X86

Ysurac, 2021-07-01:
I don't know exactly what you want to do. 
Mwan can't be used on OpenMPTCProuter. You can only bypass a domain/ip/... to a single interface.
I don't see any case where bypass to multiples interface can be a good solution.

wlatic, 2021-07-01:
Understood.

The main group this applies to is my TV group. They will not be routed over OMR, and their main connection will be ATT, if the ATT fails than it should failover to one of the Sprint connections.

With Mwan3 I'm able to do the failover easily, but from my understanding is should ATT be offline my TV group would not failover anywhere else.

Ysurac, 2021-07-01:
It's also the case with omr-bypass without setting an interface. This use interface defined as master by default, if down this use another interface.

wlatic, 2021-07-01:
My idea was that I'd keep OMR for just two devices, but then route the rest via local groups which have sticky connections on certain interfaces (Mwan3 style).

If I set ATT as master does this affect the OMR group?

If not I can just have OMR using QOS/SQM and see how that goes.

Adorfer, 2021-07-02:
I woul like to **OMR-Bypass to a second VPS**. . e.g. one with lower performance, but "with a good IP for $THINGS".

Ysurac, 2021-07-02:
@Adorfer It's not related to this issue.

wlatic, 2021-07-02:
@Ysurac got it running with just the bypass and it does seem to be working really well.

Im going to look into mwan3 to see if there are any manual rules that can be taken out of it and also how they setup their groups and do i manually. If I can will provide all the information here so we can see if its possibly to use that to integrate.

storm1er, 2021-09-29:
> @Ysurac said:
> I don't see any case where bypass to multiples interface can be a good solution.

Example: I can only use 4G where I am and can't have a wired connection.
I have multiple 4G cards, multiple routers, and using OpenMTCPRouter for convenience.

Once a 4G connection reaches the data limit, I have to switch the interface in OMR bypass for every line using the interface which is now down. 

I would like this to be handled automatically: If we can't load balance, at least a "fallback interface" feature would be great.

It would us to select a second interface for each OMR Bypass line, so when the first connection goes down, it uses the fallback one.

Ysurac, 2021-09-29:
@storm1er when no interface is set on omr-bypass, then it use interface defined as master and fallback to another when down.

wlatic, 2021-09-29:
@Ysurac the problem is in real life you want to be able to control how things failover. Especially when most people have 1 decent connection and 2 or more connections which vary from usable to junk.

Ysurac, 2021-09-29:
Connection set as master should always be the best connection, so this shouldn't be a big problem :)

wlatic, 2021-09-29:
Apart from if you have a home business and need a stable connection you might not want "junk traffic" (TV, game consoles etc.) using the main connection during work hours.

Its fine if you aren't taking the project that way, but the majority of people using your project are people coming from a 4g environment. Is there anyway we can set donation requirements for features being built to help out?

storm1er, 2021-09-30:
Damned, it's written and translated in the UI...

Maybe you could add the fallback story in the same message ? <3

[image]

----

My case aside, the problem still remains: the actual feature means one connection (the master) takes all, limiting the usage balance to one connection for every bypass line.

Being able to load balance every bypass rules would be awesome for people using 4G connections only =)

closed 2022-01-12 by github-actions[bot]