Ysurac/openmptcprouter #1931: image for mini pc

ssdeol84, 2021-07-03:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

Dear Sir,

How can i install the image on Mini PC Intel celeron J4125?

do your have any image for this?

**Describe why you are asking for this support?**

**Additional context**

Ysurac, 2021-07-03:
It's an Intel 64bits processor, so a x86_64 image should work.
Documentation here: https://github.com/Ysurac/openmptcprouter/wiki/Router-install#for-x86-and-x86_64
So if you can, connect the hard disk to your PC and follow the doc. Else you can use an USB bootable key (or copy OpenMPTCProuter on a key, it's bootable), copy the image on it and follow the doc.

ssdeol84, 2021-07-03:
thanks for reply...one thing more i would like to know..can we use other programs on mini pc along with mptcp router on usb key? bcoz i want to use it as a nginx rtmp streaming server along with mptcp router

Ysurac, 2021-07-03:
You can install what OpenWrt support, but installing nginx is not a good idea. When OpenMPTCProuter is updated all data are removed and only config are set back in.

ssdeol84, 2021-07-03:
is there any way to use both if we have good hardware resources?

Ysurac, 2021-07-03:
By using virtualisation, and using one VM for OpenMPTCProuter.