Ysurac/openmptcprouter #1938: OMR LAN access from VPS LAN not using MPTCP

Rookie805, 2021-07-06:
Hello Ysurac!

Thank you for your awesome project! Keep up and may the donations be with you! ;)

Expected Behavior
Bidirectional access to/from OMR LAN from VPS LAN and some other subnetworks at VPS side utilizing all WANs in OMR. VPS is inside LAN. 

 Current Behavior
I can't access OMR LAN from VPS LAN. And can't access from OMR LAN to neighbour LANS on VPS side (see config below).

The network config is:

VPS side:
Internet (1 IP) -> Router1(192.168.16.1) -> LAN(192.168.16.0/24)

VPS is in LAN(192.168.16.100) with all needed ports forwarded to it from WAN side of router. OMR is working using all LTE interfaces.

The router (192.168.16.1) also has multiple sites (with their LANs) connected to it using VPNs and routing between sites is done using OSPF. 

One site is connected using OMR. 

OMR network config:

WAN (3LTE modems using VLANs)  -> OMR(192.168.100.1) -> Router2(192.168.17.1) -> LAN(192.168.17.0/24)

For now i use L2TP from router behind OMR to router in VPS network to connect between sites. OSPF makes communication between sites very easy and comfortable. But L2TP inside OMR VPN  leads to overhead and MPTCP is not working when I access VPS LAN - it always uses one LTE. 

If i get rid of L2TP I can still access VPS LAN and other sites' LANs from OMR LAN, but not vice versa even if I create route to OMR LAN  on Router1(192.168.16.1): 

route add 192.168.17.0/24 gw 192.168.16.100

Specifications

  - OpenMPTCProuter version: v0.58.4 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter platform: RPI4 or x86_64

To summarize th issue:

Expected: access from VPS LAN to OMR LAN and back using MPTCP on all LTE interfaces on OMR.

Any help and instructions very appreciated. Thank you in advance.

Ysurac, 2021-07-06:
For that you need to add all needed routes on both side and change firewall rule for VPN zone in a more open way.

Rookie805, 2021-07-06:
Thanks for reply, Ysurac!

Could you give more detailed instructions?
1. How to add routes on VPS? What gateway to put in them?

2. Default GW on VPS is the Router in VPS LAN, so I don't need to add routes to all LANS on OMR, right?

3. I am completely confused by firewall section in OMR - could you explain what to change in VPS Zone?
 I mean Firewall -> Zones -> vpn zone -> what? remove masquarading? Or Edit -> General -> allow forward? Or Edit -> Advanced -> ...?

Regards

neoseason, 2021-07-09:
Hi @Rookie805

I have a very similar problem but my topology is a quiet different. My VPS have two interfaces, one act as external  (ens192) and handle all vpn traffic and one internal (ens160) for my vdc infrastructure. My objective was simple, need to use vdc as VPS for branch traffic and have access to the vdc server resources.

I have modified shorewall configuratio (/etc/shotewall)

Have added a new zone called VDC in "zones"

```
###############################################################################
#ZONE TYPE OPTIONS IN OUT
# OPTIONS OPTIONS
fw firewall
net ipv4
vpn ipv4
vdc ipv4
```

Have added interface to this new zone in "interfaces"

```
###############################################################################
#ZONE INTERFACE OPTIONS
net $NET_IFACE dhcp,tcpflags,routefilter,nosmurfs,sourceroute=0
vpn gt-tun+ nosmurfs,tcpflags
vpn gt-udp-tun+ nosmurfs,tcpflags
vpn mlvpn+ nosmurfs,tcpflags
vpn tun+ nosmurfs,tcpflags
vpn wg+ nosmurfs,tcpflags
vpn dsvpn+ nosmurfs,tcpflags
vpn gre-user+ nosmurfs,tcpflags
vpn omr-bonding nosmurfs,tcpflags
vdc ens160 nosmurfs,tcpflags

Modified policy in "policies"
```

```
###############################################################################
#SOURCE DEST POLICY LOG LEVEL LIMIT:BURST

vpn net ACCEPT
vpn fw ACCEPT
fw vpn ACCEPT
fw net ACCEPT
net all DROP
vpn vpn DROP

vpn vdc ACCEPT
vdc vpn ACCEPT
```

Then changed masquereding in "snat" where I have changed 192.168.0.0/16 with only my private class behind OMR.

For me it is working correctly. Don't know if its the right way but seems to. Maybe Ysurac can confirm.