Ysurac/openmptcprouter #1939: Error message: Network interface MAC address duplicated

Rookie805, 2021-07-06:
## Expected Behavior
All is green on status page as was before 0.58 in same config.

## Current Behavior
[image]

## Possible Solution
Nope.

## Steps to Reproduce the Problem

  1. Add 3 WAN interfaces using wizard. WAN interfaces are  added as VLANs (eth0.2, eth0.3, eth0.5)
  2. Sometimes it's working, and sometimes I get error in subject on one or two not master interfaces.
  3. After restarting interfaces with error in Network->Interfaces error is gone.
  4. Error comes back if I remove one of interfaces in wizard or change config somehow.

## Context (Environment)
During all actions in Network->Interfaces all MACs are different. When error - this interface is not used in normal operation.

Sorry, couldn't catch the log at the moment error appearing:

> Tue Jul  6 19:59:50 2021 user.notice omr-bypass: Starting OMR-ByPass...
Tue Jul  6 19:59:51 2021 user.notice omr-bypass: Restart dnsmasq...
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[24617]: exiting on receipt of SIGTERM
Tue Jul  6 19:59:51 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jul  6 19:59:51 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jul  6 19:59:51 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: started, version 2.84 cachesize 150
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: DNS service limited to local subnets
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: DNSSEC validation enabled
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: configured with trust anchor for <root> keytag 20326
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq-dhcp[8057]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using only locally-known addresses for domain test
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using only locally-known addresses for domain onion
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using only locally-known addresses for domain localhost
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using only locally-known addresses for domain local
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using only locally-known addresses for domain invalid
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using only locally-known addresses for domain bind
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using only locally-known addresses for domain use-application-dns.net
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using only locally-known addresses for domain lan
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using nameserver 127.0.0.1#5353
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: using 1 more local addresses
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: read /etc/hosts - 4 addresses
Tue Jul  6 19:59:51 2021 user.notice omr-bypass: OMR-ByPass is running
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq[8057]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Jul  6 19:59:51 2021 daemon.info dnsmasq-dhcp[8057]: read /etc/ethers - 0 addresses
Tue Jul  6 19:59:52 2021 user.notice omr-bypass: Reload dnsmasq...
Tue Jul  6 19:59:52 2021 daemon.info dnsmasq[8057]: read /etc/hosts - 4 addresses
Tue Jul  6 19:59:52 2021 daemon.info dnsmasq[8057]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Jul  6 19:59:52 2021 daemon.info dnsmasq-dhcp[8057]: read /etc/ethers - 0 addresses
Tue Jul  6 19:59:52 2021 user.notice omr-bypass: OMR-ByPass is running
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-local[9590]: using tcp fast open
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-local[9590]: enable TCP no-delay
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-local[9590]: initializing ciphers... aes-256-gcm
Tue Jul  6 19:59:53 2021 user.info : using tcp fast open
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9591]: enable TCP no-delay
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9591]: resolving hostname to IPv6 address first
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9591]: initializing ciphers... aes-256-gcm
Tue Jul  6 19:59:53 2021 user.info : using tcp fast open
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9592]: enable TCP no-delay
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9592]: initializing ciphers... aes-256-gcm
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-local[9590]: listening at 127.0.0.1:1111
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-local[9590]: tcp port reuse enabled
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-local[9590]: udprelay enabled
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9592]: listening at 0.0.0.0:1101
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9591]: listening at :::1100
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9592]: tcp port reuse enabled
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9591]: tcp port reuse enabled
Tue Jul  6 19:59:53 2021 daemon.info /usr/bin/ss-redir[9592]: UDP relay enabled
[... 610 lines cut ...]
Tue Jul  6 20:30:41 2021 user.notice ModemManager: hotplug: event reported: action=add, name=ifb4tun0, subsystem=net
Tue Jul  6 20:30:41 2021 user.notice SQM: piece_of_cake.qos was started on tun0 successfully
Tue Jul  6 20:30:41 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Jul  6 20:30:41 2021 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Tue Jul  6 20:30:42 2021 daemon.err glorytun[10271]: xxx.xxx.xxx.xxx.65001: connected
Tue Jul  6 20:30:42 2021 daemon.info glorytun[10271]: STARTED tun0
Tue Jul  6 20:30:42 2021 daemon.debug dnsmasq[9638]: listening on tun0(#144): 10.255.255.2 port 53
Tue Jul  6 20:30:42 2021 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Tue Jul  6 20:30:46 2021 user.notice omr-bypass: Starting OMR-ByPass...
Tue Jul  6 20:30:47 2021 user.notice omr-bypass: Reload dnsmasq...
Tue Jul  6 20:30:47 2021 daemon.info dnsmasq[9638]: read /etc/hosts - 4 addresses
Tue Jul  6 20:30:47 2021 daemon.info dnsmasq[9638]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Jul  6 20:30:47 2021 daemon.info dnsmasq-dhcp[9638]: read /etc/ethers - 0 addresses
Tue Jul  6 20:30:47 2021 user.notice omr-bypass: OMR-ByPass is running
Tue Jul  6 20:30:51 2021 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
Tue Jul  6 20:31:14 2021 daemon.notice netifd: wan3 (3103): udhcpc: sending renew to 10.177.0.2
Tue Jul  6 20:31:14 2021 daemon.notice netifd: wan3 (3103): udhcpc: lease of 10.50.249.115 obtained, lease time 122

## Specifications

  - OpenMPTCProuter version:v0.58.4 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter platform: x86_64

Regards

Ysurac, 2021-07-07:
I compiled a snapshot image with a fix: https://download.openmptcprouter.com/develop/5.4/
If you can test, I will then backport to stable release.

Rookie805, 2021-07-08:
Sorry, I don't know how to use snapshots - how to test the snapshot on my system. 

I could test, if you give instructions on how to install snapshot on my OMR.

Ysurac, 2021-07-08:
Exactly the same as release images. Download the .img.gz image, backup your config, and upgrade via System->Backup / upgrade firmware menu and flash firmware with keep settings.
To downgrade to release, same way but doesn't keep settings and then import your backup

Rookie805, 2021-07-08:
OK, I upgraded as you said. Everything seems to be fine - I added and removed WAN interfaces a couple of times and haven't got Duplicated MAC error.

Ysurac, 2021-07-08:
ok thanks.
If all is still ok tomorrow, I will backport and create a new stable image.

kinkyninja, 2021-07-08:
Same problem with my Rasp4B & v58.4, so will appreciate the new version 🙏

Ysurac, 2021-07-08:
You can also test snapshot :)

Ysurac, 2021-07-10:
Can you test this futures v0.58.5 ? https://download.openmptcprouter.com/master/5.4/

Rookie805, 2021-07-10:
> 
> 
> Can you test this futures v0.58.5 ? https://download.openmptcprouter.com/master/5.4/

Unfortunately doesn't work.

[image]

System log:
`
Sat Jul 10 11:41:44 2021 daemon.notice netifd: Interface 'lan' is setting up now
Sat Jul 10 11:41:44 2021 daemon.debug dnsmasq[5358]: listening on eth0(#10): 192.168.100.1 port 53
Sat Jul 10 11:41:44 2021 daemon.notice netifd: Interface 'lan' is now up
Sat Jul 10 11:41:44 2021 daemon.notice netifd: Interface 'wan1' is enabled
Sat Jul 10 11:41:44 2021 kern.notice kernel: [   38.465491] r8169 0000:02:00.0 eth0: No native access to PCI extended config space, falling back to CSI
Sat Jul 10 11:41:44 2021 kern.info kernel: [   38.466843] r8169 0000:02:00.0 eth0: Link is Down
Sat Jul 10 11:41:44 2021 daemon.notice netifd: Interface 'wan3' is enabled
Sat Jul 10 11:41:44 2021 user.notice ModemManager: hotplug: add network interface eth0.2: event processed
Sat Jul 10 11:41:44 2021 user.notice ModemManager: hotplug: event reported: action=add, name=eth0.2, subsystem=net
Sat Jul 10 11:41:44 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Jul 10 11:41:44 2021 user.notice ModemManager: hotplug: add network interface eth0.5: event processed
Sat Jul 10 11:41:44 2021 user.notice ModemManager: hotplug: event reported: action=add, name=eth0.5, subsystem=net
Sat Jul 10 11:41:44 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Jul 10 11:41:44 2021 user.notice firewall: Reloading firewall due to ifup of lan (eth0)
Sat Jul 10 11:41:45 2021 daemon.err rmmod: module is not loaded
Sat Jul 10 11:41:45 2021 user.notice Shadowsocks: Rules UP
Sat Jul 10 11:41:45 2021 daemon.info vnstatd[7366]: Info: vnStat daemon 2.6 started. (pid:7366 uid:0 gid:0 64-bit)
Sat Jul 10 11:41:45 2021 daemon.info vnstatd[7366]: Info: Monitoring (1): eth0 (1000 Mbit)
Sat Jul 10 11:41:45 2021 daemon.err procd: failed to open pidfile for writing: : No such file or directory
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/dhcp
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/glorytun
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/glorytun-udp
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/dsvpn
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/mptcpovervpn
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/omr6in4
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/omr-quota
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/omr-tracker
Sat Jul 10 11:41:45 2021 authpriv.info dropbear[7649]: Not backgrounding
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/network reload dependency on /etc/config/openmptcprouter
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/wireless reload dependency on /etc/config/network
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/luci-splash
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/qos
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/miniupnpd
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/firewall reload dependency on /etc/config/sqm
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/dhcp reload dependency on /etc/config/odhcpd
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up non-init /etc/config/fstab reload handler: /sbin/block mount
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/system reload trigger for non-procd /etc/init.d/led
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/luci_statistics
Sat Jul 10 11:41:45 2021 user.notice ucitrack: Setting up /etc/config/system reload dependency on /etc/config/dhcp
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq[5358]: exiting on receipt of SIGTERM
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq[7931]: started, version 2.84 cachesize 150
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq[7931]: DNS service limited to local subnets
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq[7931]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq[7931]: DNSSEC validation enabled
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq[7931]: configured with trust anchor for <root> keytag 20326
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq-dhcp[7931]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq[7931]: using only locally-known addresses for domain test
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq[7931]: using only locally-known addresses for domain onion
Sat Jul 10 11:41:45 2021 daemon.info dnsmasq[7931]: using only locally-known addresses for domain localhost
[... cut ...]
Sat Jul 10 11:43:52 2021 daemon.notice netifd: wan3 (9224): udhcpc: lease of 10.50.249.115 obtained, lease time 122
Sat Jul 10 11:43:52 2021 daemon.notice netifd: wan1 (9227): udhcpc: sending renew to 100.68.156.172
Sat Jul 10 11:43:52 2021 daemon.notice netifd: wan1 (9227): udhcpc: lease of 100.68.156.171 obtained, lease time 122
Sat Jul 10 11:43:58 2021 daemon.err omr-tracker[14487]: Flush terminated
Sat Jul 10 11:44:03 2021 daemon.err glorytun[24947]: the name server returned a failure
`

Reboot doesn't help. Upgrade was done via System ->Backup/Flash firmware

Ysurac, 2021-07-10:
You didn't keep settings ? Because settings from snapshot can't work on 0.58.*.
In any case I saw an error that it also fixed in snapshot but wasn't backported.

Rookie805, 2021-07-10:
I kept settings. Shouldn't I?

What should I do now? Reset settings somehow, or just run wizard again?

Ysurac, 2021-07-10:
You can import a previous backup, or flash again without keeping settings and use wizard.

Rookie805, 2021-07-10:
OK, previous backup imported.

The status is all green. 

I also added and removed one of WAN interfaces, there was no duplicate MAC problem. 

What else should be tested?

Ysurac, 2021-07-10:
it's all ;)

closed 2021-07-10 by Rookie805

Ysurac, 2021-07-10:
I add another little patch for dashboard, compile, and I will release 0.58.5 officially.

Rookie805, 2021-07-10:
Thank you, Ysurac. Have a nice day!