Ysurac/openmptcprouter #1940: Open port TCP UDP

sonnyxp2008, 2021-07-07:
## Expected Behavior
open the tcp and udp port 44158

## Current Behavior
I re-address all the ports on the OMR and Port Forwards firewall I open port 44158 on the affected ip which in this case is my pc. The door remains closed

## Specifications

  - OpenMPTCProuter version: v0.58rc5
  - OpenMPTCProuter VPS version: 5.4.100-mptcp 0.1026
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-07-07:
You followed https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug ?

sonnyxp2008, 2021-07-07:
I'm sorry but I have no results, the door remains closed. this is what omr tells me

root@OpenMPTCProuter:~# tcpdump -i tun0 port 44158
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun0, link-type RAW (Raw IP), capture size 262144 bytes
13:08:40.196086 IP 198.xxx.xxx.xxx.47660 > 10.255.255.2.44158: Flags [S], seq 180                              1328523, win 14600, options [mss 1460,sackOK,TS val 59463399 ecr 0,nop,wscale 8]                              , length 0
13:08:40.196141 IP 10.255.255.2.44158 > 198.xxx.xxx.xxx.47660: Flags [R.], seq 0,                               ack 1801328524, win 0, length 0
13:08:40.403029 IP 198.xxx.xxx.xxx.47662 > 10.255.255.2.44158: Flags [S], seq 391                              7028417, win 14600, options [mss 1460,sackOK,TS val 59463451 ecr 0,nop,wscale 8]                              , length 0
13:08:40.403119 IP 10.255.255.2.44158 > 198.xxx.xxx.xxx.47662: Flags [R.], seq 0,                               ack 3917028418, win 0, length 0
13:08:40.610922 IP 198.xxx.xxx.xxx.47666 > 10.255.255.2.44158: Flags [S], seq 356                              7829753, win 14600, options [mss 1460,sackOK,TS val 59463504 ecr 0,nop,wscale 8]                              , length 0
13:08:40.610965 IP 10.255.255.2.44158 > 198.xxx.xxx.xxx.47666: Flags [R.], seq 0,                               ack 3567829754, win 0, length 0
13:14:40.253493 IP 198.xxx.xxx.xxx.49045 > 10.255.255.2.44158: Flags [S], seq 522484037, win 14600, options [mss 1460,sackOK,TS val 59553414 ecr 0,nop,wscale 8], length 0
13:14:40.253547 IP 10.255.255.2.44158 > 198.xxx.xxx.xxx.49045: Flags [R.], seq 0, ack 522484038, win 0, length 0
13:14:40.511660 IP 198.xxx.xxx.xxx.49049 > 10.255.255.2.44158: Flags [S], seq 2291546252, win 14600, options [mss 1460,sackOK,TS val 59553479 ecr 0,nop,wscale 8], length 0
13:14:40.511711 IP 10.255.255.2.44158 > 198.xxx.xxx.xxx.49049: Flags [R.], seq 0, ack 2291546253, win 0, length 0
13:14:40.766233 IP 198.xxx.xxx.xxx.49051 > 10.255.255.2.44158: Flags [S], seq 2696496405, win 14600, options [mss 1460,sackOK,TS val 59553543 ecr 0,nop,wscale 8], length 0
13:14:40.766397 IP 10.255.255.2.44158 > 198.xxx.xxx.xxx.49051: Flags [R.], seq 0, ack 2696496406, win 0, length 0

Ysurac, 2021-07-07:
At least all is redirected to router.
What is the firewall rule you set ? you can get the info via SSH with ```uci show firewall```. I would also neet the result of ```iptables-save```

sonnyxp2008, 2021-07-07:
root@OpenMPTCProuter:~# uci show firewall
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].input='REJECT'
firewall.@defaults[0].output='REJECT'
firewall.zone_lan=zone
firewall.zone_lan.name='lan'
firewall.zone_lan.input='ACCEPT'
firewall.zone_lan.output='ACCEPT'
firewall.zone_lan.forward='ACCEPT'
firewall.zone_lan.auto_helper='0'
firewall.zone_lan.mtu_fix='1'
firewall.zone_lan.network='lan'
firewall.zone_wan=zone
firewall.zone_wan.name='wan'
firewall.zone_wan.input='REJECT'
firewall.zone_wan.output='ACCEPT'
firewall.zone_wan.forward='REJECT'
firewall.zone_wan.masq='1'
firewall.zone_wan.mtu_fix='1'
firewall.zone_wan.network='wan1' 'wan2' '4g'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
firewall.@rule[1].proto='icmp'
firewall.@rule[1].icmp_type='echo-request'
firewall.@rule[1].family='ipv4'
firewall.@rule[1].target='ACCEPT'
firewall.@rule[2]=rule
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[2].src='wan'
firewall.@rule[2].proto='igmp'
firewall.@rule[2].family='ipv4'
firewall.@rule[2].target='ACCEPT'
firewall.@rule[3]=rule
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[3].src='wan'
firewall.@rule[3].proto='udp'
firewall.@rule[3].src_ip='fc00::/6'
firewall.@rule[3].dest_ip='fc00::/6'
firewall.@rule[3].dest_port='546'
firewall.@rule[3].family='ipv6'
firewall.@rule[3].target='ACCEPT'
firewall.@rule[4]=rule
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[4].src='wan'
firewall.@rule[4].proto='icmp'
firewall.@rule[4].src_ip='fe80::/10'
firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
firewall.@rule[4].family='ipv6'
firewall.@rule[4].target='ACCEPT'
firewall.@rule[5]=rule
firewall.@rule[5].name='Allow-ICMPv6-Forward'
firewall.@rule[5].src='wan'
firewall.@rule[5].dest='*'
firewall.@rule[5].proto='icmp'
firewall.@rule[5].icmp_type='echo-request' 'echo-reply' 'destination-unreachable' 'packet-too-big' 'time-exceeded' 'bad-header' 'unknown-header-type'
firewall.@rule[5].limit='1000/sec'
firewall.@rule[5].family='ipv6'
[... 499 lines cut ...]
-A zone_wan_dest_ACCEPT -o eth1 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o eth2 -m conntrack --ctstate INVALID -m comment --comm             ent "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth2 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_ACCEPT -o eth4 -m conntrack --ctstate INVALID -m comment --comm             ent "!fw3: Prevent NAT leakage" -j DROP
-A zone_wan_dest_ACCEPT -o eth4 -m comment --comment "!fw3" -j ACCEPT
-A zone_wan_dest_REJECT -o eth1 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth2 -m comment --comment "!fw3" -j reject
-A zone_wan_dest_REJECT -o eth4 -m comment --comment "!fw3" -j reject
-A zone_wan_forward -m comment --comment "!fw3: Custom wan forwarding rule chain             " -j forwarding_wan_rule
-A zone_wan_forward -p esp -m comment --comment "!fw3: Allow-IPSec-ESP" -j zone_             lan_dest_ACCEPT
-A zone_wan_forward -p udp -m udp --dport 500 -m comment --comment "!fw3: Allow-             ISAKMP" -j zone_lan_dest_ACCEPT
-A zone_wan_forward -m conntrack --ctstate DNAT -m comment --comment "!fw3: Acce             pt port forwards" -j ACCEPT
-A zone_wan_forward -m comment --comment "!fw3" -j zone_wan_dest_REJECT
-A zone_wan_input -m comment --comment "!fw3: Custom wan input rule chain" -j in             put_wan_rule
-A zone_wan_input -p udp -m udp --dport 68 -m comment --comment "!fw3: Allow-DHC             P-Renew" -j ACCEPT
-A zone_wan_input -p icmp -m icmp --icmp-type 8 -m comment --comment "!fw3: Allo             w-Ping" -j ACCEPT
-A zone_wan_input -p igmp -m comment --comment "!fw3: Allow-IGMP" -j ACCEPT
-A zone_wan_input -m conntrack --ctstate DNAT -m comment --comment "!fw3: Accept              port redirections" -j ACCEPT
-A zone_wan_input -m comment --comment "!fw3" -j zone_wan_src_REJECT
-A zone_wan_output -m comment --comment "!fw3: Custom wan output rule chain" -j              output_wan_rule
-A zone_wan_output -m comment --comment "!fw3" -j zone_wan_dest_ACCEPT
-A zone_wan_src_REJECT -i eth1 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth2 -m comment --comment "!fw3" -j reject
-A zone_wan_src_REJECT -i eth4 -m comment --comment "!fw3" -j reject
COMMIT