Ysurac/openmptcprouter #1944: All work, but speed is slow

dmn61, 2021-07-09:
## Expected Behavior

My system Raspberry Pi 3B+, usb splitter, dc 5v 5a

I have 2 modem 4G, the speed of the modem inserted into the computer 10-15 mbit, in openmptcprouter 2 modems speed 7-12  mbit.

My shadowsocks server 100mbit channel.

[image]

## Current Behavior

My system log
Fri Jul  9 01:29:03 2021 kern.info kernel: [   29.724647] usbcore: registered new interface driver rt2800usb
Fri Jul  9 01:29:03 2021 kern.info kernel: [   29.751622] usbcore: registered new interface driver rtl8192cu
Fri Jul  9 01:29:03 2021 kern.info kernel: [   29.767460] usbcore: registered new interface driver ath9k_htc
Fri Jul  9 01:29:03 2021 user.info kernel: [   29.779372] kmodloader: done loading kernel modules from /etc/modules.d/*
Fri Jul  9 01:29:04 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Fri Jul  9 01:29:04 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Fri Jul  9 01:29:04 2021 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: started, version 2.84 cachesize 150
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: DNS service limited to local subnets
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: DNSSEC validation enabled
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: DNSSEC signature timestamps not checked until receipt of SIGINT
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: configured with trust anchor for <root> keytag 20326
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using only locally-known addresses for domain test
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using only locally-known addresses for domain onion
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using only locally-known addresses for domain localhost
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using only locally-known addresses for domain local
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using only locally-known addresses for domain invalid
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using only locally-known addresses for domain bind
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using only locally-known addresses for domain use-application-dns.net
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using only locally-known addresses for domain lan
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using nameserver 127.0.0.1#5353
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: using 1 more local addresses
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: read /etc/hosts - 4 addresses
Fri Jul  9 01:29:04 2021 daemon.info dnsmasq[3254]: read /tmp/hosts/dhcp.cfg01411c - 0 addresses
Fri Jul  9 01:29:04 2021 authpriv.info dropbear[3316]: Not backgrounding
Fri Jul  9 01:29:04 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Failed to connect to ubus
Fri Jul  9 01:29:04 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'lan'
Fri Jul  9 01:29:04 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan1'
Fri Jul  9 01:29:04 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan2'
Fri Jul  9 01:29:04 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'glorytun'
Fri Jul  9 01:29:04 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Fri Jul  9 01:29:04 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Fri Jul  9 01:29:04 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @rule[9] (Allow-All-LAN-to-VPN) does not specify a protocol, assuming TCP+UDP
Fri Jul  9 01:29:04 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @rule[12] (Allow-Lan-to-Wan) does not specify a protocol, assuming TCP+UDP
Fri Jul  9 01:29:04 2021 kern.info kernel: [   31.412572] usb 1-1.2.1: USB disconnect, device number 5
Fri Jul  9 01:29:04 2021 kern.info kernel: [   31.420012] rndis_host 1-1.2.1:1.0 eth1: unregister 'rndis_host' usb-3f980000.usb-1.2.1, RNDIS device
Fri Jul  9 01:29:05 2021 kern.info kernel: [   31.697913] usb 1-1.2.1: new high-speed USB device number 8 using dwc_otg
Fri Jul  9 01:29:05 2021 user.notice Shadowsocks: Rules UP
Fri Jul  9 01:29:05 2021 kern.warn kernel: [   31.830330] usb 1-1.2.1: config 1 has an invalid interface number: 4 but max is 3
Fri Jul  9 01:29:05 2021 kern.warn kernel: [   31.840551] usb 1-1.2.1: config 1 has no interface number 3
Fri Jul  9 01:29:05 2021 kern.info kernel: [   31.848333] usb 1-1.2.1: New USB device found, idVendor=1286, idProduct=4e31, bcdDevice= 1.00
Fri Jul  9 01:29:05 2021 kern.info kernel: [   31.859361] usb 1-1.2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Fri Jul  9 01:29:05 2021 kern.info kernel: [   31.869284] usb 1-1.2.1: Product: Mobile Composite Device Bus
Fri Jul  9 01:29:05 2021 kern.info kernel: [   31.876404] usb 1-1.2.1: Manufacturer: Marvell
Fri Jul  9 01:29:05 2021 kern.info kernel: [   31.882172] usb 1-1.2.1: SerialNumber: 200806006809080000
Fri Jul  9 01:29:05 2021 kern.debug kernel: [   31.889208] usb 1-1.2.1: no of_node; not parsing pinctrl DT
Fri Jul  9 01:29:05 2021 kern.debug kernel: [   31.889711] rndis_host 1-1.2.1:1.0: no of_node; not parsing pinctrl DT
Fri Jul  9 01:29:05 2021 user.notice Shadowsocks: Reload omr-bypass rules
Fri Jul  9 01:29:05 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Failed to connect to ubus
Fri Jul  9 01:29:05 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_lan' cannot resolve device of network 'lan'
Fri Jul  9 01:29:05 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan1'
Fri Jul  9 01:29:05 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_wan' cannot resolve device of network 'wan2'
Fri Jul  9 01:29:05 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'glorytun'
Fri Jul  9 01:29:05 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omrvpn'
Fri Jul  9 01:29:05 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section 'zone_vpn' cannot resolve device of network 'omr6in4'
Fri Jul  9 01:29:05 2021 daemon.notice procd: /etc/rc.d/S19firewall: Warning: Section @rule[9] (Allow-All-LAN-to-VPN) does not specify a protocol, assuming TCP+UDP
[... 532 lines cut ...]
Fri Jul  9 01:47:49 2021 daemon.info unbound: [10623:0] info:    0.000000    0.000001 8
Fri Jul  9 01:47:49 2021 daemon.info unbound: [10623:0] info:    0.065536    0.131072 1
Fri Jul  9 01:47:49 2021 daemon.info unbound: [10623:0] info:    0.131072    0.262144 14
Fri Jul  9 01:47:49 2021 daemon.info unbound: [10623:0] info:    0.262144    0.524288 3
Fri Jul  9 01:47:49 2021 daemon.info unbound: [10623:0] info:    0.524288    1.000000 2
Fri Jul  9 01:47:49 2021 daemon.info unbound: [10623:0] info:    1.000000    2.000000 5
Fri Jul  9 01:47:49 2021 daemon.info unbound: [10623:0] info:    2.000000    4.000000 1
Fri Jul  9 01:47:49 2021 user.notice post-tracking-post-tracking: Check API configuration...
Fri Jul  9 01:47:49 2021 daemon.notice unbound: [26044:0] notice: init module 0: validator
Fri Jul  9 01:47:49 2021 daemon.notice unbound: [26044:0] notice: init module 1: iterator
Fri Jul  9 01:47:49 2021 daemon.info unbound: [26044:0] info: start of service (unbound 1.13.1).
Fri Jul  9 01:47:50 2021 daemon.info unbound: [26044:0] info: generate keytag query _ta-4f66. NULL IN
Fri Jul  9 01:48:00 2021 user.notice post-tracking-post-tracking: Check API configuration... Done
Fri Jul  9 01:51:23 2021 kern.notice kernel: [  316.347451] EXT4-fs (mmcblk0p2): error count since last fsck: 1
Fri Jul  9 01:51:23 2021 kern.notice kernel: [  316.354682] EXT4-fs (mmcblk0p2): initial error at time 1625794142: __ext4_new_inode:1114
Fri Jul  9 01:51:23 2021 kern.notice kernel: [  316.365281] EXT4-fs (mmcblk0p2): last error at time 1625794142: __ext4_new_inode:1114
Fri Jul  9 02:00:00 2021 cron.err crond[4637]: USER root pid 14566 cmd /etc/init.d/vnstat_backup backup
Fri Jul  9 02:00:00 2021 cron.err crond[4637]: USER root pid 14567 cmd /etc/init.d/omr-bypass bypass_asn
Fri Jul  9 02:08:34 2021 daemon.err /usr/bin/ss-redir[14089]: remote recv: Operation timed out

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Ysurac, 2021-07-09:
Some strange errors in log like the FS error on SD card.
You can test speed using ```omr-test-speed``` for aggregated speed and ```omr-test-speed eth1``` for eth1 speed.
You are using some usb to ethernet adapter ? Not a great idea on RPI3 because USB and ethernet port are shared.

dmn61, 2021-07-09:
1 screenshot omr-test-speed
[image]

2 screenshot omr-test-speed eth1
[image]

My system photo
[image]

How to use a usb modem if not ethernet adapter?

Ysurac, 2021-07-09:
RPI4 would be better for that, USB is not shared with ethernet.
You can also run ```omr-test-speed eth2``` ?
eth1 speed seems to be really low here.

dmn61, 2021-07-09:
[image]

Ysurac, 2021-07-09:
eth2 is also not really fast now. (it's in KB/s, so *8 to get in Kb/s.
So I think there is a problem with modems (location maybe)

closed 2021-10-12 by github-actions[bot]