Ysurac/openmptcprouter #1947: Low TX Power mt7615

SonicFM, 2021-07-10:
## Expected Behavior
Installing the kmods for the mt7615 card and receiving at least 20dbm (Germany) TX power over 5GHz and 2,4GHz.

## Current Behavior
I am currently using the kmods for my mt7615 wifi-card.
Unfortunately I am only getting 6dbm max transmit power from this card.

Searching google has led me to several posts stating that this could be a problem with the eeprom of the card.

The User "Borromini" in my request https://github.com/openwrt/mt76/issues/555 pointed me towards a possible problem with the offset within the eeprom of the card (see his linked forum post).

Could you have a look at it, if this is maybe the issue? If you could tell me where to put some patches, i'm also willed to recompile the image by myself.

## Specifications

  - OpenMPTCProuter version: v0.58.4
  - OpenMPTCProuter VPS version: currently none (not installed)
  - OpenMPTCProuter platform: Bananapi BPI-R2

Ysurac, 2021-07-10:
Can you check if you have mt7615 firmware in ```/lib/firmware/``` or ```/lib/firmware/mediatek``` ?
If not, can you try to add all mt7615* files from https://github.com/openwrt/mt76/tree/master/firmware in ```/lib/firmware/mediatek```, reboot and try ?

SonicFM, 2021-07-10:
The files are there, but i also tried replacing them and rebooted. But nothing changed, still capped at 6dbm.
```
root@OpenMPTCProuter:~# ls -lah /lib/firmware/mediatek/
drwxr-xr-x    2 root     root        3.4K Jul 10 21:22 .
drwxr-xr-x    1 root     root        3.4K Jul 10 21:22 ..
-rw-r--r--    1 root     root      119.8K Mar 18 09:53 mt7615_cr4.bin
-rw-r--r--    1 root     root      447.3K Mar 18 09:53 mt7615_n9.bin
-rw-r--r--    1 root     root       10.8K Mar 18 09:53 mt7615_rom_patch.bin
```
I installed the following packages:
wpad-mesh-openssl
dcwpad (thanks for adding kmod-macremapper ;))
kmod-mt76-connac
kmod-mt7615-common 
kmod-mt7615-firmware
kmod-mt7615e
kmod-mt76-core

Ysurac, 2021-07-11:
Can you test snapshot release : https://download.openmptcprouter.com/develop/5.4/ ?
Do a backup of your config before update, you will need it for downgrade.

This will check if mt76 driver is now fixed, or not.

SonicFM, 2021-07-12:
Which file(s) do you mean exactly?
Reflashing https://download.openmptcprouter.com/develop/5.4/bpi-r2/targets/mediatek/mt7623/openmptcprouter-v0.58.4-r0+17035-1818b038d7-mediatek-mt7623-bpi_bananapi-r2-squashfs-img.gz or installing the kmods/firmware from https://download.openmptcprouter.com/develop/5.4/bpi-r2/targets/mediatek/mt7623/packages/ ?

Ysurac, 2021-07-12:
Reflash with this image and then install driver

SonicFM, 2021-07-12:
Reflashed and installed necessary drivers as wished.
Cards gets detected and loaded without critical errors, as far as i can tell, but sadly the problem persists.

Firmware files are where they are supposed to be (/lib/firmware/mediatek/).
Modprobe say its loaded.
```
root@OpenMPTCProuter:~# modprobe -v mt7615e
mt7615e is already loaded
```
See maybe useful information in the attachment
[logs.zip](https://github.com/Ysurac/openmptcprouter/files/6804172/logs.zip)
.

SonicFM, 2021-07-22:
Hi,
where you able to dig deeper into this topic?
Is there a way i can help you? Sadly im not a dev.

SonicFM, 2021-09-02:
Hello again,
is there any news about this problem?

The not working WiFi card is the only thing missing in my OpenMPTCPRouter+BananaPi R2 setup.

closed 2021-12-07 by github-actions[bot]