Ysurac/openmptcprouter #1948: USB devices are listed, but not connecting automatically to internet on reboot

Bumpalert, 2021-07-10:
## Expected Behavior

 I am using PI 3b and 2 powered USB hub of 4 ports each to connect multiple 4g dongles & all of them are configured in DHCP mode.all 4g dongles are in different gateway address of lan so that no conflict of address in route command. First set up work perfectly  and connect to vps. It is expected to work on power cycle when configured interfaced are enumerated correctly on lsusb.

## Current Behavior
On reboot/power cycle some dongle won't connect to network(not present in route command), but listed in lsusb command. As these devices are powered by usb, they power on on different time and get detected in different order of usb0 or usb1 or eth1 or eth2 and get swapped positions.

In interface page it will say Error: Unknown error (DEVICE_CLAIM_FAILED) on some devices which are on USB bus & if look into the interface edit, we can see a mac id of some other device is trying to get assigned to this device, by mac override issue. in the below example eth1 and eth2 has same mac due to override setting which comes automatically after reboot.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Johnz/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Johnz/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

root@OpenMPTCProuter:~#   ifconfig
--
eth0        Link encap:Ethernet  HWaddr   B8:27:EB:94:F4:07
inet addr:192.168.100.1    Bcast:192.168.100.255    Mask:255.255.255.0
UP BROADCAST RUNNING MULTICAST    MTU:1500  Metric:1
RX packets:20845 errors:0 dropped:0 overruns:0 frame:0
TX packets:28137 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:3870443 (3.6 MiB)  TX   bytes:17372048 (16.5 MiB)
 
eth1        Link encap:Ethernet  HWaddr   FC:DE:56:FF:01:06
inet addr:192.168.8.102    Bcast:192.168.8.255    Mask:255.255.255.0
UP BROADCAST RUNNING MULTICAST    MTU:1500  Metric:1
RX packets:62449 errors:0 dropped:0 overruns:0 frame:0
TX packets:63904 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:100
RX bytes:13765425 (13.1 MiB)  TX   bytes:7347731 (7.0 MiB)
 
eth2        Link encap:Ethernet  HWaddr   FC:DE:56:FF:01:06
inet addr:192.168.1.100    Bcast:192.168.255.255    Mask:255.255.0.0
UP BROADCAST RUNNING MULTICAST    MTU:1500  Metric:1
RX packets:1993 errors:0 dropped:0 overruns:0 frame:0
TX packets:2356 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:100
RX bytes:351766 (343.5 KiB)  TX   bytes:340821 (332.8 KiB)
 
lo          Link encap:Local Loopback
inet addr:127.0.0.1    Mask:255.0.0.0
inet6 addr: ::1/128 Scope:Host
UP LOOPBACK RUNNING    MTU:65536  Metric:1
RX packets:47748 errors:0 dropped:0 overruns:0 frame:0
TX packets:47748 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:9439450 (9.0 MiB)  TX   bytes:9439450 (9.0 MiB)
 
tun0        Link encap:UNSPEC  HWaddr   00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
inet addr:10.255.255.2    P-t-P:10.255.255.1    Mask:255.255.255.255
inet6 addr: fe80::d8d4:4209:4af8:a571/64 Scope:Link
UP POINTOPOINT RUNNING NOARP MULTICAST    MTU:1500  Metric:1
RX packets:10739 errors:0 dropped:0 overruns:0 frame:0
TX packets:10972 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:100
RX bytes:1845918 (1.7 MiB)  TX   bytes:963511 (940.9 KiB)
 
usb0        Link encap:Ethernet  HWaddr   2A:C8:C5:51:44:2B
inet addr:192.168.225.144    Bcast:192.168.255.255    Mask:255.255.0.0
UP BROADCAST RUNNING MULTICAST    MTU:1500  Metric:1
RX packets:48559 errors:0 dropped:0 overruns:0 frame:0
TX packets:57894 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:100
RX bytes:11028202 (10.5 MiB)  TX   bytes:6411176 (6.1 MiB)
 
root@OpenMPTCProuter:~#

</body>

</html>

## Possible Solution
1. Disable mac override and no automatic turn on
2. Error: Unknown error (DEVICE_CLAIM_FAILED) reason as openwrt router had same reason when enabled broadcast flag. My settings doesnt have broadcast flag enabled.

## Steps to Reproduce the Problem

  1. Connect 3 usb to lan devices on usb port on different wan networks.
  2. set up interfaces and reboot
  3. Observe the device order of changed. Mac override came. Some devices are not connecting to internet & some are keeping not responding to 1 sec and coming back to wan connection in status page of server connection

## Context (Environment)

I am trying to make a portable 4g router setup adding off-the-shelf  powered usb hub. Initially I suspected usb power issue, but later I measured and made sure that no usb devices are missing from lsusb.

## Specifications

  - OpenMPTCProuter version: 58.4
  - OpenMPTCProuter VPS version: 1026
  - OpenMPTCProuter platform:RPI3
[image]
[image]
[image]
[image]
[image]

Ysurac, 2021-07-10:
Check in System->OpenMPTCProuter, advanced settings tab, if "Disable interfaces auto rename" is unchecked.

Bumpalert, 2021-07-10:
Thanks for the quick response.

[image]

Device USB0 is present but not connecting in network. But present is lsusb

[image]

root@OpenMPTCProuter:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         10.255.255.1    0.0.0.0         UG    0      0        0 tun0
default         192.168.1.1     0.0.0.0         UG    10     0        0 eth1
default         192.168.8.1     0.0.0.0         UG    13     0        0 eth2
10.255.255.1    *               255.255.255.255 UH    0      0        0 tun0
10.255.255.2    *               255.255.255.255 UH    1200   0        0 tun0
13.235.79.244   192.168.8.1     255.255.255.255 UGH   13     0        0 eth2
127.0.0.0       *               255.0.0.0       U     5      0        0 lo
192.168.0.0     *               255.255.0.0     U     10     0        0 eth1
192.168.8.0     *               255.255.255.0   U     13     0        0 eth2
root@OpenMPTCProuter:~# ifconfig
eth0      Link encap:Ethernet  HWaddr B8:27:EB:94:F4:07
          inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1544 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1926 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:292896 (286.0 KiB)  TX bytes:941646 (919.5 KiB)

eth1      Link encap:Ethernet  HWaddr 0C:5B:8F:27:9A:64
          inet addr:192.168.1.102  Bcast:192.168.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1109 errors:226 dropped:0 overruns:0 frame:0
          TX packets:1144 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:287584 (280.8 KiB)  TX bytes:165642 (161.7 KiB)

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:53:44:58
          inet addr:192.168.8.100  Bcast:192.168.8.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1596 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1866 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:402000 (392.5 KiB)  TX bytes:230655 (225.2 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3119 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3119 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4315209 (4.1 MiB)  TX bytes:4315209 (4.1 MiB)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.255.255.2  P-t-P:10.255.255.1  Mask:255.255.255.255
          inet6 addr: fe80::c9cd:1bc0:7d2d:5b31/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:94 errors:0 dropped:0 overruns:0 frame:0
          TX packets:106 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:12829 (12.5 KiB)  TX bytes:9554 (9.3 KiB)

root@OpenMPTCProuter:~# lsusb
Bus 001 Device 053: ID 2109:2815
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 056: ID 05c6:aaff
Bus 001 Device 051: ID 0424:2514
Bus 001 Device 058: ID 12d1:14db
Bus 001 Device 057: ID 05c6:f00e
Bus 001 Device 050: ID 0424:2514
Bus 001 Device 054: ID 0424:7800
Bus 001 Device 052: ID 2109:2815
root@OpenMPTCProuter:~# date
Sat Jul 10 23:43:28 IST 2021
root@OpenMPTCProuter:~#

05c6:aaff Qualcomm, Inc.  is the USB0 which has Error: Unknown error (DEVICE_CLAIM_FAILED).

All 3 interface usb0,eth1,eth2 has override settings came for mac address.

Auto rename option unchecked as shown below.

[image]

Ysurac, 2021-07-10:
I think that I only save mac for macvlan. So you can remove mac settings on all interfaces.
If not working with auto rename, you can check the box.

Bumpalert, 2021-07-11:
I never use macvlan option.  So it saves mac even when normal option is used. Another issue i found is even perfectly working interface will come as interface error after some time, without changing anything. on restarting interface  it will  show ip for some time and work normal. Even if 2 interfaces has error claiming interface, resetting any of them will fix all interfaces. 3 devices are from 3 vendors and all of them has the device claiming issue.

[image]

1. how can we see the actual mac address of the interface, not the overridden mac address
2. how can we see the interface error in command line and give a restart of interface from command line as a workaround
3. how can we use a command to use default address instead of overriden mac using command line.
4. how to recover from no-data as now only solution is reinstall the firmware. The interfaces are still connected to vps

Another problem is no data is show
[image]

Ysurac, 2021-07-11:
Check Status->System log to find what is detected.
Try to remove set mac address in Network->Interfaces, if this doesn't work remove /etc/init.d.macvlan
ifstatus usb0, and ifup/iddown usb0
no data mean something took too long. Check log, fix issue and it's fixed.
RPI3 always have issues with USB.

Bumpalert, 2021-07-11:
Your last line was very useful.. I moved to PI4B and all problems looks solved. 
Another problem i noticed , when I try to connect to 1025-test version upgraded to 1026 and displayed as 1026 wont connect to both 58.4 & 58.5 version.

Ifstatus is not working as shown below. It was nice to have any interface restart commands handy to use as a work around.

root@OpenMPTCProuter:~# ifstatus eth1
Interface eth1 not found
root@OpenMPTCProuter:~# ifconfig
eth0      Link encap:Ethernet  HWaddr B8:27:EB:94:F4:07
          inet addr:192.168.100.1  Bcast:192.168.100.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:113173 errors:0 dropped:0 overruns:0 frame:0
          TX packets:184942 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:58881978 (56.1 MiB)  TX bytes:193817434 (184.8 MiB)

eth1      Link encap:Ethernet  HWaddr 0C:5B:8F:27:9A:64
          inet addr:192.168.8.102  Bcast:192.168.8.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:77311 errors:0 dropped:0 overruns:0 frame:0
          TX packets:74690 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:62947222 (60.0 MiB)  TX bytes:33178478 (31.6 MiB)

Bumpalert, 2021-07-18:
I got a clue on the "no data issue". Whenever eth0 is not in route web interface will show as no data.

------------------------------------------------------------------------------
   (r0+16336-b36068d35d)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.58.5
 TARGET:      rpi4
 ARCH:        aarch64_cortex-a72

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Sat Jul 10 17:04:09 UTC 2021
-----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenMPTCProuter:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.106 (runner@fv-az180-858) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+16336-b36068d35d)) #0 SMP Thu Mar 25 05:03:24 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45208 r8192 d32616 u86016
[    0.000000] pcpu-alloc: s45208 r8192 d32616 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:35:4B:CF vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 coherent_pool=8M root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rw rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 1836856K/2019328K available (7166K kernel code, 434K rwdata, 1864K rodata, 512K init, 539K bss, 116936K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000156] Console: colour dummy device 80x25
[    0.000387] printk: console [tty1] enabled
[    0.000420] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000442] pid_max: default: 32768 minimum: 301
[    0.000602] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.000654] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001325] Disabling memory control group subsystem
[    0.002027] ASID allocator initialised with 32768 entries
[    0.002117] rcu: Hierarchical SRCU implementation.
[    0.002735] smp: Bringing up secondary CPUs ...
[    0.003122] Detected PIPT I-cache on CPU1
[    0.003173] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003619] Detected PIPT I-cache on CPU2
[    0.003653] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004152] Detected PIPT I-cache on CPU3
[    0.004185] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[... 1600 lines cut ...]
[    9.724903] usb 1-1.3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    9.732395] usb 1-1.3.1: Product: Android
[    9.733061] kmodloader: done loading kernel modules from /etc/modules.d/*
[    9.736406] usb 1-1.3.1: Manufacturer: Android
[    9.736409] usb 1-1.3.1: SerialNumber: 0123456789ABCDEF
[    9.736694] usb 1-1.3.1: no of_node; not parsing pinctrl DT
[    9.755376] uas 1-1.3.1:1.0: no of_node; not parsing pinctrl DT
[    9.755418] usb-storage 1-1.3.1:1.0: no of_node; not parsing pinctrl DT
[    9.755433] usb-storage 1-1.3.1:1.0: USB Mass Storage device detected
[    9.762257] scsi host0: usb-storage 1-1.3.1:1.0
[   10.143592] usb 1-1.3.1: USB disconnect, device number 6
[... cut ...]
[   11.034410] usb 1-1.3.1: no of_node; not parsing pinctrl DT
[   11.041971] cdc_ether 1-1.3.1:1.0: no of_node; not parsing pinctrl DT
[   11.052432] cdc_ether 1-1.3.1:1.0 usb0: register 'cdc_ether' at usb-0000:01:00.0-1.3.1, CDC Ethernet Device, aa:d2:a1:32:e2:d8
[   12.851833] bcmgenet: Skipping UMAC reset
[   12.857367] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[   12.865064] bcmgenet fd580000.ethernet eth0: Link is Up - 10Mbps/Half - flow control off
root@OpenMPTCProuter:~#
root@OpenMPTCProuter:~#

Bumpalert, 2021-07-24:
The no

> 
> 
> I got a clue on the "no data issue". Whenever eth0 is not in route web interface will show as no data.
> ## (r0+16336-b36068d35d)
> 
> PACKAGE: openmptcprouter
> VERSION: v0.58.5
> TARGET: rpi4
> ARCH: aarch64_cortex-a72
> ## BUILD REPO: https://github.com/ysurac/openmptcprouter
> BUILD DATE: Sat Jul 10 17:04:09 UTC 2021
> ## === WARNING! =====================================
> There is no root password defined on this device!
> Use the "passwd" command to set up a new password
> in order to prevent unauthorized SSH logins.
> 
> root@OpenMPTCProuter:~# dmesg [ 0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083] [ 0.000000] Linux version 5.4.106 (runner@fv-az180-858) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+16336-b36068d35d)) #0 SMP Thu Mar 25 05:03:24 2021 [ 0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.1 [ 0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB [ 0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool [ 0.000000] On node 0 totalpages: 504832 [ 0.000000] DMA zone: 3792 pages used for memmap [ 0.000000] DMA zone: 0 pages reserved [ 0.000000] DMA zone: 242688 pages, LIFO batch:63 [ 0.000000] DMA32 zone: 4096 pages used for memmap [ 0.000000] DMA32 zone: 262144 pages, LIFO batch:63 [ 0.000000] percpu: Embedded 21 pages/cpu s45208 r8192 d32616 u86016 [ 0.000000] pcpu-alloc: s45208 r8192 d32616 u86016 alloc=21_4096 [ 0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [ 0.000000] Detected PIPT I-cache on CPU0 [ 0.000000] CPU features: detected: EL2 vector hardening [ 0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware [ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 496944 [ 0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:35:4B:CF vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000 console=ttyAMA0,115200 console=tty1 coherent_pool=8M root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rw rootwait [ 0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear) [ 0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear) [ 0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off [ 0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB) [ 0.000000] Memory: 1836856K/2019328K available (7166K kernel code, 434K rwdata, 1864K rodata, 512K init, 539K bss, 116936K reserved, 65536K cma-reserved) [ 0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1 [ 0.000000] rcu: Hierarchical RCU implementation. [ 0.000000] rcu: CONFIG_RCU_FANOUT set to non-default value of 32. [ 0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies. [ 0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0 [ 0.000000] GIC: Using split EOI/Deactivate mode [ 0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0 [ 0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys). [ 0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns [ 0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns [ 0.000156] Console: colour dummy device 80x25 [ 0.000387] printk: console [tty1] enabled [ 0.000420] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000) [ 0.000442] pid_max: default: 32768 minimum: 301 [ 0.000602] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear) [ 0.000654] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear) [ 0.001325] Disabling memory control group subsystem [ 0.002027] ASID allocator initialised with 32768 entries [ 0.002117] rcu: Hierarchical SRCU implementation. [ 0.002735] smp: Bringing up secondary CPUs ... [ 0.003122] Detected PIPT I-cache on CPU1 [ 0.003173] CPU1: Booted secondary processor 0x0000000001 [0x410fd083] [ 0.003619] Detected PIPT I-cache on CPU2 [ 0.003653] CPU2: Booted secondary processor 0x0000000002 [0x410fd083] [ 0.004152] Detected PIPT I-cache on CPU3 [ 0.004185] CPU3: Booted secondary processor 0x0000000003 [0x410fd083] [ 0.004254] smp: Brought up 1 node, 4 CPUs [ 0.004281] SMP: Total of 4 processors activated. [ 0.004294] CPU features: detected: 32-bit EL0 Support [ 0.004307] CPU features: detected: CRC32 instructions [ 0.004363] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching [ 0.004381] CPU: All CPU(s) started at EL2 [ 0.004410] alternatives: patching kernel code [ 0.005394] devtmpfs: initialized [ 0.012376] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns [ 0.012417] futex hash table entries: 1024 (order: 4, 65536 bytes, linear) [ 0.014181] pinctrl core: initialized pinctrl subsystem [ 0.014472] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT [ 0.015050] NET: Registered protocol family 16 [ 0.017207] DMA: preallocated 8192 KiB pool for atomic allocations [ 0.017759] cpuidle: using governor ladder [ 0.017788] cpuidle: using governor menu [ 0.018099] Serial: AMBA PL011 UART driver [ 0.019552] bcm2835-mbox fe00b880.mailbox: mailbox enabled [ 0.028039] raspberrypi-firmware soc:firmware: Attached to firmware from 2021-02-16 14:17, variant start [ 0.032046] raspberrypi-firmware soc:firmware: Firmware hash is d6d82cf99 [... cut ...] data will go and show the devices are not present. Even if you plug the usb 4g dongles one by one, even if single 4g dongle connected it will show nodata. Then I removed all interfaces and tried re adding interface. again nodata came. Only thing worked is restored a blank backup(changes SSH key)  & then added one by one then no data disappeared. I had even tried a fresh installation and restored a last working backup(which will restore ssh signature  of "nodata" setup also) which failed. @Ysurac