Ysurac/openmptcprouter #1950: Supermicro Atom board

gitxmax, 2021-07-12:
Hello,
I'm looking at low TDP boards out there and came across Supermicro A1SAM-2750F server board, which has Atom  C2550 8core-8 threads, Quad GbE LAN ports. It's for sale locally, for around $50 (minus Ram), so really tempting.
Just want to check, if there can be ay potential problems running the openmptcprouter on it, since the hardware is rather "exotic".
Thank you

Ysurac, 2021-07-12:
Should work and CPU have AES-NI support that will help.
LAN ports should be supported by igb kernel module that is installed by default on OpenMPTCProuter.

gitxmax, 2021-07-12:
Great then. I will certainly have more questions, when I receive it.
Thanks again

closed 2021-10-15 by github-actions[bot]