Ysurac/openmptcprouter #1951: How to configure a client to be on same Vpn as OMR and VPS?

highfreq, 2021-07-12:
## Expected Behavior
I need to install a client on a machine to reach OMR trough VPN (i can use any needed VPN on the OMR+VPS system)
Basicly i need to talk to a client that is in the OMR lan (i know how to set port forwarding on the router)
## Current Behavior
Don't know how to connect my pc on same VPN and comunicate with OMR. Maybe i can just use same keys and certificates as OMR and be on same VPN?

highfreq, 2021-07-12:
I need the appliance that has OMR to use mptcp to VPS so that link is fast, i just need another server or pc to get in the same vpn as OMR so that i can enter OMR lan, i need the OMR link to the VPS to stay.
thank you for your help.

milankragujevic, 2021-07-12:
In that case I recommend you install PPTP VPN server on the VPS server with OMR VPNs already and then port forward on OMR device to your appliance, but only where source IP is in PPTP VPN LAN. And then masquerade PPTP VPN clients to the internal IP which is allowed to access OMR, and then access OMR with it's 10.255.255.1 or 10.255.254.1 or whatever IP, it will be visible to you from PPTP. 

This isn't exactly what you want but is similar.

You can't really connect random devices to existing OMR VPNs on the VPS.

highfreq, 2021-07-12:
What if i use port forwarding?
From a public server i need to to pull an RTSP stream from a device on the OMR lan and another tcp stream from the same machine. Maybe with oport forwarding and some nat rules i can dio that?

milankragujevic, 2021-07-12:
Yes, of course, you can do it that way. You can also add firewall rules to allow certain IPs only to access the stream if you want additional security.

highfreq, 2021-07-12:
Ok that would be the way i'l test, thank you very very muche. Also could you answer to the PS i edited in the previous post about the VPS with multiple OMR.