Ysurac/openmptcprouter #1953: How to use Huawei E5785 4G+ MiFi with OMR

milankragujevic, 2021-07-13:
## Expected Behavior
E5785 is visible as cdc_ether and allows internet access using it's own gateway IP 192.168.8.1

## Current Behavior
Doesn't work at all.

It's visible as /dev/cdc-wdm0 but lsusb reports "Bus 001 Device 004: ID 12d1:1506" (which is "CD-ROM            HUAWEI   Mass Storage     2.31 PQ: 0 ANSI: 2")

Seems usb-modeswitch is not working for it.

usb-mode -l and usb-mode -s return nothing.

## Specifications

  - OpenMPTCProuter version: v0.58.5
  - OpenMPTCProuter VPS version: v0.1026 5.4.100-mptcp
  - OpenMPTCProuter platform: RPI4

## syslog
```
Tue Jul 13 09:04:33 2021 daemon.notice netifd: wan3 (12918):
Tue Jul 13 09:04:33 2021 daemon.notice netifd: wan3 (12918): Failed to get modem information
Tue Jul 13 09:04:33 2021 daemon.notice netifd: Interface 'wan3' is now down
Tue Jul 13 09:04:33 2021 daemon.notice netifd: Interface 'wan3' is setting up now
Tue Jul 13 09:04:33 2021 daemon.notice netifd: wan3 (12936): comgt 09:04:33 -> -- Error Report --
Tue Jul 13 09:04:33 2021 daemon.notice netifd: wan3 (12936): comgt 09:04:33 -> ---->                  ^
Tue Jul 13 09:04:33 2021 daemon.notice netifd: wan3 (12936): comgt 09:04:33 -> Error @118, line 9, Could not write to COM device. (1)
Tue Jul 13 09:04:33 2021 daemon.notice netifd: wan3 (12936):
Tue Jul 13 09:04:33 2021 daemon.notice netifd: wan3 (12936): Failed to get modem information
Tue Jul 13 09:04:33 2021 daemon.notice netifd: wan3 (13244): Stopping network wan3
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13244): comgt 09:04:34 -> -- Error Report --
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13244): comgt 09:04:34 -> ---->                  ^
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13244): comgt 09:04:34 -> Error @118, line 9, Could not write to COM device. (1)
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13244):
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13244): Failed to get modem information
Tue Jul 13 09:04:34 2021 daemon.notice netifd: Interface 'wan3' is now down
Tue Jul 13 09:04:34 2021 daemon.notice netifd: Interface 'wan3' is setting up now
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13256): comgt 09:04:34 -> -- Error Report --
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13256): comgt 09:04:34 -> ---->                  ^
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13256): comgt 09:04:34 -> Error @118, line 9, Could not write to COM device. (1)
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13256):
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13256): Failed to get modem information
Tue Jul 13 09:04:34 2021 daemon.notice netifd: wan3 (13409): Stopping network wan3
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13409): comgt 09:04:35 -> -- Error Report --
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13409): comgt 09:04:35 -> ---->                  ^
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13409): comgt 09:04:35 -> Error @118, line 9, Could not write to COM device. (1)
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13409):
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13409): Failed to get modem information
Tue Jul 13 09:04:35 2021 daemon.notice netifd: Interface 'wan3' is now down
Tue Jul 13 09:04:35 2021 daemon.notice netifd: Interface 'wan3' is setting up now
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13432): comgt 09:04:35 -> -- Error Report --
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13432): comgt 09:04:35 -> ---->                  ^
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13432): comgt 09:04:35 -> Error @118, line 9, Could not write to COM device. (1)
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13432):
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13432): Failed to get modem information
Tue Jul 13 09:04:35 2021 daemon.notice netifd: wan3 (13834): Stopping network wan3
Tue Jul 13 09:04:36 2021 daemon.notice netifd: wan3 (13834): comgt 09:04:36 -> -- Error Report --
Tue Jul 13 09:04:36 2021 daemon.notice netifd: wan3 (13834): comgt 09:04:36 -> ---->                  ^
Tue Jul 13 09:04:36 2021 daemon.notice netifd: wan3 (13834): comgt 09:04:36 -> Error @118, line 9, Could not write to COM device. (1)
Tue Jul 13 09:04:36 2021 daemon.notice netifd: wan3 (13834):
Tue Jul 13 09:04:36 2021 daemon.notice netifd: wan3 (13834): Failed to get modem information
Tue Jul 13 09:04:36 2021 daemon.notice netifd: Interface 'wan3' is now down
Tue Jul 13 09:04:36 2021 daemon.notice netifd: Interface 'wan3' is setting up now
Tue Jul 13 09:04:37 2021 daemon.notice netifd: wan3 (14168): comgt 09:04:37 -> -- Error Report --
Tue Jul 13 09:04:37 2021 daemon.notice netifd: wan3 (14168): comgt 09:04:37 -> ---->                  ^
Tue Jul 13 09:04:37 2021 daemon.notice netifd: wan3 (14168): comgt 09:04:37 -> Error @118, line 9, Could not write to COM device. (1)
Tue Jul 13 09:04:37 2021 daemon.notice netifd: wan3 (14168):
Tue Jul 13 09:04:37 2021 daemon.notice netifd: wan3 (14168): Failed to get modem information
Tue Jul 13 09:04:37 2021 daemon.notice netifd: wan3 (14570): Stopping network wan3
Tue Jul 13 09:04:37 2021 daemon.notice netifd: wan3 (14570): comgt 09:04:37 -> -- Error Report --
[... 1514 lines cut ...]
[39961.497338] scsi host0: usb-storage 1-1.4:1.0
[39961.786778] usb 1-1.4: USB disconnect, device number 3
[39962.157712] usb 1-1.4: new high-speed USB device number 4 using xhci_hcd
[39962.262636] usb 1-1.4: New USB device found, idVendor=12d1, idProduct=1506, bcdDevice= 1.02
[39962.271032] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[39962.278383] usb 1-1.4: Product: HUAWEI Mobile
[39962.282750] usb 1-1.4: Manufacturer: HUAWEI Mobile
[39962.287784] usb 1-1.4: no of_node; not parsing pinctrl DT
[39962.291146] option 1-1.4:1.0: no of_node; not parsing pinctrl DT
[39962.291177] option 1-1.4:1.0: GSM modem (1-port) converter detected
[39962.297654] option1 ttyUSB0: no of_node; not parsing pinctrl DT
[39962.297862] usb 1-1.4: GSM modem (1-port) converter now attached to ttyUSB0
[39962.305280] huawei_cdc_ncm 1-1.4:1.1: no of_node; not parsing pinctrl DT
[39962.332791] huawei_cdc_ncm 1-1.4:1.1: MAC-Address: 00:1e:10:1f:00:00
[39962.339182] huawei_cdc_ncm 1-1.4:1.1: setting rx_max = 16384
[39962.345558] huawei_cdc_ncm 1-1.4:1.1: NDP will be placed at end of frame for this device.
[39962.354274] huawei_cdc_ncm 1-1.4:1.1: cdc-wdm0: USB WDM device
[39962.360627] huawei_cdc_ncm 1-1.4:1.1 wwan0: register 'huawei_cdc_ncm' at usb-0000:01:00.0-1.4, Huawei CDC NCM device, 00:1e:10:1f:00:00
[39962.373707] uas 1-1.4:1.2: no of_node; not parsing pinctrl DT
[39962.373753] usb-storage 1-1.4:1.2: no of_node; not parsing pinctrl DT
[39962.373768] usb-storage 1-1.4:1.2: USB Mass Storage device detected
[39962.380477] scsi host0: usb-storage 1-1.4:1.2
[39963.386483] scsi 0:0:0:0: CD-ROM            HUAWEI   Mass Storage     2.31 PQ: 0 ANSI: 2
[39963.394945] sd 0:0:0:0: no of_node; not parsing pinctrl DT
````

Ysurac, 2021-07-13:
Device ID seems to be correct for this device.
This device is powered or use only USB ? If only USB do you use a powered hub ?

milankragujevic, 2021-07-13:
@Ysurac USB power only, not powered Hub, however the device has a battery so it's not starved of power. It works on Raspbian as cdc_ether with no problems on same RPI 4.

How should I configure the WAN, ModemManager, NCM, QMI? I'm not sure, it's not explained...

Ysurac, 2021-07-13:
It's not explained because I don't use this.
You can look at OpenWrt doc for more info.

milankragujevic, 2021-07-14:
@Ysurac I have looked at OpenWrt docs but the instructions don't work properly (doesn't get IP on interface). I'll leave the ticket open so someone else can jump in with info as I've seen a lot of people here run a 4G Huawei MiFi on OMR.

milankragujevic, 2021-08-15:
Confirmed this device is not supported at all in the OpenWrt ecosystem. Closing ticket.

closed 2021-08-15 by milankragujevic

luisimasd, 2024-04-24:
Hello, the status remains the same today? I mean, is still not supported?

gemolnar79, 2024-10-04:
Hi,
You can try using the modem over RNDIS on USB. Interface will be eth1/usb0 and can be configured in OpenWRT as you wish. (Just like an LTE stick it will provide IP over DHCP. I tried it in the past ~2 years, can confirm that works.)
RNDIS was not available in the original firmware of the device I had. I changed it to a custom one and was able to enable it.
For further details on this please see https://4pda.to/forum/index.php?showtopic=907081&st=200#entry80173699