Ysurac/openmptcprouter #1954: Traffic only goes through master interface.

jchuang1977, 2021-07-13:
## Expected Behavior
Traffic will go out on both interfaces simulatiously

## Current Behavior
But traffic is only going out over the interface which is selected as master

## Specifications

  - OpenMPTCProuter version: 58.5
  - OpenMPTCProuter VPS version:  install on around June 22
  - OpenMPTCProuter platform:  custom hardware platform (MTK)

#### debug information
I use speedtest.net to test ,  but the throughput only go through master interface.

**### WAN MPTCP test result**

1: * 
2: 10.11.225.2 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
3: 10.11.35.158  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
4: 10.9.3.153  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
5: 10.9.3.145  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
6: 10.9.166.137 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
7: 10.9.166.90  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
8: 202.144.222.114  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
9: 63.222.54.113 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
10: 63.218.250.173 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
11: 63.216.242.46  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
12: 10.71.0.246  [PARTIAL] +PartialTCP IP::TotalLength IP::TTL IP::CheckSum 
13: 10.71.3.218 TCP::DataOffset TCP::CheckSum IP::TotalLength IP::TTL IP::CheckSum +TCPOptionMaxSegSize 
14: * 
15: 45.32.24.230 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< SLL (16 bytes) :: PackeType = 0 , AddressType = 520 , AddressLength = 0 , SourceAddress = 00:00:00:00:00:00 , Pad = 0000 , Protocol = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 48 , Protocol = 0x6 , CheckSum = 0x9713 , SourceIP = 45.32.24.230 , DestinationIP = 100.76.9.91 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 62311 , SeqNumber = 1457710133 , AckNumber = 46062758 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43200 , CheckSum = 0xb238 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1394 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 3450330166240643436 , >

**### WAN1  MPTCP test result**

1: * 
2: 10.157.65.5  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
3: 10.157.67.7  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
4: 10.157.67.18  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
5: 210.65.126.214  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
6: 220.128.9.190  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
7: 220.128.9.34  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
8: 220.128.2.109  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
9: 220.128.6.233  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
10: * 
11: 120.88.54.98  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
12: * 
13: * 
14: * 
15: 45.32.24.230 TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = 7a:82:8e:82:49:71 , SourceMAC = 06:16:26:36:46:56 , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 47 , Protocol = 0x6 , CheckSum = 0xa297 , SourceIP = 45.32.24.230 , DestinationIP = 10.116.88.175 , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 46140 , SeqNumber = 3517342091 , AckNumber = 378519072 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43200 , CheckSum = 0x287 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1460 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = 7774812388133063 , >

[https://img.onl/emHHiI](url)
gateway is because my 4G provider can't support ping gateway.

**### uci show | grep openmptcprouter**
root@OpenWrt:~# uci show | grep openmptcprouter
[... 80 lines cut ...]
openmptcprouter.loopback.metric='6'
openmptcprouter.lan=interface
openmptcprouter.lan.multipath='off'
openmptcprouter.lan.metric='7'
openmptcprouter.lan.multipathvpn='0'
openmptcprouter.wan6=interface
openmptcprouter.wan6.multipath='off'
openmptcprouter.wan6.metric='8'
openmptcprouter.wan6.multipathvpn='0'
openmptcprouter.omrvpn=interface
openmptcprouter.omrvpn.multipath='off'
openmptcprouter.omrvpn.metric='1200'
openmptcprouter.omr6in4=interface
openmptcprouter.omr6in4.multipath='off'
openmptcprouter.omr6in4.metric='1201'
openmptcprouter.latest_versions=latest_versions
openmptcprouter.latest_versions.omr='0.58.5'
openmptcprouter.latest_versions.vps='0.1026'
openmptcprouter.latest_versions.lc='1626169131'
sysupgrade.server.url='https://download.openmptcprouter.com'
ucitrack.@network[0].affects='dhcp' 'radvd' 'glorytun' 'glorytun-udp' 'dsvpn' 'm                                                      ptcpovervpn' 'omr6in4' 'omr-quota' 'omr-tracker' 'openmptcprouter'
ucitrack.@glorytun[0].affects='glorytun-udp' 'openmptcprouter-vps'
ucitrack.@glorytun-udp[1].affects='openmptcprouter-vps'
ucitrack.@openmptcprouter[0]=openmptcprouter
ucitrack.@openmptcprouter[0].init='openmptcprouter'

Ysurac, 2021-07-13:
All keys are visible, it's not a great idea...
VPS is down, check in State->System log about glorytun/omrvpn.
VPS was rebooted after install ?
Use ```omr-test-speed``` to check aggregated speed, and ```omr-test-speed eth1``` for speed on eth1 real interface (check real interface with ```ip a```), do that for all wan interfaces at least 2 minutes.

jchuang1977, 2021-07-13:
Thanks for the quick reply.
I use omr-test-speed  to test "eth0"(wan1 ,master)  , "ccmni0" ( wan , it's mtk 5G/LTE interface)  and  aggregation . 
But just only  traffic in ccmni0(wan) test.
No traffic  in aggregation test.
I also reinstall the VPS and reboot.

[image]

root@OpenWrt:~# omr-test-speed eth0
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 470
host: speedtest.tokyo2.linode.com - ping: 52
host: speedtest.singapore.linode.com - ping: 67
host: speedtest.newark.linode.com - ping: 494
host: speedtest.atlanta.linode.com - ping: 311
host: speedtest.dallas.linode.com - ping: 312
host: speedtest.fremont.linode.com - ping: 311
host: speedtest.tele2.net - ping: 309
host: proof.ovh.net - ping: 303
host: speed.hetzner.de - ping: 294
host: ipv4.bouygues.testdebit.info - ping: 290
host: par.download.datapacket.com - ping: 303
host: nyc.download.datapacket.com - ping: 215
host: ams.download.datapacket.com - ping: 297
host: fra.download.datapacket.com - ping: 294
host: lon.download.datapacket.com - ping: 299
host: mad.download.datapacket.com - ping: 297
host: prg.download.datapacket.com - ping: 317
host: sto.download.datapacket.com - ping: 303
host: vie.download.datapacket.com - ping: 322
host: war.download.datapacket.com - ping: 329
host: atl.download.datapacket.com - ping: 317
host: chi.download.datapacket.com - ping: 340
host: lax.download.datapacket.com - ping: 172
host: mia.download.datapacket.com - ping: 385
host: nyc.download.datapacket.com - ping: 319
Best server is http://speedtest.tokyo2.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  309M    0  309M    0     0  2416k      0 --:--:--  0:02:11 --:--:-- 2732k

root@OpenWrt:~# omr-test-speed ccmni0
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 276
host: speedtest.tokyo2.linode.com - ping: 107
host: speedtest.singapore.linode.com - ping: 125
host: speedtest.newark.linode.com - ping: 259
host: speedtest.atlanta.linode.com - ping: 234
host: speedtest.dallas.linode.com - ping: 225
host: speedtest.fremont.linode.com - ping: 234
host: speedtest.tele2.net - ping: 352
host: proof.ovh.net - ping: 381
host: speed.hetzner.de - ping: 334
host: ipv4.bouygues.testdebit.info - ping: 297
host: par.download.datapacket.com - ping: 355
host: nyc.download.datapacket.com - ping: 315
host: ams.download.datapacket.com - ping: 240
host: fra.download.datapacket.com - ping: 485
host: lon.download.datapacket.com - ping: 321
host: mad.download.datapacket.com - ping: 284
host: prg.download.datapacket.com - ping: 294
host: sto.download.datapacket.com - ping: 320
host: vie.download.datapacket.com - ping: 271
host: war.download.datapacket.com - ping: 536
host: atl.download.datapacket.com - ping: 246
host: chi.download.datapacket.com - ping: 311
host: lax.download.datapacket.com - ping: 192
host: mia.download.datapacket.com - ping: 300
host: nyc.download.datapacket.com - ping: 307
Best server is http://speedtest.tokyo2.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024M    0 1024M    0     0  9840k      0 --:--:--  0:01:46 --:--:-- 8448k

root@OpenWrt:~# omr-test-speed
Select best test server...
host: speedtest.frankfurt.linode.com - ping: 313
host: speedtest.tokyo2.linode.com - ping: 79
host: speedtest.singapore.linode.com - ping: 81
host: speedtest.newark.linode.com - ping: 214
host: speedtest.atlanta.linode.com - ping: 213
host: speedtest.dallas.linode.com - ping: 194
host: speedtest.fremont.linode.com - ping: 146
host: speedtest.tele2.net - ping: 317
host: proof.ovh.net - ping: 459
host: speed.hetzner.de - ping: 275
host: ipv4.bouygues.testdebit.info - ping: 295
host: par.download.datapacket.com - ping: 319
host: nyc.download.datapacket.com - ping: 318
host: ams.download.datapacket.com - ping: 436
host: fra.download.datapacket.com - ping: 309
host: lon.download.datapacket.com - ping: 283
host: mad.download.datapacket.com - ping: 295
host: prg.download.datapacket.com - ping: 306
host: sto.download.datapacket.com - ping: 312
host: vie.download.datapacket.com - ping: 319
host: war.download.datapacket.com - ping: 327
host: atl.download.datapacket.com - ping: 250
host: chi.download.datapacket.com - ping: 203
host: lax.download.datapacket.com - ping: 170
host: mia.download.datapacket.com - ping: 246
host: nyc.download.datapacket.com - ping: 312
Best server is http://speedtest.tokyo2.linode.com/garbage.php?ckSize=10000, running test:
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  142M    0  142M    0     0  2819k      0 --:--:--  0:00:51 --:--:-- 2674k

Ysurac, 2021-07-13:
I see that proxy is disabled. And wan seems to be detected as down, did you check State->System log ?
I also see a high load average on router side.
It's a custom image ?

jchuang1977, 2021-07-13:
Hi Ysurac,
Thanks for you time.
Yes, it's my custom image.
I ported it on MTK solution base kernel 4.19 + openwrt 19 .
I use another router to verify my VPS, the VPS is good.
So the problem should be my custom image.
I will check the system log tomorrow.
Is there any suspect area should I check at first ?

Thanks.
Andy

Ysurac, 2021-07-13:
Kernel 4.19 is not supported, no idea if this can work with kernel on VPS.
Same for OpenWrt 19, I may use functions not available on this.

jchuang1977, 2021-07-13:
Hi Ysurac,
May I know why kernel 4.19 is not supported ?  Which driver/module  can not meet the OMR's requirement  ?
Because I found the MPTCP patch in 
https://github.com/Ysurac/openmptcprouter/tree/develop/root/target/linux/generic/hack-4.19
So I thought OMR also can support 4.19.

Best,
Andy

Ysurac, 2021-07-13:
I didn't updated 4.19 patches since a long time and some are missing. I only support 5.4 kernel.

jchuang1977, 2021-07-14:
Hi Ysurac,
Sorry to bother you again.
Still need your professional comment.
I change my 2 wan interfaces into stable ethernet  in my custom image.
[image]

All status looks great ( all green ),  but the omr-test-speed traffic still only goes  through the master interface.

Best,
Andy

jchuang1977, 2021-07-16:
Hi ,
I rebuild the latest OMR git  on xiaomi 3G.
But I still have the same problem which traffic only go through the master interface.
What configuration should I notice ?

[image]

[image]

Thanks for your help.

Best,
Andy

Ysurac, 2021-07-16:
You have a really high load average.
Check with ```htop``` via SSH what is using so much CPU.

jchuang1977, 2021-07-16:
Hi Ysurac,
Thanks for the reply.
Because I choose the openmptcprouter-mini option in this build,  I found the luci-proto-bonding package is not selected.
Is the luci-proto-bonding essential and must  to the mptcp  function ?
Is it the reason I can't aggregate the 2 WAN traffic ?

Best,
Andy

Ysurac, 2021-07-16:
No, luci-proto-bonding is only used for OpenVPN bonding, not related to MPTCP.

jchuang1977, 2021-07-21:
Hi Ysurac,
   Sorry to bother you again, still need your professional comment here.

   I still try to find the root cause to the traffic only goes through only one interface.
I compare several pages with the RPI4 and my custom fw, I found the "established connections" is very different.

RPI4
[image]

my custom fw
[image]

When I run the omr-test speed,  two different situations happen.
RPI4 will connect to my VPS(45.32.24.230==E618202D)  65101 and 65001 port with 4~5 connections.
But my custom fw will only connect to my wan1 ip (192.168.8.100==6408A8C0) , and port is not 65xxx port.
  Do you have any comment about this area ?  Thanks you very much again.

best,
Andy