Ysurac/openmptcprouter #1955: Second WAN aren't reachable by openmptcprouter

xillibit, 2021-07-14:
## Expected Behavior

The second wan should be reachable by openmptcprouter

## Current Behavior
I have done a clean install of openmptcprouter 0.58.5, the second isn't reachable by openmptcprouter. If i connet directly my PC on the huawei router, i can access the web interface on his IP: 192.168.8.1

[image]

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-07-14:
wan2 is a macvlan interface ?
192.168.8.1 can be pinged or blocked by huawei modem ?

xillibit, 2021-07-14:
It was macVLAN by default for wan2 and other wan, i don't changed it

I don't know why huawei B715-23c should block is own ip 192.168.8.1, i will try to connect a pc on huawei to check on the web interface

xillibit, 2021-07-24:
I had disconnected the huawei router and reconnected now, it's works : 

[image]

closed 2021-07-24 by xillibit