Ysurac/openmptcprouter #1956: Error: Device can not be enslaved while up

flimo44, 2021-07-14:
## Expected Behavior
I have a OVH VPS. I just do a fresh install. ( it's a new VPS debian 10 server )
when i connect on console i have many errors in continu on display. ( not in rsh )

## Current Behavior

Jul 14 17:47:18 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:18 vps-ba1bab01 kernel: [69955.800983] omr-bonding: (slave ovpnbonding8): slave is up - this may be due to an out of date ifenslave
Jul 14 17:47:18 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:18 vps-ba1bab01 omr-service[821]: RTNETLINK answers: File exists
Jul 14 17:47:18 vps-ba1bab01 omr-service[821]: RTNETLINK answers: File exists
Jul 14 17:47:18 vps-ba1bab01 omr-service[821]: RTNETLINK answers: File exists
Jul 14 17:47:21 vps-ba1bab01 v2ray[636]: 2021/07/14 17:47:21 127.0.0.1:39873 accepted tcp:127.0.0.1:0 [api]
Jul 14 17:47:21 vps-ba1bab01 v2ray[636]: 2021/07/14 17:47:21 127.0.0.1:39875 accepted tcp:127.0.0.1:0 [api]
Jul 14 17:47:30 vps-ba1bab01 v2ray[636]: 2021/07/14 17:47:30 127.0.0.1:39887 accepted tcp:127.0.0.1:0 [api]
Jul 14 17:47:30 vps-ba1bab01 v2ray[636]: 2021/07/14 17:47:30 127.0.0.1:39889 accepted tcp:127.0.0.1:0 [api]
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: jq: error (at <stdin>:1): Cannot iterate over null (null)
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:32 vps-ba1bab01 kernel: [69969.995214] omr-bonding: (slave ovpnbonding1): slave is up - this may be due to an out of date ifenslave
Jul 14 17:47:32 vps-ba1bab01 kernel: [69970.000805] omr-bonding: (slave ovpnbonding2): slave is up - this may be due to an out of date ifenslave
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:32 vps-ba1bab01 kernel: [69970.005194] omr-bonding: (slave ovpnbonding3): slave is up - this may be due to an out of date ifenslave
Jul 14 17:47:32 vps-ba1bab01 kernel: [69970.009267] omr-bonding: (slave ovpnbonding4): slave is up - this may be due to an out of date ifenslave
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:32 vps-ba1bab01 kernel: [69970.013584] omr-bonding: (slave ovpnbonding5): slave is up - this may be due to an out of date ifenslave
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:32 vps-ba1bab01 kernel: [69970.017811] omr-bonding: (slave ovpnbonding6): slave is up - this may be due to an out of date ifenslave
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:32 vps-ba1bab01 kernel: [69970.021960] omr-bonding: (slave ovpnbonding7): slave is up - this may be due to an out of date ifenslave
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:32 vps-ba1bab01 kernel: [69970.026100] omr-bonding: (slave ovpnbonding8): slave is up - this may be due to an out of date ifenslave
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: Error: Device can not be enslaved while up.
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: RTNETLINK answers: File exists
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: RTNETLINK answers: File exists
Jul 14 17:47:32 vps-ba1bab01 omr-service[821]: RTNETLINK answers: File exists
root@vps-ba1bab01:/var/log#

## Possible Solution

## Steps to Reproduce the Problem

  1. all time
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  0.58.5
  - OpenMPTCProuter VPS version: 0.1226 5.4
  - OpenMPTCProuter platform:  VM x86_64

Ysurac, 2021-07-14:
It's in log, not really a big issue but I will try to remove these warnings

flimo44, 2021-07-14:
Hello @Ysurac ,
Can you do also something for this error on the Router log ? ( merci )

Wed Jul 14 18:20:09 2021 daemon.info dnsmasq-dhcp[31992]: DHCPOFFER(eth0) 192.168.100.100 02:11:32:21:d4:56
Wed Jul 14 18:20:11 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%LAN_FB@eth0 (Permission denied)
Wed Jul 14 18:20:11 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Wed Jul 14 18:20:27 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%LAN_FB@eth0 (Permission denied)
Wed Jul 14 18:20:27 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Wed Jul 14 18:20:43 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%LAN_FB@eth0 (Permission denied)
Wed Jul 14 18:20:43 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Wed Jul 14 18:20:59 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%LAN_FB@eth0 (Permission denied)
Wed Jul 14 18:20:59 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Wed Jul 14 18:21:15 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%LAN_FB@eth0 (Permission denied)
Wed Jul 14 18:21:15 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Wed Jul 14 18:21:31 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%LAN_FB@eth0 (Permission denied)
Wed Jul 14 18:21:31 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Wed Jul 14 18:21:47 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%LAN_FB@eth0 (Permission denied)
Wed Jul 14 18:21:47 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Wed Jul 14 18:22:03 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%LAN_FB@eth0 (Permission denied)
Wed Jul 14 18:22:03 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%lan@eth0 (Permission denied)
Wed Jul 14 18:22:19 2021 daemon.err odhcpd[20785]: Failed to send to ff02::1%LAN_FB@eth0 (Permission denied)

flimo44, 2021-07-14:
I have also a last issue :/ since i have change my VPS with IPV6.
Impossible to have IPV6 with the DHCP. 
[image]

ping V6 is OK both router and VPS. No error anything.

One day that I'm looking for without finding where the problem comes from.

flimo44, 2021-07-15:
Resolved my ipv6 Dhcp. 
Problem solved by remove IPV6 and IV6 Gateway on my Freebox WAN....

closed 2021-07-16 by flimo44