Ysurac/openmptcprouter #1957: Router installed and working, but need help to install capture device

hi10bha, 2021-07-15:
Hi , I have installed the router and is working the way it is designed. I am looking for help and want to know if I can install my usb 3.0 capture card on the same system and need help with my drivers , my device is Magewell SDI/HDMI capture device, if someone can guide me through it will be a great help

Ysurac, 2021-07-15:
If it's supported by OpenWrt, this should be supported by OpenMPTCProuter.
But I don't think OpenWrt support this device.

hi10bha, 2021-07-15:
Thanks for so quick reply

I am attaching a pic , looking to make a device like this. So only one
device is used for bonding and streaming both.

hi10bha, 2021-07-15:
thanks for your reply,

i have one more query and want to know how many lan cards can be installed
and how do i install the drivers for my lan cards, m really sorry to ask
these questions but. m completely new, but m happy that i am able to run my
mptcprouter is installed and working fine, all i want to do is to install
multiple ethernet cards

thanks

closed 2022-01-18 by github-actions[bot]