Ysurac/openmptcprouter #1961: OpenMPTCProuter on esxi-arm vm

jjak0b, 2021-07-16:
## Expected Behavior
Try to setup openMPTCProuter on ESXi 7.0.0 virtual machine running on RPI4 

## Current Behavior
I did the following steps:
1. got ext4 image: [openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz](https://download.openmptcprouter.com/release/v0.58.5/rpi4/targets/bcm27xx/bcm2711/openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2711-rpi-4-ext4-factory.img.gz)
2. extracted image
3. converted img to vmdk: `qemu-img convert -f raw -O vmdk "openmptcprouter-v0.58.5-r0+16336-b36068d35d-bcm27xx-bcm2711-rpi-4-ext4-factory.img" "openmptcprouter-v0.58.5-r0-rpi-4-ext4-factory2.vmdk`
4. uploaded the vmdk to esxi
5. ssh to esxi
6. converted vmdk for esxi: `vmkfstools -i openmptcprouter-v0.58.5-r0-rpi-4-ext4-factory2.vmdk -d zeroedthick open
mptcprouter-v0.58.5-r0-rpi-4-ext4-factory.esxi.vmdk`
7. setup VM using #87 configuration
8. Run VM 
9. VM doesn't boot

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version:  N/A
  - OpenMPTCProuter platform: RPI4

Adorfer, 2021-07-16:
what are your mass storage/boot options set for the vm?

jjak0b, 2021-07-16:
The boot options in boot options order are:
1. EFI Hard Drive
2. EFI Network
3. EFI Network 1
4. EFI Internal shell
the vm try to boot try by Network, and the boot through Hard drive fail immediately even if i try to boot selecting the Hard drive option

jjak0b, 2021-07-16:
Can the issue be related to the fact that the image must be an EFI image for RPI4 ?
To install the EFI Firmware for RPI4 i followed [this guide](https://download3.vmware.com/software/vmw-tools/esxi-on-arm/Fling-on-Raspberry-Pi.pdf).
But how can i generate an efi image for RPI4 ?

Adorfer, 2021-07-17:
my question was more: what hard drive emulation did you offer? virtio? sata? ide?
perhaps try with a plain openwrt first. if that works then we can look here. 
(At least my guess is it's more about the general esxi driver setup)
in german it may look like this: 
[image]

jjak0b, 2021-07-17:
Oh i'm sorry. these are the hard disk settings:

[image]

About Openwrt, when i have enough time i'm going to test [this](https://forum.openwrt.org/t/arm-fling-installation/83740/4)

closed 2021-10-20 by github-actions[bot]

thelastdreamer, 2022-01-04:
hello, did someone manage to make it work? i tried on Esxi arm 7.0 as mentioned in the top but i coudn't manage to make it work

[image]
[image]
[image]

thanks a lot!

xunil79, 2022-06-15:
I have the same issue. Would be helpful to have a vmdk tailored for esxi fling

darkmaniac7, 2022-08-10:
Also had issues getting this to work initially. Here was the steps to follow for me to get it working. Now have OMR & PfSense virtualized in 6.7u3

Step 1 Download latest version combined.img.gz
Step 2 using this guide from OpenWRT - https://wi1dcard.dev/posts/convert-openwrt-image-to-esxi-vmdk/

Download qemu-img - https://cloudbase.it/qemu-img-windows/
Unrar/Zip *.img.gz
type command 'qemu-img convert -f raw -O vmdk image_name_and_directory_here.img Converted_image_name_and_directory_here.vmdk'
Take that *.vmdk and upload it to your datastore. Open SSH and navigate to your datastore

"cd /vmfs/volumes/datastore_name_here/
type the command vmkfstools -i 'image_name_and_directory_here.vmdk' Converted_image_name_and_directory_here*.vmdk -d thin
Step 3 Create VM

Compatibility - Latest available
OS - Linux
Version - Other 4.x 64-bit

Select Storage

Delete existing USB Controller, Sata controller, Hard Disk, CD Drive, etc. Add existing hard disk, select converted VMDK.
Adjust CPU / Memory as needed
Set SCSI Controller to LSI Logic SAS
Set Network Adapter to E1000e

go to Boot Options and change from EFI to BIOS

Step 4
Finish and power on

thelastdreamer, 2022-08-12:
> Also had issues getting this to work initially. Here was the steps to follow for me to get it working. Now have OMR & PfSense virtualized in 6.7u3
>
> Step 1 Download latest version combined.img.gz [...]

thanks man, but can you please tell me which file of Openmptcrouter you used? 

ext4 efi for example?

thanks man

darkmaniac7, 2022-08-12:
> thanks man, but can you please tell me which file of Openmptcrouter you used?
> 
> ext4 efi for example?

Sure -

https://download.openmptcprouter.com/release/v0.59rc1-5.4/x86_64/targets/x86/64/openmptcprouter-v0.59rc1-5.4-r0+16594-ce92de8c8c-x86-64-generic-ext4-combined-efi.img.gz