Ysurac/openmptcprouter #1962: Strange errors with my Linksys USB Ethernet Adapter

darthclide, 2021-07-17:
## Expected Behavior
Stable connection to Twitch and the system log should just be normal dhcp and information.

## Current Behavior
Internet randomly loses connection to my Twitch server for ~20 seconds but all other connections (discord, etc) work perfectly fine during that 20 second drop out. (repeatable many times for many weeks now). I can't remember the exact logs, but I have had my raspberry completely lock up right after I refresh my system log to see a full page of "r8152 1-1.3:1.0 eth1 Rx status -71" errors

`Fri Jul 16 23:38:37 2021 kern.info kernel: [55082.686866] NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55082.693374] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x2b8/0x2c0
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55082.701631] Modules linked in: ath9k ath9k_htc ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial pppoe ppp_async option ipw cdc_mbim brcmfmac ath9k_hw ath wireguard usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx smsc75xx sierra_net sierra rtlwifi rtl8xxxu rtl8812au rtl8187 rt2x00usb rt2x00lib rndis_host qmi_wwan pppox ppp_generic pl2303 oti6858 mt7601u mos7720 mmc_spi mct_u232 mcs7830 mac80211 lzo libchacha20poly1305 libblake2s keyspan kalmia ipt_REJECT huawei_cdc_ncm garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cypress_m8 cp210x chacha_neon ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa ax88179_178a asix ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55082.701696]  xt_cgroup xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor via_velocity via_rhine usbserial usbnet usbhid tulip ts_fsm ts_bm solos_pci slhc sky2 skge sis900 sis190 sch_cake rtl8150 r8712u(C) r8169 r8152 r8125 r6040 poly1305_neon pegasus pcnet32 of_mmc_spi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci macvlan lzo_decompress lzo_compress libcurve25519_generic libchacha libblake2s_generic kaweth iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55082.789388]  ip6table_raw ip_tables hso hid_generic forcedeth ezusb ethoc et131x ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 e1000e e100 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_acm brcmutil bnx2 atl2 atl1e atl1c atl1 asn1_decoder arptable_filter arpt_mangle arp_tables 8390 8250_pci 8139too 8139cp tcp_nanqinlang tcp_bbr2 sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835(C) hid evdev i2c_gpio i2c_algo_bit i2c_dev spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55082.876130]  ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig bonding 3c59x ip6_gre ip_gre gre vmxnet3 e1000 ifb nat46 ip6_udp_tunnel udp_tunnel sit sctp libcrc32c ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel netlink_diag tunnel6 tunnel4 ip_tunnel hfcpci hfcmulti veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo autofs4 br2684 atm nls_utf8 zram zsmalloc natsemi eeprom_93cx6 sha1_generic md5 kpp ghash_generic gf128mul gcm echainiv des_generic libdes deflate zlib_inflate zlib_deflate authenc crypto_acompress vfat fat nls_iso8859_1 nls_cp437 ahci libahci libata fsl_mph_dr_of ehci_platform ehci_fsl
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55082.963312]  ehci_hcd gpio_button_hotplug tg3 b44 ssb ptp realtek pps_core
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.057647] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.106 #0
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.065034] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.070859] pstate: 40400005 (nZcv daif +PAN -UAO)
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.075644] pc : dev_watchdog+0x2b8/0x2c0
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.079644] lr : dev_watchdog+0x2b8/0x2c0
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.083643] sp : ffffffc010003da0
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.086947] x29: ffffffc010003da0 x28: 00000000000000e0
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.092251] x27: 0000000000000140 x26: 00000000ffffffff
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.097554] x25: 0000000000000000 x24: 0000000000000000
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.102857] x23: 0000000000000001 x22: ffffff8079780000
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.108159] x21: ffffff8079780480 x20: ffffffc0109e6000
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.113462] x19: 0000000000000000 x18: 0000000000000000
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.118764] x17: 0000000000000000 x16: 0000000000000000
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.124066] x15: 0000000000000001 x14: 0720072007200720
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.129369] x13: 0720072007200720 x12: 07740775076f0720
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.134672] x11: 07640765076d0769 x10: 0774072007300720
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.139974] x9 : 0765077507650775 x8 : 0771072007740769
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.145277] x7 : 076d0773076e0761 x6 : 0000000000000001
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.150579] x5 : 0000000000000000 x4 : 0000000000000000
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.155882] x3 : 0000000000000000 x2 : ffffff807fb97258
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.161184] x1 : ffffffc06f1c0000 x0 : 0000000000000039
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.166486] Call trace:
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.168925]  dev_watchdog+0x2b8/0x2c0
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.172579]  call_timer_fn.isra.34+0x20/0x78
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.176839]  run_timer_softirq+0x1e8/0x478
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.180926]  __do_softirq+0x11c/0x250
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.184580]  irq_exit+0x9c/0xb8
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.187714]  __handle_domain_irq+0x64/0xb8
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.191800]  gic_handle_irq+0x5c/0xb8
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.195452]  el1_irq+0xf0/0x1c0
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.198586]  arch_cpu_idle+0x10/0x18
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.202153]  do_idle+0x1e4/0x258
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.205371]  cpu_startup_entry+0x20/0x78
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.209286]  rest_init+0xb0/0xbc
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.212504]  arch_call_rest_init+0xc/0x14
Fri Jul 16 23:38:37 2021 kern.warn kernel: [55083.216504]  start_kernel+0x3c4/0x3dc
[... cut ...]
Some kind of driver/kernel change does better with this Linksys USB Ethernet Adapter? This thread looks slightly related:
https://bugzilla.kernel.org/show_bug.cgi?id=198931

## Steps to Reproduce the Problem

Even without streaming, these errors will randomly show up in the system log.

## Context (Environment)
Having my raspberry randomly lose connection to my eth1 as well as having to restart due to lockups is very frustrating.

## Specifications

  - OpenMPTCProuter version: 58.4
  - OpenMPTCProuter VPS version: Version 0.1026 5.4.100-mptcp
  - OpenMPTCProuter platform: RPI4

Ysurac, 2021-07-17:
And if you try ```ethtool -k eth1 tx off``` via SSH on the router ?

darthclide, 2021-07-17:
For now I just switched the device over to a USB 2.0 port. If I see these errors again I will try your command both when I am plugged into this port, as well as going back to the USB 3.0 port. Weird how this issue wasn't happening for many months, but now it is all of a sudden.

darthclide, 2021-07-18:
Woke up and realized I could just copy my kernel log from the past couple days:

[55082.686866] NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out
[55082.693374] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x2b8/0x2c0
[55082.701631] Modules linked in: ath9k ath9k_htc ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial pppoe ppp_async option ipw cdc_mbim brcmfmac ath9k_hw ath wireguard usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx smsc75xx sierra_net sierra rtlwifi rtl8xxxu rtl8812au rtl8187 rt2x00usb rt2x00lib rndis_host qmi_wwan pppox ppp_generic pl2303 oti6858 mt7601u mos7720 mmc_spi mct_u232 mcs7830 mac80211 lzo libchacha20poly1305 libblake2s keyspan kalmia ipt_REJECT huawei_cdc_ncm garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cypress_m8 cp210x chacha_neon ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa ax88179_178a asix ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment
[55082.701696]  xt_cgroup xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor via_velocity via_rhine usbserial usbnet usbhid tulip ts_fsm ts_bm solos_pci slhc sky2 skge sis900 sis190 sch_cake rtl8150 r8712u(C) r8169 r8152 r8125 r6040 poly1305_neon pegasus pcnet32 of_mmc_spi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci macvlan lzo_decompress lzo_compress libcurve25519_generic libchacha libblake2s_generic kaweth iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth
[55082.789388]  ip6table_raw ip_tables hso hid_generic forcedeth ezusb ethoc et131x ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 e1000e e100 crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_acm brcmutil bnx2 atl2 atl1e atl1c atl1 asn1_decoder arptable_filter arpt_mangle arp_tables 8390 8250_pci 8139too 8139cp tcp_nanqinlang tcp_bbr2 sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred snd_bcm2835(C) hid evdev i2c_gpio i2c_algo_bit i2c_dev spi_ks8995 xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark
[55082.876130]  ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig bonding 3c59x ip6_gre ip_gre gre vmxnet3 e1000 ifb nat46 ip6_udp_tunnel udp_tunnel sit sctp libcrc32c ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel netlink_diag tunnel6 tunnel4 ip_tunnel hfcpci hfcmulti veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo autofs4 br2684 atm nls_utf8 zram zsmalloc natsemi eeprom_93cx6 sha1_generic md5 kpp ghash_generic gf128mul gcm echainiv des_generic libdes deflate zlib_inflate zlib_deflate authenc crypto_acompress vfat fat nls_iso8859_1 nls_cp437 ahci libahci libata fsl_mph_dr_of ehci_platform ehci_fsl
[55082.963312]  ehci_hcd gpio_button_hotplug tg3 b44 ssb ptp realtek pps_core
[55083.057647] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.106 #0
[55083.065034] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[55083.070859] pstate: 40400005 (nZcv daif +PAN -UAO)
[55083.075644] pc : dev_watchdog+0x2b8/0x2c0
[55083.079644] lr : dev_watchdog+0x2b8/0x2c0
[55083.083643] sp : ffffffc010003da0
[55083.086947] x29: ffffffc010003da0 x28: 00000000000000e0 
[55083.092251] x27: 0000000000000140 x26: 00000000ffffffff 
[55083.097554] x25: 0000000000000000 x24: 0000000000000000 
[55083.102857] x23: 0000000000000001 x22: ffffff8079780000 
[55083.108159] x21: ffffff8079780480 x20: ffffffc0109e6000 
[55083.113462] x19: 0000000000000000 x18: 0000000000000000 
[55083.118764] x17: 0000000000000000 x16: 0000000000000000 
[55083.124066] x15: 0000000000000001 x14: 0720072007200720 
[55083.129369] x13: 0720072007200720 x12: 07740775076f0720 
[55083.134672] x11: 07640765076d0769 x10: 0774072007300720 
[55083.139974] x9 : 0765077507650775 x8 : 0771072007740769 
[55083.145277] x7 : 076d0773076e0761 x6 : 0000000000000001 
[55083.150579] x5 : 0000000000000000 x4 : 0000000000000000 
[... cut ...]
[123556.997780]  irq_exit+0x9c/0xb8
[123557.001001]  __handle_domain_irq+0x64/0xb8
[123557.005174]  gic_handle_irq+0x5c/0xb8
[123557.008913]  el1_irq+0xf0/0x1c0
[123557.012133]  arch_cpu_idle+0x10/0x18
[123557.015787]  do_idle+0x1e4/0x258
[123557.019093]  cpu_startup_entry+0x24/0x78
[123557.023094]  rest_init+0xb0/0xbc
[123557.026400]  arch_call_rest_init+0xc/0x14
[123557.030486]  start_kernel+0x3c4/0x3dc
[123557.034225] ---[ end trace c79dfe43bd051e3a ]---
[123557.039024] ipheth 1-1.1:4.2: ipheth_tx: usb_submit_urb: -16

Adorfer, 2021-07-20:
what kind of powered hub do you have between the RPI4 and the Stick? 
I strongly recommend one with 12V PSU since the ones with 5V very often deliver only 4.6V to the USB port when loaded (and if that happens then you have no benefit of that HUB, voltage weak like plugging it directly to the Raspi)

darthclide, 2021-07-20:
If you have any recommendations that have almost no chance of feeding back power into the raspberry, please do tell. My current powered hub delivers at most 2.4A across all ports. If I could get a powered USB hub that delivers up to 4.0A across all ports while not destroying my rpi4 it would be amazing.