Ysurac/openmptcprouter #1963: Every morning raspberry 4 has day before date

highfreq, 2021-07-17:
Every morning my rasp 4 has day before date. It doesn't update it and nothing works because of the difference with server. As soon as i force it manually everything start but tomorrow will be same prob. Looks like can't contact ntp.

Fri Jul 16 17:16:15 2021 daemon.info dnsmasq[7334]: using nameserver 127.0.0.1#5353
Fri Jul 16 17:16:15 2021 daemon.info dnsmasq[7334]: using 1 more local addresses
Fri Jul 16 17:16:15 2021 daemon.info dnsmasq[7334]: read /etc/hosts - 4 addresses
Fri Jul 16 17:16:15 2021 daemon.info dnsmasq[7334]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Fri Jul 16 17:16:15 2021 daemon.info dnsmasq-dhcp[7334]: read /etc/ethers - 0 addresses
Fri Jul 16 17:16:15 2021 daemon.info dnsmasq[7334]: read /etc/hosts - 4 addresses
Fri Jul 16 17:16:15 2021 daemon.info dnsmasq[7334]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Fri Jul 16 17:16:15 2021 daemon.info dnsmasq-dhcp[7334]: read /etc/ethers - 0 addresses
Fri Jul 16 17:16:15 2021 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/glorytun-udp
Fri Jul 16 17:16:15 2021 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/openmptcprouter-vps
Fri Jul 16 17:16:15 2021 user.notice ucitrack: Setting up /etc/config/glorytun-udp reload dependency on /etc/config/openmptcprouter-vps
Fri Jul 16 17:16:16 2021 user.notice ucitrack: Setting up /etc/config/shadowsocks-libev reload dependency on /etc/config/omr-tracker
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: checking if ModemManager is available...
Fri Jul 16 17:16:16 2021 user.notice ucitrack: Setting up /etc/config/mlvpn reload trigger for non-procd /etc/init.d/mlvpn
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty28, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty28
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty28, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty56, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty56
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty56, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty18, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty18
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty18, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty1, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty1
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty1, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty46, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty46
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty46, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 daemon.err odhcpd[4277]: Failed to send to ff02::1%lan@eth0 (Address not available)
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty36, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty36
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty36, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty26, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty26
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty26, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty54, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty54
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty54, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty16, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty16
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty16, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty44, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty44
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty44, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=console, subsystem=tty, sysfspath=/sys/devices/virtual/tty/console
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=console, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice mptcp: Reloading mptcp config due to ifup of lan (eth0)
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty34, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty34
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty34, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty62, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty62
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty62, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty24, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty24
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty24, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty8, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty8
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty8, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty52, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty52
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty52, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty14, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty14
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: event reported: action=add, name=tty14, subsystem=tty
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 16 17:16:16 2021 user.notice ModemManager: hotplug: cached event found: action=add, name=tty42, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty42
[... 533 lines cut ...]
[... cut ...]
Fri Jul 16 17:23:22 2021 daemon.err odhcpd[4277]: Failed to send to ff02::1%lan@eth0 (Address not available)
Fri Jul 16 17:23:38 2021 daemon.err odhcpd[4277]: Failed to send to ff02::1%lan@eth0 (Address not available)
Fri Jul 16 17:23:54 2021 daemon.err odhcpd[4277]: Failed to send to ff02::1%lan@eth0 (Address not available)
Fri Jul 16 17:24:10 2021 daemon.err odhcpd[4277]: Failed to send to ff02::1%lan@eth0 (Address not available)
Fri Jul 16 17:24:26 2021 daemon.err odhcpd[4277]: Failed to send to ff02::1%lan@eth0 (Address not available)

Ysurac, 2021-07-17:
Time is synchronized via NTP, you can check in System->System, "Time Synchronization" tab that it's enabled. You can also change ntp servers if you want. By default timezone is set to UTC.

highfreq, 2021-07-17:
First 2 checks enabled on system->time synchro. Now enable dhcp server. Saved and enabled and restarted but date is still at yesterday.

highfreq, 2021-07-17:
It looks like router can't resolve names untill a connection is established but a connection  can't be established with vps because time is off but time can't be set because it can't resolve 0.openwrt.ntp........

highfreq, 2021-07-17:
I can't ping www.google.com from router cli. As soon as i synchro time with browser than i can ping and resolve names. So i think the problem is in resolving names when the connection to vps is not up, but i don't know where to set it.

Ysurac, 2021-07-17:
You should be able to ping all before VPS is up, at least if System->OpenMPTCProuter, "advanced settings" tab, "Disable default gateway" is not checked.

highfreq, 2021-07-17:
It is not disabled, i can ping before vps is up but only ip numbers, i can't resolve names.

highfreq, 2021-07-17:
so my guess is that it can't resolve ntp name so it never gets an updated date without wich vps doesn't come up because of a too much different time and so on................

Ysurac, 2021-07-17:
You use default DNS or something else ?
DNS is disabled on all interfaces if they are set in DHCP ?

highfreq, 2021-07-17:
Deafault DNS.
All interfaces in DHCP, how do i disable DNS on interfaces? (if it is needed)

highfreq, 2021-07-18:
This morning same as yesterday, date not updated and dns not working. All default with a Rpi 4. Any hints?

highfreq, 2021-07-18:
I am not using the ethernet lan interface at the moment, router is only attached to usb modems, in future lan will be used for a streaming board.
Is it that the router looks for dns on lan/ethernet port instead of wan?

Ysurac, 2021-07-18:
I see that in log:
```
using nameserver 8.8.4.4#53
using nameserver 8.8.8.8#53
```
It's not default configuration for DNS.
In any case, you shouldn't have a date so far, RPI4 is rebooting during night ?
To remove DNS announced by WANs, it's in Network->Interfaces, on each wan interfaces, advanced parameters tab.

highfreq, 2021-07-18:
I added 8.8.8.8 & 8.8.4.4 this morning because it wasn't working again, but on first start was at default as per log from yesterday.
RPI is rebooting, what do you mean i shouldn't have a date?

highfreq, 2021-07-18:
In network-->interfaces->advanced i have nothing like DNS announced by WAN, i just have quite a few IPv6 parameters the first of wich Use builtin IPv6-management, is the only one checked.

Ysurac, 2021-07-18:
If WANs are set as DHCP client, you should have "Use DNS servers advertised by peer" unchecked. Same for "Use default gateway" that should be unchecked.
Why RPI is rebooting during night ?

highfreq, 2021-07-18:
It is rebooting because it is a portable unit we use to stream video.
Use DNS servers advertised by peer is NOT present in Network->interfaces->edit->advanced settings
Here what i  have:

Use builtin IPv6-management

Force link

Set interface properties regardless of the link carrier (If set, carrier sense events do not invoke hotplug handlers).
IPv6 assignment length
disabled
▾
Assign a part of given length of every public IPv6-prefix to this interface
IPv6 address
Add IPv6 address…
+
IPv6 gateway
IPv6 routed prefix
Public prefix routed to this device for distribution to clients.
IPv6 suffix
::1
Optional. Allowed values: 'eui64', 'random', fixed value like '::1' or '::1:2'. When IPv6 prefix (like 'a:b:c:d::') is received from a delegating server, use the suffix (like '::1') to form the IPv6 address ('a:b:c:d::1') for the interface.
Multipath setting

Enabled
Only one interface must be set as Master.
Additional latency
0
Use broadcast flag

Required for certain ISPs, e.g. Charter with DOCSIS 3
Client ID to send when requesting DHCP
Vendor Class to send when requesting DHCP
Override MAC address
32:94:ca:9d:d7:9e
Override MTU
1500

Ysurac, 2021-07-18:
I would need your configuration available in System->OpenMPTCProuter, "Show all settings" tab.
I also need a screenshot of status page when it's not working.

highfreq, 2021-07-18:
asterisk.general=asterisk
asterisk.general.enabled='0'
asterisk.general.log_stderr='1'
asterisk.general.log_stdout='1'
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.@dnsmasq[0].dnssec='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].server='8.8.8.8' '8.8.4.4'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/zoom.us/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.ra_slaac='1'
dhcp.lan.force='1'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra='server'
dhcp.lan.ra_default='1'
dhcp.lan.ra_preference='high'
dhcp.lan.ra_management='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs7'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs4'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs4'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs4'
[... 1786 lines cut ...]
v4l2rtspserver.core.device='/dev/video0'
v4l2rtspserver.core.port='554'
v4l2rtspserver.core.resolution='640x480'
v4l2rtspserver.core.fps='15'
v4l2rtspserver.core.path='stream'
v4l2rtspserver.core.username='openwrt'
v4l2rtspserver.core.password='opxxxxxx'
v4l2rtspserver.core.format='h264'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.channel='auto'
wireless.radio0.country='IT'
wireless.radio0.cell_density='0'
wireless.radio0.hwmode='11g'
wireless.wifinet0=wifi-iface
wireless.wifinet0.device='radio0'
wireless.wifinet0.mode='sta'
wireless.wifinet0.network='wwan'
wireless.wifinet0.ssid='stream'
wireless.wifinet0.encryption='psk2'
wireless.wifinet0.key='oxen7til'
Powered by openmptcprouter v0.58.5 r0+16336-b36068d35d

highfreq, 2021-07-18:
When not working status page says "no data"

Ysurac, 2021-07-18:
I would also need the result of ```ip r``` via SSH on the router and ```dig google.com```.
Also using '8.8.8.8' '8.8.4.4' as DNS server is a bad idea, these IP are used by omr-tracker to check if connection is up or down and during check these IPs can not answer.

highfreq, 2021-07-18:
default via 10.255.255.1 dev tun0 
default via 192.168.8.1 dev eth1 metric 3 
default via 192.168.1.1 dev wlan0 metric 9 
default via 10.255.255.1 dev tun0 metric 1200 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1200 
80.211.11.99 metric 1 
	nexthop via 192.168.8.1 dev eth1 weight 1 
	nexthop via 192.168.1.1 dev wlan0 weight 10 
127.0.0.0/8 dev lo proto static scope link metric 5 
192.168.1.0/24 dev wlan0 scope link metric 9 
192.168.8.0/24 dev eth1 scope link metric 3

highfreq, 2021-07-18:
; <<>> DiG 9.17.11 <<>> google.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 23
;; flags: qr rd ra; QUERY: 1, ANSWER: 6, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;google.com.			IN	A

;; ANSWER SECTION:
google.com.		299	IN	A	173.194.76.138
google.com.		299	IN	A	173.194.76.100
google.com.		299	IN	A	173.194.76.101
google.com.		299	IN	A	173.194.76.102
google.com.		299	IN	A	173.194.76.139
google.com.		299	IN	A	173.194.76.113

;; Query time: 107 msec
;; SERVER: ::1#53(::1) (UDP)
;; WHEN: Sun Jul 18 10:00:31 CEST 2021
;; MSG SIZE  rcvd: 135

highfreq, 2021-07-18:
Changed now dns to 1.1.1.1 & 1.0.0.1

highfreq, 2021-07-18:
Reinstalled router clean, on first boot everything was ok, i than turned it off and went to lunch, on returning and switching on time is still at 2 hours ago and system doesn't come up. Usual problems, no name resolution, no time server name resolution and system down.
From router cli i can only ping numeric addresses.

Ysurac, 2021-07-18:
1.1.1.1 and 1.0.0.1 are also used by omr-tracker.
I need "ip r" and "dig google.com" when it's not working, I think the previous result was when it was working.

highfreq, 2021-07-18:
Hopefully tomorrow morning when i'll turn it on it'll do it again. I will post ip r and dig google.com.

Thanks for all your help

highfreq, 2021-07-19:
default via 10.255.255.1 dev tun0 
default metric 1 
	nexthop via 192.168.8.1 dev eth1 weight 2 
	nexthop via 192.168.1.1 dev wlan0 weight 1 
default via 192.168.8.1 dev eth1 metric 3 
default via 192.168.1.1 dev wlan0 metric 8 
default via 10.255.255.1 dev tun0 metric 1200 
10.255.255.1 dev tun0 proto kernel scope link src 10.255.255.2 
10.255.255.2 dev tun0 scope link metric 1200 
80.211.11.99 metric 1 
	nexthop via 192.168.8.1 dev eth1 weight 10 
	nexthop via 192.168.1.1 dev wlan0 weight 1 
127.0.0.0/8 dev lo proto static scope link metric 5 
192.168.1.0/24 dev wlan0 scope link metric 8 
192.168.8.0/24 dev eth1 scope link metric 3

highfreq, 2021-07-19:
; <<>> DiG 9.17.11 <<>> google.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 40778
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;google.com.			IN	A

;; Query time: 108 msec
;; SERVER: ::1#53(::1) (UDP)
;; WHEN: Sun Jul 18 19:06:03 CEST 2021
;; MSG SIZE  rcvd: 39

highfreq, 2021-07-19:
System status -----> NO DATA

highfreq, 2021-07-19:
Date of the system is stopped at yesterday time when i switched it off.

highfreq, 2021-07-19:
Ping of anything by name on router cli fails.

highfreq, 2021-07-19:
If i ping www.google.com from server it pings it on ipv6:

ping www.google.com
PING www.google.com(wo-in-x69.1e100.net (2a00:1450:400c:c07::69)) 56 data bytes
64 bytes from wo-in-x69.1e100.net (2a00:1450:400c:c07::69): icmp_seq=1 ttl=110 time=22.4 ms
64 bytes from wo-in-x69.1e100.net (2a00:1450:400c:c07::69): icmp_seq=2 ttl=110 time=22.4 ms
64 bytes from wo-in-x69.1e100.net (2a00:1450:400c:c07::69): icmp_seq=3 ttl=110 time=22.4 ms
64 bytes from wo-in-x69.1e100.net (2a00:1450:400c:c07::69): icmp_seq=4 ttl=110 time=22.5 ms
64 bytes from wo-in-x69.1e100.net (2a00:1450:400c:c07::69): icmp_seq=5 ttl=110 time=22.4 ms
64 bytes from wo-in-x69.1e100.net (2a00:1450:400c:c07::69): icmp_seq=6 ttl=110 time=22.4 ms

Ysurac, 2021-07-19:
VPN seems to be up.
For IPv6, it's enabled on the wizard ? If not check that a modem is not giving IPv6.

highfreq, 2021-07-19:
IP6 is enabled in wizard, i enabled it since my server has an ip6 address. Modem is giving ip4 addresses.
In Services---->shadowsocks-libev   i have ss_tunnel.dns DISABLED, is it normal?

Ysurac, 2021-07-19:
yes it's normal.
If ipv6 is enabled and working, ping should work too as it's also using same VPN.
Can you try to ping 10.255.255.1 (it's VPN VPS side ip) ?

highfreq, 2021-07-19:
I can ping 10.255.255.1

Ysurac, 2021-07-19:
ok, so VPN is working correctly.
From router can you ping 1.1.1.1 or not ? If not, can you ping vps public ip from router ?

highfreq, 2021-07-19:
I can ping 1.1.1.1 AND vps public ip.

Ysurac, 2021-07-19:
```dig google.com``` is still not working ?

highfreq, 2021-07-19:
; <<>> DiG 9.17.11 <<>> google.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 6832
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;google.com.			IN	A

;; Query time: 176 msec
;; SERVER: ::1#53(::1) (UDP)
;; WHEN: Sun Jul 18 20:00:37 CEST 2021
;; MSG SIZE  rcvd: 39

Ysurac, 2021-07-19:
Hum, I have an idea.
check the result of ```dig google.com +cd```

highfreq, 2021-07-19:
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 58272
;; flags: qr rd ra cd; QUERY: 1, ANSWER: 6, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;google.com.			IN	A

;; ANSWER SECTION:
google.com.		300	IN	A	108.177.15.139
google.com.		300	IN	A	108.177.15.102
google.com.		300	IN	A	108.177.15.101
google.com.		300	IN	A	108.177.15.100
google.com.		300	IN	A	108.177.15.113
google.com.		300	IN	A	108.177.15.138

;; Query time: 68 msec
;; SERVER: ::1#53(::1) (UDP)
;; WHEN: Sun Jul 18 20:15:41 CEST 2021
;; MSG SIZE  rcvd: 135

Ysurac, 2021-07-19:
OK
So the problem is that dnssec need date to be accurate but when ntp try to contact ntp servers, date is not yet accurate and so dnssec block queries. So it's a loop, should not append.
Can you try to disable DNSSEC in Network->DHCP & DNS, "Advanced settings" tab, and uncheck "DNSSEC" ?

highfreq, 2021-07-19:
Should i reboot after unchecking?

Ysurac, 2021-07-19:
You can try yes.

highfreq, 2021-07-19:
Still the same, date at last night

Ysurac, 2021-07-19:
ok...
What is the result of ```dig 0.openwrt.pool.ntp.org``` ? If this give you an IP, what is the result of ```ntpd -dnq -p 0.openwrt.pool.ntp.org```, it this doesn't work, the result of ```ntpd -dnq -I eth1 -p 0.openwrt.pool.ntp.org```

highfreq, 2021-07-19:
; <<>> DiG 9.17.11 <<>> 0.openwrt.pool.ntp.org
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 32464
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;0.openwrt.pool.ntp.org.		IN	A

;; Query time: 0 msec
;; SERVER: ::1#53(::1) (UDP)
;; WHEN: Sun Jul 18 20:25:11 CEST 2021
;; MSG SIZE  rcvd: 51

highfreq, 2021-07-19:
root@OpenMPTCProuter:~# ntpd -dnq -p 0.openwrt.pool.ntp.org
ntpd: bad address '0.openwrt.pool.ntp.org'
ntpd: bad address '0.openwrt.pool.ntp.org'

highfreq, 2021-07-19:
root@OpenMPTCProuter:~# ntpd -dnq -I wlan0 -p 0.openwrt.pool.ntp.org
ntpd: bad address '0.openwrt.pool.ntp.org'

highfreq, 2021-07-19:
I did it on wlan0 because is the interface i use now

Ysurac, 2021-07-19:
Domain doesn't resolve.
Can you try ```dig +cd 0.openwrt.pool.ntp.org```, if you get an IP, disable also DNSSEC in Services->Recursive DNS, uncheck "Enable DNSSEC" in basic tab.

highfreq, 2021-07-19:
Still date at yesterday

Ysurac, 2021-07-19:
retry ```dig 0.openwrt.pool.ntp.org```, still no IP ?

highfreq, 2021-07-19:
root@OpenMPTCProuter:~# dig 0.openwrt.pool.ntp.org

; <<>> DiG 9.17.11 <<>> 0.openwrt.pool.ntp.org
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 43243
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;0.openwrt.pool.ntp.org.		IN	A

;; Query time: 0 msec
;; SERVER: ::1#53(::1) (UDP)
;; WHEN: Sun Jul 18 20:24:04 CEST 2021
;; MSG SIZE  rcvd: 51

Ysurac, 2021-07-19:
I made some tests, and with DNSSEC disabled in Network->DHCP & DNS (DNSSEC & DNSSEC check unsigned in advanced settings tab) and in Services->Recursive DNS, this seems to work.
I will make more test to fix this issue.

Ttyd, 2021-08-17:
what are you doing&nbsp;

------------------&nbsp;原始邮件&nbsp;------------------
发件人:                                                                                                                        "Ysurac/openmptcprouter"                                                                                    ***@***.***&gt;;
发送时间:&nbsp;2021年7月18日(星期天) 下午3:46
***@***.***&gt;;
***@***.******@***.***&gt;;
主题:&nbsp;Re: [Ysurac/openmptcprouter] Every morning raspberry 4 has day before date (#1963)

 
asterisk.general=asterisk
 asterisk.general.enabled='0'
 asterisk.general.log_stderr='1'
 asterisk.general.log_stdout='1'
 ***@***.***[0]=dnsmasq
 ***@***.***[0].domainneeded='1'
 ***@***.***[0].localise_queries='1'
 ***@***.***[0].rebind_protection='1'
 ***@***.***[0].rebind_localhost='1'
 ***@***.***[0].local='/lan/'
 ***@***.***[0].domain='lan'
 ***@***.***[0].expandhosts='1'
 ***@***.***[0].authoritative='1'
 ***@***.***[0].readethers='1'
 ***@***.***[0].leasefile='/tmp/dhcp.leases'
 ***@***.***[0].localservice='1'
 ***@***.***[0].ednspacket_max='1232'
 ***@***.***[0].dnssec='1'
 ***@***.***[0].noresolv='1'
 ***@***.***[0].nonegcache='1'
 ***@***.***[0].rebind_domain='plex.direct'
 ***@***.***[0].server='8.8.8.8' '8.8.4.4'
 ***@***.***[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/zoom.us/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
 dhcp.lan=dhcp
 dhcp.lan.interface='lan'
 dhcp.lan.start='100'
 dhcp.lan.limit='150'
 dhcp.lan.leasetime='12h'
 dhcp.lan.dhcpv4='server'
 dhcp.lan.ra_slaac='1'
 dhcp.lan.force='1'
 dhcp.lan.dhcpv6='server'
 dhcp.lan.ra='server'
 dhcp.lan.ra_default='1'
 dhcp.lan.ra_preference='high'
 dhcp.lan.ra_management='1'
 dhcp.lan.ra_flags='managed-config' 'other-config'
 dhcp.wan=dhcp
 dhcp.wan.interface='wan'
 dhcp.wan.ignore='1'
 dhcp.odhcpd=odhcpd
 dhcp.odhcpd.maindhcp='0'
 dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
 dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
 dhcp.odhcpd.loglevel='4'
 ***@***.***[0]=dropbear
 ***@***.***[0].PasswordAuth='on'
 ***@***.***[0].RootPasswordAuth='on'
 ***@***.***[0].Port='22'
 ***@***.***[0]=classify
 ***@***.***[0].direction='both'
 ***@***.***[0].proto='icmp'
 ***@***.***[0].class='cs7'
 ***@***.***[0].comment='ICMP'
 ***@***.***[1]=classify
 ***@***.***[1].direction='both'
 ***@***.***[1].proto='udp'
 ***@***.***[1].class='cs4'
 ***@***.***[1].src_port='53,123,5353'
 ***@***.***[1].comment='DNS udp and NTP'
[... 1605 lines cut ...]
 unbound.fwd_adguard_standard.tls_upstream='1'
 unbound.fwd_adguard_standard.zone_type='forward_zone'
 unbound.fwd_adguard_standard.server='176.103.130.130' '176.103.130.131'
 unbound.fwd_adguard_standard.zone_name='.'
 unbound.fwd_cloudflare_family=zone
 unbound.fwd_cloudflare_family.enabled='0'
 unbound.fwd_cloudflare_family.fallback='1'
 unbound.fwd_cloudflare_family.tls_index='family.cloudflare-dns.com'
 unbound.fwd_cloudflare_family.tls_upstream='1'
 unbound.fwd_cloudflare_family.zone_type='forward_zone'
 unbound.fwd_cloudflare_family.server='1.1.1.3' '1.0.0.3'
 unbound.fwd_cloudflare_family.zone_name='.'
 unbound.fwd_cloudflare_malware=zone
 unbound.fwd_cloudflare_malware.enabled='0'
 unbound.fwd_cloudflare_malware.fallback='1'
 unbound.fwd_cloudflare_malware.tls_index='security.cloudflare-dns.com'
 unbound.fwd_cloudflare_malware.tls_upstream='1'
 unbound.fwd_cloudflare_malware.zone_type='forward_zone'
 unbound.fwd_cloudflare_malware.server='1.1.1.2' '1.0.0.2'
 unbound.fwd_cloudflare_malware.zone_name='.'
 unbound.fwd_odvr=zone
 unbound.fwd_odvr.enabled='0'
 unbound.fwd_odvr.fallback='1'
 unbound.fwd_odvr.tls_index='odvr.nic.cz'
 unbound.fwd_odvr.tl

tanganellilore, 2021-09-19:
@Ysurac I have similar issue, but in my case i can't ping Nothing and resolve name via default route.

Via eth0 or wan1 I'm qbe to resolve and ping everything but without specofoed it this is not possibile.

Moreover I'm able to ping tun0 and viceversa. 

I see that in the default route the first entry (with metric 0) is the Tun0 via vpn tunnel.
So i think the issue is here, default vpn can't reach internet and cant resolve it.

In fact if i disabile the vpn Interface, everything work well, because default route was removed.

It's possibile something related fw rule in the vps or omr side?

Ysurac, 2021-09-19:
@tanganellilore I don't understand what you mean. This issue is about NTP not working correctly with a big diff when DNSSEC enabled (it's the default setting). This will be solved in next release.

tanganellilore, 2021-09-19:
I mean, i read the discussione and i see other people that try to resolve ntp via dig or similar command.
I have exatly same issue, big time differance, but after some check i discover that everything, without Interface specified, not work.

I just avoid to open other issue for similar problem

[4 more replies]

closed 2022-01-07 by github-actions[bot]