Ysurac/openmptcprouter #1964: On reboot NIC - eth changes

5UP3RR3D, 2021-07-19:
Since the last release NICs don't remember their configuration. My LAN interface ( eth0) changes to eth2 or eth3 and the WAN interfaces changes to eth0 - so my hardware routing won't fit to the config.

Ysurac, 2021-07-19:
Did you try to uncheck "Disable interfaces auto rename" in System->OpenMPTCProuter, "advanced settings" tab ?

5UP3RR3D, 2021-07-19:
Yes! That's why I think it's buggy. I use tp-link usb3 - Eth Adapters - no problems with Huawei/ZTE USB Modems (which simulate an eth) - on the previous version everything is fine.

Ysurac, 2021-07-19:
I made no changes on this since previous major release, so this should work same way.
I would need State->System log when there is info about this interface.

5UP3RR3D, 2021-08-17:
A rewrite of a non efi image + reinstall of the vps did it. Now everything works fine. Layer8 error - my bad!