Ysurac/openmptcprouter #1969: MBIM modem protocol support

starkovv, 2021-07-20:
There is no "MBIM" option listed in Network –> Interfaces –> [some_interface] –> Protocol drop-down menu.

How can I add MBIM modem to the interfaces list?

I can do it via CLI, but it seems the web GUI doesn't support this thing.

starkovv, 2021-07-20:
Here is what I do to make MBIM working:
1. Install bash package and all *mbim* packages.

2. Insert USB modem into my NanoPI R4S.

3. In GUI:
Add interface of type "Unmanaged" and physical device "wwan0".

4. In CLI:
```
cat << EOF > /etc/mbim-network.conf
APN=internet 
APN_USER=
APN_PASS=
APN_AUTH=
PROXY=yes
EOF

cd /usr/bin
wget https://raw.githubusercontent.com/elementzonline/GSMModem/master/SIM7600/mbim-set-ip/mbim-set-ip
chmod +x mbim-set-ip

mbim-network /dev/cdc-wdm0 start

mbim-set-ip /dev/cdc-wdm0 wwan0
```

Done.

Ysurac, 2021-07-20:
You should be able to use modemmanger

starkovv, 2021-07-20:
If I choose protocol "ModemManager" then the new interface is added and it returns "Error: Unknown error (MM_TEARDOWN_IN_PROGRESS)"

starkovv, 2021-07-22:
UP

How to properly debug the "Unknown error" issue?
Thanks.

Ysurac, 2021-07-22:
The unknown error are not real errors, only message not know by interface, this is "fixed" in snapshot release.

starkovv, 2021-07-22:
Looking forward to test it.

I never tried snapshot release. Where can I find & download one for NanoPI R4S?

Ysurac, 2021-07-23:
Images are here: https://download.openmptcprouter.com/develop/5.4/

starkovv, 2021-07-23:
I reflashed SD card with the v0.58.5-r0+17035-1818b038d7.
Boot and installed mbim-utils package. Rebooted.

Connected a modem with USB 3.0 in MBIM mode to the NanoPI R4S.
Added the modem as WAN2 interface (physical device wwan0) with protocol ModemManager. Now it shows `Error: Network device is not present`.

How to debug it further?
Thanks.

Ysurac, 2021-07-23:
I don't use any MBIM device. You can refer to OpenWrt documentation.
For ModemManager you can use ```mmcli``` command to get all informations.

starkovv, 2021-07-23:
Do you use modems in QMI Cellular protocol? Which modem protocol best suits the OMR?

Ysurac, 2021-07-23:
I personally use a Huawei HiLink, so it's see as a normal ethernet interface on the router.
But some users are using QMI at least and maybe MBIM but I don't remember.

starkovv, 2021-07-23:
I tried OMR with four Huawei E3372 HiLink, works like a charm.

starkovv, 2021-07-23:
Rebooted the router and now the modem continuously connecting -> disconnecting from the carrier.

This line are repeated in System log every few seconds:
```
[...]
Fri Jul 23 19:14:19 2021 daemon.notice netifd: Interface 'wan3' is setting up now
Fri Jul 23 19:14:19 2021 daemon.notice netifd: wan3 (29192): modem available at /org/freedesktop/ModemManager1/Modem/0
Fri Jul 23 19:14:19 2021 daemon.notice netifd: wan3 (29192): starting connection with apn 'internet'...
Fri Jul 23 19:14:19 2021 daemon.info [4911]: <info>  [modem0] simple connect started...
Fri Jul 23 19:14:19 2021 daemon.info [4911]: <info>  [modem0] simple connect state (3/8): enable
Fri Jul 23 19:14:19 2021 daemon.info [4911]: <info>  [modem0] state changed (disabled -> enabling)
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] state changed (enabling -> enabled)
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] simple connect state (5/8): register
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] 3GPP registration state changed (unknown -> registering)
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] 3GPP registration state changed (registering -> home)
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] state changed (enabled -> registered)
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] simple connect state (6/8): bearer
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] simple connect state (7/8): connect
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] state changed (registered -> connecting)
Fri Jul 23 19:14:20 2021 daemon.debug [4911]: Using dynamic session ID 0
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0/bearer9] net link mbimmux0.0 created (session id 0)
Fri Jul 23 19:14:20 2021 kern.warn kernel: [  170.559212] netlink: 'ModemManager': attribute type 1 has an invalid length.
Fri Jul 23 19:14:20 2021 user.notice ModemManager: hotplug: add network interface mbimmux0.0: event processed
Fri Jul 23 19:14:20 2021 user.notice ModemManager: hotplug: event reported: action=add, name=mbimmux0.0, subsystem=net
Fri Jul 23 19:14:20 2021 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] state changed (connecting -> connected)
Fri Jul 23 19:14:20 2021 daemon.info [4911]: <info>  [modem0] simple connect state (8/8): all done
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): successfully connected the modem
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): signal refresh rate is not set
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): network operator name: Beeline
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): network operator MCCMNC: 25099
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): registration type: home
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): access technology: lte
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): signal quality: 22%
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): IPv4 connection setup required in interface wan3: static
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): adding IPv4 address 10.102.143.33, netmask 255.255.255.252
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): adding default IPv4 route via 10.102.143.34
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): adding primary DNS at 10.10.32.131
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): adding secondary DNS at 8.8.8.8
Fri Jul 23 19:14:20 2021 daemon.notice netifd: wan3 (29192): Command failed: Unknown error
Fri Jul 23 19:14:21 2021 daemon.notice netifd: wan3 (29534): stopping network
Fri Jul 23 19:14:21 2021 daemon.notice netifd: wan3 (29534): IPv4 connection teardown required in interface wan3: static
Fri Jul 23 19:14:21 2021 daemon.notice netifd: wan3 (29534): running disconnection (common)
Fri Jul 23 19:14:21 2021 daemon.notice netifd: wan3 (29534): Command failed: Permission denied
Fri Jul 23 19:14:21 2021 daemon.info [4911]: <info>  [modem0] state changed (connected -> disconnecting)
Fri Jul 23 19:14:21 2021 daemon.info [4911]: <info>  [modem0] state changed (disconnecting -> registered)
Fri Jul 23 19:14:21 2021 daemon.info [4911]: <info>  [modem0/bearer9] connection #1 finished: duration 0s, tx: 0 bytes, rx :0 bytes
Fri Jul 23 19:14:21 2021 daemon.notice netifd: wan3 (29534): successfully disconnected all bearers in the modem
Fri Jul 23 19:14:21 2021 daemon.info [4911]: <info>  [modem0] state changed (registered -> disabling)
Fri Jul 23 19:14:21 2021 daemon.info [4911]: <info>  [modem0] 3GPP registration state changed (home -> unknown)
Fri Jul 23 19:14:21 2021 daemon.notice netifd: wan3 (29534): successfully disabled the modem
Fri Jul 23 19:14:21 2021 daemon.info [4911]: <info>  [modem0] state changed (disabling -> disabled)
Fri Jul 23 19:14:21 2021 daemon.notice netifd: Interface 'wan3' is now down
[...]
```

Ysurac, 2021-07-24:
Make sure that "Use DNS servers advertised by peer" is unchecked in advanced settings of the interface in network->interfaces

starkovv, 2021-07-24:
I've unchecked the "Use DNS servers advertised by peer" and rebooted router. Same result. System logs appears in cycle.

I've added custom DNS 8.8.8.8 to the interface and rebooted router. Same result. System logs appears in cycle.
Also the line `daemon.notice netifd: wan3 (29192): adding primary DNS at 10.10.32.131` continues to appear, like I never performed all these operations above or OMR ignores it.

starkovv, 2021-07-25:
Is it possible to enable debug mode for system logs, so I could see what ModemManager commands cause the errors?

starkovv, 2021-08-11:
UP

Just tested all the modems on Ubuntu 20.04, everything works smoothly both with ModemManager (mmcli) and NetworkManager (nmcli).

With OMR v0.58.5 r4s (aarch64_generic) the `mmcli -m 0` displays that the modem constantly change state in circle: enabling –> enabled –> registering –> registered –> connecting –> connected –> disconnecting –> disconnected –> disabling –> disabled. Each round literally takes 1-2 seconds.

Should we mark this issue as "bug"?

rumynio, 2021-08-23:
I have the same problem with mbim in version 0.58.5.
Before that I was using v0.57.3 and everything worked fine in it.
I would very much like to fix the problem.
In addition, in version 0.58.5, one of the 4 QMI modems constantly falls off

Ysurac, 2021-08-24:
You can test latest snapshot image, I modified mbim/qmi/modemmanager packages release.

closed 2021-11-29 by github-actions[bot]