Ysurac/openmptcprouter #1971: PR_CONNECT_RESET_ERROR

MarcSN311, 2021-07-21:
## Expected Behavior
Using openmptcprouter schould not break the internet connection

## Current Behavior
- When opening a Website in Firefox I get PR_CONNECT_RESET_ERROR
- Chrome also doesn't work

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 0.58.5
  - OpenMPTCProuter platform: x86_64
 
[image]

Ysurac, 2021-07-21:
You have a very high usage on the VPS. Check via htop for example what is using so much CPU.
You get the error for all websites ?

MarcSN311, 2021-07-21:
It is a Hetzner root server with i7-4770 and 32GB memory, that also had a few docker containers running.
I get the error for all websites, doesn't look like IP blocking or something like that.

Ysurac, 2021-07-21:
Check log on the VPS, in /var/log/daemon.log on Debian.

MarcSN311, 2021-07-21:
Nothing in there looks suspicious to me. Only some entries like this
```
Jul 21 15:47:42 hive v2ray[668]: 2021/07/21 15:47:42 127.0.0.1:15317 accepted tcp:127.0.0.1:0 [api]
```
While I'm testing.

Ysurac, 2021-07-21:
By default shadowsocks is used as proxy, so it's not related.
```curl https://1.1.1.1/``` via SSH on the router is working ?

MarcSN311, 2021-07-21:
yeah curl on the router seems to work. looks like there is something wrong between my computer and the openmptcprouter.

Ysurac, 2021-07-21:
No IPv6 on client computer ? DNS and gateway are OpenMPTCProuter IP ? No other DHCP server on network ? No other box/firewall/router/AP/... between client computer and OpenMPTCProuter ? You tested with another browser ?

MarcSN311, 2021-07-21:
- [x] No IPv6 on client computer ?
- [x] DNS and gateway are OpenMPTCProuter IP ?
- [ ] No other DHCP server on network ?
- [x] No other box/firewall/router/AP/... between client computer and OpenMPTCProuter ?
- [x] You tested with another browser ?

I do have another DHCP on the network, but I blacklisted my test devices MAC on that one, so it gets its config from OpenMPTCProuter.

I moved OpenMPTCProuter and my test device to a new VLAN and now it works. Not sure if it was related to DHCP (why?!) or what else on the other vlan interferes with my connection but I will find it.

Thanks for your help and for a great project.

closed 2021-07-21 by MarcSN311