Ysurac/openmptcprouter #1975: RTMP and SRT streaming

mvmaksnet, 2021-07-23:
When I do speedtest than my usb lte modems divide bandwidth  traffic. But When I try to do streaming via rtmp or srt eveything go to one usb modem. How can I make working on all modems?

Ysurac, 2021-07-23:
You mean watching or uploading a stream ?
Modem with lowest latency is used in priority, when full bandwidth is used next modem with higher latency is used.

mvmaksnet, 2021-07-23:
I do stream from pc rtmp and srt to internet (upload). I have 3 usb lte modems. I am sending 6mbps, so every modem should send 2mbps (2+2+2). Can this be done?

Ysurac, 2021-07-23:
should work yes, even if upload speed if often lower than download speed.

mvmaksnet, 2021-07-23:
But it is not working

mvmaksnet, 2021-07-26:
So is there any solution?

Adorfer, 2021-07-26:
provide more data please. what do iperf tests (like yabs.sh) result?

mvmaksnet, 2021-07-26:
Sorry, I don't know how to do that. Can you help me how can I test on your way.
I used openmptcp older versions and rtmp and srt worked on all modems simultaneously.

mouse2rice, 2021-07-28:
试试更改算法，不要用BBR 改成其他的

mvmaksnet, 2021-07-29:
> 试试更改算法，不要用BBR 改成其他的

Don't understand your language.

satan67, 2021-07-30:
> 抱歉，我不知道该怎么做。你能帮我我如何测试你的方式。
> 我使用了旧版本的 openmptcp 并且 rtmp 和 srt 同时在所有调制解调器上工作。

What version is it?

closed 2021-11-08 by github-actions[bot]