Ysurac/openmptcprouter #1976: Pihole as DNS server for wireguard peer

simon2cyrene, 2021-07-23:
J'utilise le VPN Wireguard du VPS quand je suis en itinérance avec mon laptop. Tout le trafic passe par le tunnel. En revanche, je n'arrive pas à faire en sorte que le PiHole réponde aux requêtes DNS du client VPN.
J'ai essaye de faire des modifications dans la configuration de shorewall et celle de pihole, mais sans succès.
Vous auriez une idée ?
--
I use the Wireguard VPN of the VPS when I am roaming with my laptop. All traffic goes through the tunnel. However, I can't get the PiHole to respond to DNS requests from the VPN client.
I tried to make changes in the shorewall and pihole configuration, but without success.
Do you have an idea?

Ysurac, 2021-07-23:
You may need to add interface in ```/etc/dnsmasq.d/99-omr.conf```

simon2cyrene, 2021-07-24:
Thanks for your reply.
I added `interface=wg0` in `/etc/dnsmasq.d/99-omr.conf`. It works.