Ysurac/openmptcprouter #1978: IPv6 Port Forwarding

iamfreaky, 2021-07-24:
## Expected Behavior
IPv6 Forwarding for a Webserver should work like IPv4

## Current Behavior
IPv6 Does not work

## Specifications

  - OpenMPTCProuter version: v0.58.5 r0+16336-b36068d35d
  - OpenMPTCProuter VPS version: 0.1026 5.4.100-mptcp
  - OpenMPTCProuter platform: x86_64

In /etc/shorewall6/rules you can see a DNAT Rule after changing the Rules on the MPTCP Router. 

Config Rule on Router

config rule
        option src 'vpn'
        option target 'ACCEPT'
        option family 'ipv6'
        list proto 'tcp'
        option dest_port '80'
        option name 'HTTP'
        option dest 'lan'
        list dest_ip 'IPv6 Address'

Config on VPS in /etc/shorewall6/rules

DNAT            net             vpn:$OMR_ADDR   tcp     80      # OMR openmptcprouter redirect router 80 port tcp

Config on VPS IPv4 (IPv4 is working)

DNAT            net             vpn:$OMR_ADDR   tcp     80      # OMR openmptcprouter redirect router 80 port tcp

Internet Access (IPv4 and IPv6) is completly working.

Adorfer, 2021-07-24:
same here. i want to have a webserver in LAN available via VPS-IPV6.

Ysurac, 2021-07-25:
For IPv6 you should have an "ACCEPT" rule on shorewall6, DNAT is not used. I will check how a DNAT can be created.

iamfreaky, 2021-07-25:
well the DNAT Rule get created automatically, so this is a bug or ? i will try to change the shorewall6 rule myself.

iamfreaky, 2021-07-25:
If i change the exact rule from DNAT to ACCEPT i still wont work

Adorfer, 2021-07-25:
> For IPv6 you should have an "ACCEPT" rule on shorewall6, DNAT is not used. I will check how a DNAT can be created.

So it does not work with the UI alone, we have to use ssh to the vps for that?

Ysurac, 2021-07-25:
You can try to add "OMR_ADDR=fe80::a00:2" in ```/etc/shorewall6/rules``` and ```/etc/init.d/shorewall6 restart```

iamfreaky, 2021-07-25:
@Ysurac 

● shorewall6.service - Shorewall IPv6 firewall
   Loaded: loaded (/lib/systemd/system/shorewall6.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2021-07-25 10:59:39 PDT; 33s ago
  Process: 6254 ExecStart=/sbin/shorewall -6 $OPTIONS start $STARTOPTIONS (code=exited, status=25)
 Main PID: 6254 (code=exited, status=25)

Jul 25 10:59:39  shorewall[6254]: Adding rules for DHCP
Jul 25 10:59:39  shorewall[6254]: Compiling TCP Flags filtering...
Jul 25 10:59:39  shorewall[6254]: Compiling /etc/shorewall6/snat...
Jul 25 10:59:39  shorewall[6254]: Compiling MAC Filtration -- Phase 1...
Jul 25 10:59:39  shorewall[6254]: Compiling /etc/shorewall6/rules...
Jul 25 10:59:39  shorewall[6254]:    ERROR: Invalid parameter (OMR_ADDR=fe80::a00:2) /etc/shorewall6/rules (line 24)
Jul 25 10:59:39  root[6400]: ERROR:Shorewall6 start failed
Jul 25 10:59:39  systemd[1]: shorewall6.service: Main process exited, code=exited, status=25/n/a
Jul 25 10:59:39  systemd[1]: shorewall6.service: Failed with result 'exit-code'.
Jul 25 10:59:39  systemd[1]: Failed to start Shorewall IPv6 firewall.

Thats doesnt work, inside the /etc/shorewall6/params.vpn there is also the same IPv6 Address.

iamfreaky, 2021-07-25:
Changing the line from 

`ACCEPT         net             vpn:$OMR_ADDR   tcp     80      # OMR openmptcprouter redirect router 80 port tcp`

to 

`ACCEPT         net             vpn:fe80::a00:2   tcp     80      # OMR openmptcprouter redirect router 80 port tcp`

Will work with shorewall. but traffic still wont be passed down to openmptcp router. tcpdump will show no Packets on the 
omr-6in4-user0 interface.

Ysurac, 2021-07-25:
Sorry ```OMR_ADDR=fe80::a00:2``` should be in ```/etc/shorewall6/params.vpn```
With this, rule should use DNAT.

iamfreaky, 2021-07-25:
Well it was in there allready from the beginning. at least the same IPv6 Address just with a Prefix like so `OMR_ADDR=fe80::a00:2/126`

iamfreaky, 2021-07-25:
Okay so with `OMR_ADDR=fe80::a00:2` in `/etc/shorewall6/params.vpn` and `DNAT            net             vpn:$OMR_ADDR   tcp     80      # OMR openmptcprouter redirect router 80 port tcp` in `/etc/shorewall6/rules` i get an Error Message ` ICMP6, destination unreachable, unreachable address`

Ysurac, 2021-07-25:
Can you ping the IP from the VPS ?

iamfreaky, 2021-07-25:
Yes that is working.

iamfreaky, 2021-07-25:
And of course the other way Arround also works.

Ysurac, 2021-07-25:
Ping the IP you want to redirect to.

iamfreaky, 2021-07-25:
From VPS to Client is also working.

Ysurac, 2021-07-26:
Did you check using ```tcpdump -i eth0 -vv ip6 and port 80``` (or something like that) on the router to see if you have something when you connect to external IP ?

iamfreaky, 2021-07-26:
So yes i did, but now i got something different. So again. 

`Ping on VPS -> OpenMPTCP -> Client` that is working. This basically describes the Routing which takes places. 

`Ping on External Server -> VPS -> OpenMPTCP -> Client` that is NOT working. 

The Other way Round `Ping from Client -> OpenMPTCP -> VPS -> External Server` is working.

For both ping tests im using the Global Address. 

if i do a ` tcpdump -i eth0 -n host XXXX::XXXX` on the VPS i can see that there are ICMP6 Request are incoming for that client, but the same is not present on the  `omr-6in4-user` Interface on the VPS Side.

iamfreaky, 2021-08-02:
Any more Ideas ? @Ysurac

closed 2021-11-06 by github-actions[bot]