Ysurac/openmptcprouter #1980: All Google services returns 403

wnabil, 2021-07-25:
## Expected Behavior

Http requests to Google servers should work
## Current Behavior

When requesting any http service "google play, firebase, analytics, admob, etc..." I received an error message that I do not have permission to view the page with error 403
I'm not sure if google blocking mptcp connections but in my case its very annoying, when I request google search or youtube I have to do the robot check every time.
I noticed when I'm downloading something from google services "Android libraries, Gradle, SDK related requests" using curl or wget I got many "connection reset by peer" errors
Disabling the router and connecting directly with my network works fine with no issues
 
[image]

## Specifications

  - OpenMPTCProuter version: v0.58.4 / v0.58.5
  - OpenMPTCProuter VPS version: v0.58.4
  - OpenMPTCProuter platform: x86_64

Ysurac, 2021-07-25:
Google doesn't block MPTCP, but can block some IP range of VPS/server providers after abuses.
You may have for your VPS an IP that was banned.

wnabil, 2021-07-25:
@Ysurac Thanks for the info, any ideas what to do in that case ?
I know its not related to the project but I could not find anything to help

Ysurac, 2021-07-25:
Using another VPS. Or try to make the IP removed from google ban. When it's only the IP banned, ban should be removed after some weeks.
What is you VPS provider ?

wnabil, 2021-07-25:
OVH and I have many servers for other projects, none blocked only this one
Do you know where I can submit to google a request to remove the ban ? Google support forums are specific support platforms and my issue occurs in all of them 
closing this as unrelated and many thanks for the help

closed 2021-07-25 by wnabil