Ysurac/openmptcprouter #1981: QMI bug Signal not showing up Internet connection lost sometimes

zxwanghongyi, 2021-07-25:
Qmi bug tested with quectel EC20CFHC
## Expected Behavior
1, In openmptcp status, I can see the network type and signal icon at the top right of my 4g module.(also dbus)
2, Get internet connection.

## Current Behavior
1, (Everytime) The "LTE" and signal icon will dissapper after getting internet conncetion(also dbus)
Before:
`
"networktype": "lte",
"donglestate": "",
"signal": "77",
`
After:
`
 "networktype": "",
 "donglestate": "",
 "signal": "",
`

2,Some times(Don't know how to reproduce) in openmptcp status, showing [interface:wwan0,phone number:Failed to conncet to server] and there is no wwan0 in ifconfig. No internet conncetion in openwrt, but when I query with AT command, the module reports lte network is connected. So I think there are someting wrong with the qmi dial software.(may be only works perfect with Qualcomm 4g module? If I use quectel's, the qmi dial software need to be replace by quectel custom ver?)

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: no need
  - OpenMPTCProuter platform: ALL

Ysurac, 2021-07-28:
1. It's only some info I retrieve by a request to uqmi when qmi is used. If for any reason uqmi doesn't answer in less than 1s then it timeout and I have no info. This happen sometimes.
2. Should work with quectel without issues. You can check OpenWrt forum ( https://forum.openwrt.org/ ) to check if there is a solution (if you found something I can then add it to OpenMPTCProuter)

zxwanghongyi, 2021-07-28:
Update of bugs 2
When internet gone(openwrt's internet, not 4g module lost conncetion), uqmi will not respond of any command I send.(such as uqmi -d /dev/cdc-wdm0 --get-data-status) press enter, nothing show up, until I hit ctrl+c, Failed to connect to service will show up.

After I go to htop and kill all the process contains uqmi and /dev/cdc-wdm0(the 4g module that openwrt said no internet), the connection comes back, everything works fine until uqmi freeze next time.

related links
https://forum.openwrt.org/t/solved-lte-qmi-troubles/28379
https://forum.gl-inet.com/t/uqmi-hangs-when-modem-is-active/5982

closed 2021-11-03 by github-actions[bot]