Ysurac/openmptcprouter #1985: Multipath seems to be blocked on specific router only

GTiDon, 2021-07-26:
[image]

Hello

I have this error only on my Mikrotik Router R11e
On my other Mikrotik SXT LTE routers I dont have this problem with the same Sim card

Does anyone know what configuration in the RouterOS setting can cause this problem?
I have tried with firewall disabled and reset to defaults. I still cant get connected

Thanks

GTiDon, 2021-07-27:
Everytime I see this error the whois address is incorrect. It seems to pull from one of the other routers.

I suspect some sort of DNS issue?

Ysurac, 2021-07-28:
Please use the template when you open an issue.
If you are on latest release, the check use interface detected public IP. Delay are possible before data is updated.
It's not related to DNS, and it's only a warning. If all is working there is no issues.

closed 2021-10-31 by github-actions[bot]