Ysurac/openmptcprouter #1986: OpenMPTCPRouter slow Name Resolving?

xemles, 2021-07-26:
Hello,
I'm having troubles understanding why OpenMPTCPRouter is doing that to me:
Whenever I try to go to a website or anything, it takes sometimes up to 20 seconds to actually load it, or I might even have to start the request again.

I was wondering if there was anything to tweak regarding that because that's a really annoying issue... The speed is there, the ping averages 30-50ms on each of my 4Gs (there are 3), yet it seems like I have missed something.

The only settings I changed after reconfiguring, since I thought that was my issue, is switching Glorytun TCP to MLVPN since I had troubles with Discord.

I believe it's something related to DNS but DNS in a router are such an unknown thing to me...

So if anybody was able to help me, that'd be appreciated, thanks!

## Specifications

  - OpenMPTCProuter version: 0.58.5
  - OpenMPTCProuter VPS version: 0.1026
  - OpenMPTCProuter platform: Linksys WRT3200ACM

Ysurac, 2021-07-28:
For DNS, root servers DNS are used by default and this use VPN as it's UDP traffic in most cases.
Check that you don't have a WAN that announce it's own DNS if set as DHCP in Network->Interfaces if you didn't use the wizard to set them.
If it's not the case, you can set another DNS in Network->DHCP&DNS and remove "127.0.0.1#5353". Don't use "8.8.8.8, 4.4.4.4,1.1.1.1,..." as they are used by OMR-Tracker and this can give some strange result.

xemles, 2021-08-01:
[image]
Still seems to be the same, I have to refresh multiple times until it works...
The first case scenario didn't seem to apply so I did the second with these DNS
[image]
Yet it seems to make no difference

Ysurac, 2021-08-01:
Check DNS used by your computer client.
Check also that DNS is always up in State->System log

closed 2021-11-05 by github-actions[bot]